zephyrproject-rtos/zephyr #67908: Add support for `-fsanitize=thread`

clamattia, 2024-01-22:
# Desired Solution
As a developer of zephyr applications, I would like to ensure, that they are correct by building with the flag `-fsanitize=thread` at least for `native_posix` and similar targets.

Ideally there is a Kconfig, that just does the right thing in the background analogously to `CONFIG_ASAN`.

# Alternative
I tried adding `zphyr_compile_options(-fsanitize=thread)`. This led to a lot of linker errors.
Please let me know if there are further alternatives/workarounds.

Let me know, if this is a duplicate (I did not find anything yet with a quick search).

aescolar, 2024-01-23:
Getting something like this to work in the native targets would probably be tricky.
Pasting here the discord discussion for background:

-----
```
aescolar
 — 
I just gave it a quick spin here, it does produce a lot of output (I presume false positives). So it does not get blinded by the thread synchronization, it actually seems to miss it 🙂
einenndamen
 — 
So the mutex do not propperly mark the dirty memory as clean again?
aescolar
 — 
I don't know 🙂
aescolar
 — 
I just tried a silly example:

diff --git a/samples/hello_world/src/main.c b/samples/hello_world/src/main.c
index 2758d75d3f..a916871325 100644
--- a/samples/hello_world/src/main.c
+++ b/samples/hello_world/src/main.c
@@ -5,9 +5,29 @@
  */
 
 #include <stdio.h>
+#include <string.h>
+#include <zephyr/kernel.h>
+
+char smashy_buffer[10];
+
+static K_THREAD_STACK_DEFINE(stacky,128);
+static struct k_thread thready;
+
+void thready_f(void *unused, void *unused1, void *unused2)
+{
+       memcpy(smashy_buffer, "pijamas", 8);
+       printf("%s\n", smashy_buffer);
+}
 
 int main(void)
 {
+       k_thread_create(&thready, &stacky[0], 128,
+                       thready_f, NULL, NULL, NULL,
+                       0, K_USER, K_NO_WAIT);
+
        printf("Hello World! %s\n", CONFIG_BOARD);
+       k_yield();
+       memcpy(smashy_buffer, "bananas", 8);
+       printf("%s\n", smashy_buffer);
        return 0;
 }
And the thread sanitizer had no problem about smashy_buffer (with or without the yield).
I think it may be a bit tricky to have this kind of detection work.
einenndamen
 — 
Hm ok. If native_sim would just map all construct to system-constructs (mutexes, threads etc..), I would have hoped, that it would be easy. But I see, that it is probably not.
aescolar
 — 
it is not working that way. The whole Zephyr OS is running inside the process
einenndamen
 — 
Hm, I see. Maybe this could be a separate use-case?
aescolar
 — 
That would be a very different one 🙂
einenndamen
 — 
Or I mean, A single Process is fine, I think, as long as threads/mutexes/semaphores are mapped? No?
aescolar
 — 
The aim of native_posix/sim is to run the Zephyr kernel itself, both to test it, but also to test the libraries and apps with the kernel
What you are thinking about is a compatibility layer (akin to wine) to translate Zephyr calls into Linux calls. 
einenndamen
 — 
Exactly. From the discussion with you, I understand such a layer is not (yet) available.
Do you know if this is planned?
aescolar
 — 
It is not planned 🙂, I dont think I ever heard of anybody asking for it until today 
einenndamen
 — 
An alternative might be to implement tsan natively for the zephyr kernel.
einenndamen
 — 
😄 But you have heared of somebody asking it 😄
einenndamen
 — 
Might be more usefull, since it might allow to check kernel-threads?
(and the libraries and apps with it)
einenndamen
 — 
Maybe it is possible to write a custom implementation of libtsan for zephyr and link that one?
aescolar
 — 
I have no clue. I never looked into how that library does its thing.
einenndamen
 — 
This way maybe it could be implemented for many ( all) boards?

closed 2024-02-08 by nashif

clamattia, 2024-02-08:
Understandable. Thanks anyway for helping me understand the situation better.