zephyrproject-rtos/zephyr #67924: Add dt overlays during compile time (after zephyr package is included)

JHirsbrunner, 2024-01-22:
**Is your enhancement proposal related to a problem? Please describe.**
I need to generate an API (.h) with an implementation (.c) and devicetree overlay, depending on a json script that the user provides. The generator, a python script, is executed once the build process starts with `west build -p -b ...`. It has its own cmake and CMakeLists.txt wherein the process is executed with [`execute_process`](https://cmake.org/cmake/help/latest/command/execute_process.html#execute-process) or [`add_custom_target`](https://cmake.org/cmake/help/latest/command/add_custom_target.html) / [`add_custom_command`](https://cmake.org/cmake/help/latest/command/add_custom_command.html) and the files are added to the build, like you'd do with normal driver APIs and implementations. The output is in the build directory, where sub-directories are created to store these files (mainly to follow the namespacing guidelines of our company). This works all perfectly, except for the overlay, which is added too late.

**Describe the solution you'd like**
I'd like to add a target that has to be build before the dtc is called. In fact, it doesn't even have to be dt-specific. I think adding scripts that are executed before and after build might be a benefit in general. Something among these lines
```{cmake}
set(MY_GENERATOR_SCRIPT path/to/generator.py)
set(
    FLAGS
    --json ${JSON_PATH}  # The user has to provide this with either set(), or -- -DJSON_PATH="path/to/script.json"
    ...
)
add_custom_target(
    foo
    COMMAND ${PYTHON_EXECUTABLE} ${MY_GENERATOR_SCRIPT} ${FLAGS}
    ...
)
add_dependencies(zephyr_pre_build, foo)
```

**Describe alternatives you've considered**
Currently, I force the user to include an additional cmake file before `find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})` is called in the CMakeLists.txt of the app directory.

JHirsbrunner, 2024-02-12:
Has anyone taken a look at this yet? Is it feasible?
BR, Joel

nordicjm, 2024-03-12:
After discussing this on discord I do not see merit in this idea. If overlays are needed, then the overlays should be present and selected as needed, not generated as part of a build. This requests would be a misuse of overlays

closed 2024-03-12 by JHirsbrunner