zephyrproject-rtos/zephyr #67927: ninja: error: loading 'build.ninja': No such file or directory

alwa-nordic, 2024-01-22:
If a pristine build fails during CMake configure, it leaves a build folder that breaks incremental building. The telltale log message is `ninja: error: loading 'build.ninja': No such file or directory`. This problem does not appear in vanilla non-Zephyr CMake.

**To Reproduce**
Steps to reproduce the behavior:
 - Start with the hello world sample.
 - Make sure there is no build folder yet.
 - Add some error to CMakeLists.txt. E.g. append `this_is_not_defined()`.
 - Run `west build -b native_posix`.
 - Observe telltale error.
 - Fix error in CMakeLists.txt.
 - Run `west build` for incremental build.
 - Observe bug.

The workaround is to delete the build folder.

**Impact**
User confusion, major annoyance.

junaidaslamRS, 2024-04-18:
Looks like this is not fixed yet. I did a fresh clone of the mainline today, and it is still there...

theob-pro, 2024-06-18:
I am experiencing this issue too, is there some plan to investigate it?

tejlmand, 2024-09-12:
This is not a `west build` or Zephyr issue.

Incremental builds are done using `cmake --build <build-folder>` and same behavior is seen both with CMake and ninja when trying to do incremental builds.

For example:
```
$ cmake -GNinja -DBOARD=nrf52840dk/nrf52840 -Bbuild samples/hello_world
Loading Zephyr default modules (Zephyr repository).
-- Application: /projects/github/ncs/zephyr/samples/hello_world
...
CMake Error at CMakeLists.txt:7 (foobar):
  Unknown CMake command "foobar".

```

As described in the issue, then fixing the issue and trying to do an incremental build results in:
```
$ cmake --build build
ninja: error: loading 'build.ninja': No such file or directory
```

same with `ninja`:
```
$ ninja -Cbuild
ninja: Entering directory `build'
ninja: error: loading 'build.ninja': No such file or directory
```

`west` can be to forced to re-run CMake in the above case, by using `-c` / `--cmake`.
```
$ west build
ninja: error: loading 'build.ninja': No such file or directory
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /projects/github/ncs/zephyr/build
$ west build -c
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: /projects/github/ncs/zephyr/samples/hello_world
...
[135/135] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       18584 B         1 MB      1.77%
             RAM:        5696 B       256 KB      2.17%
        IDT_LIST:          0 GB        32 KB      0.00%
Generating files from /projects/github/ncs/zephyr/build/zephyr/zephyr.elf for board: nrf52840dk
```

alwa-nordic, 2024-09-12:
I did not say it was a west issue. I added the "area: Build System" label. @thedjnK changed it to "area: West". This is an issue with Zephyr's CMake files. A simple vanilla CMake project does not have this problem. This is a DX issue that should be addressed, so I'll reopen this.

tejlmand, 2024-09-13:
@alwa-nordic you also wrote:
> This problem does not appear in vanilla non-Zephyr CMake.

`west build` is just a wrapper around CMake for convenience.
And as I showed, the exact same errors happens if you try to invoke `cmake --build` or `ninja` on a build folder where the build.ninja is missing.

When you refer to `This problem does not appear in vanilla non-Zephyr CMake.` then I interprete that as you are doing a manual CMake run, and the equivalent to that, when using `west build`, is to run `west build -c` as I wrote in earlier reply.

So the solution to your issue is to use `west build -c` for your incremental build.
Therefore I don't consider this a bug, and thus nothing needs to be fixed.

> The workaround is to delete the build folder.

Use `west build -c` to insure cmake is invoked.

tejlmand, 2024-09-13:
and just to make it clear, from `west help build`:
```
$ west help build
usage: west build ....

Convenience wrapper for building Zephyr applications.

...
cmake and build tool:
  -c, --cmake           force a cmake run
...
```

mmahadevan108, 2024-09-17:
@alwa-nordic , can this issue be closed?

jori-nordic, 2024-09-18:
I still think it's a `west build` failure.
If it was really a "convenient wrapper" it would:
- if there's a `build.ninja` just run `ninja build`
- if there is no `build.ninja`, do a cmake run before

Doesn't seem like rocket science to me, and would solve this UX issue.

tejlmand, 2024-09-18:
> if there's a build.ninja just run ninja build

because it's a wrapper then `west build` technically doesn't know that `ninja` is to be used.
You can run `west build ... -- -G'Unix Makefiles'` and CMake will generate Makefiles instead of ninja files.
Or you can use another generator of your choice, of course depending on your OS.

This is exactly why `west` doesn't call `ninja` (or any other build tool) but calls `cmake --build` because then it's CMake's responsibility to call the correct build tool.

So what you're basically asking is that CMake should re-run itself when there is a CMake cache but no build file, but such request is better placed at CMake, not west.

jori-nordic, 2024-09-19:
> CMake cache but no build file, but such request is better placed at CMake

I was not aware of that, and the comments before hinted that `west` was calling ninja itself.

tejlmand, 2024-09-19:
> and the comments before hinted that west was calling ninja itself.

I actually tried to avoid such misunderstanding in my first comment here: https://github.com/zephyrproject-rtos/zephyr/issues/67927#issuecomment-2346356231

> This is not a `west build` or Zephyr issue.
>
> Incremental builds are done using `cmake --build <build-folder>` ...

alwa-nordic, 2024-09-20:
Ok, so from a purely DX-perspective, the issue is thus: It appears `west build` is a unitary command that will normally perform both the configure and build steps as needed. But this mental picture fails in the described case.

There is an argument that in CMake, the underlying system, the configure and build tasks are entirely separate. Counterargument: Since these are not separate steps in West, this should be hidden as an implementation detail if possible. Especially since the resulting error is not good (it's not instructive for new users).

How does the "configure when needed" work today?: Internally, there two separate triggers for starting the configure: West does it when there is no build folder yet. And the generated Makefile has deps on the CMake files and reruns the configure step when those change.

How does it fail?: A failure during the configure step will leave a build folder, but not a Makefile. West only checks for the presence of the build folder to determine if the Makefile can take the responsibility of "configuring when needed". But with no Makefile, the make program does not take this responsibility.

Solutions:
 - Make West by default always perform the configure step. This is the safest option. It can be made very fast. The configure step is also incremental in CMake and uses the CMake cache that can be used to speed up the reruns. I'm for this.
 
   Modern CMake has "workflows" that can be defined in CMakePresets.txt, and in that system it seems intended to just run the configure step every time.
 - Make West's initial configure test more strict. I looked over the variables in a failed CMakeCache.txt, but I can't see anything that plainly records the success or failure, but I noticed there is a file CMakeFiles/TargetDirectories.txt that only written if in the case of success. This file exists both with Ninja and GNU Make as generator. I don't know what this file is tho and documentation is hard to find. Could we use the presence of this file as the check?

alwa-nordic, 2024-09-20:
Solution:
 - West leaves a marker file in the build directory upon the first successful configure (looking at exit code of CMake) and uses the presence of that file to determine the handover.

tejlmand, 2024-11-14:
> * Make West by default always perform the configure step. This is the safest option. It can be made very fast. The configure step is also incremental in CMake, in the sense that it should use the CMake cache to speed up the reruns. I'm for this.

and this behavior is already supported by `west build` as I mentioned here: https://github.com/zephyrproject-rtos/zephyr/issues/67927#issuecomment-2348989172.
> > So the solution to your issue is to use `west build -c` for your incremental build.```

A reason why this should **not** be default, is because if you always re-run CMake, then you always get:
```
$ west build -b nrf52840dk/nrf52840 samples/hello_world -c
-- west build: generating a build system
Re-run cmake no build system arguments
Loading Zephyr default modules (Zephyr base (cached)).
...
-- west build: building application
[1/7] ...
[7/7] ...
Memory region         Used Size  Region Size  %age Used
           FLASH:       18684 B         1 MB      1.78%
             RAM:        5696 B       256 KB      2.17%
        IDT_LIST:          0 GB        32 KB      0.00%
Generating files from /projects/github/ncs/zephyr/build/zephyr/zephyr.elf for board: nrf52840dk
$ 
```
Sure that is just builds 7 steps, basically because of a re-link, and sounds like @alwa-nordic is fine with that, but I would like to see:
```
$ west build -b nrf52840dk/nrf52840 samples/hello_world/
ninja: no work to do.
$
```

which is not possible if you always run CMake configure.

What I could consider acceptable is to support a `west config build.cmake single` / `west config build.cmake always` which would then allow users to omit the `-c` / `--cmake` flag, and still have CMake to re-run on each incremental build.

Because `west build` is intended to be a wrapper, then I'm not in favor of trying to outsmart the system or make a build system on top a build system. One request for a certain behavior leads to another, and soon the complexity grows.

But i am in favor of allowing users to adjust the default behavior of `west build` using config settings for `cmake`, `pristine`, etc.