zephyrproject-rtos/zephyr #67930: drivers: nxp_enet_experimental: Performance is very poor

decsny, 2024-01-22:
TX/RX performance of the nxp enet driver is very poor compared to older eth_mcux driver, despite almost the exact same code paths in tx/rx, issue is probably due to differences in flexspi prefetching?

Difference can be seen by using the zperf sample. Interestingly, enabling debug optimizations or speed optimizations will fix the issue, again probably due to a different image causing different prefetch pattern in flexspi?

ngphibang, 2024-10-09:
+1 for this.
We use ENET1G for a demo and enabled speed optimization as well as relocating to itcm but the speed is still very poor than expected.

decsny, 2025-02-12:
> Is NXP planning to refactor the network driver soon?

can you clarify what you mean

ofirshe, 2025-02-12:
@decsny Apply the suggestions you mentioned above (or any other solutions) to improve the driver's performance.

decsny, 2025-02-12:
> [@decsny](https://github.com/decsny) Apply the suggestions you mentioned above (or any other solutions) to improve the driver's performance.

It's not scheduled. There was someone working on some of it but they are no longer with NXP now from what I heard. Are you having a problem with the performance?

decsny, 2025-02-12:
> [@decsny](https://github.com/decsny) Apply the suggestions you mentioned above (or any other solutions) to improve the driver's performance.

The general recommendation to improve performance is to relocate the entire networking stack to ITCM on the RT parts to avoid the flexspi fetching and this allows basically the maximum performance of the part. Relocating the ethernet driver itself however doesn't have much effect on the performance which makes us think the bottleneck is not really in the driver, so it's not a priority to "refactor" the driver