zephyrproject-rtos/zephyr #67954: [Coverity CID: 330660] Out-of-bounds access in subsys/bluetooth/controller/ll_sw/ull_adv_iso.c

zephyrbot, 2024-01-23:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/69522b8694e11a0597935169c7d5c0ece0309496/subsys/bluetooth/controller/ll_sw/ull_adv_iso.c

Category: Memory - corruptions
Function: `ll_big_create`
Component: Bluetooth
CID: [330660](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=330660)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/69522b8694e11a0597935169c7d5c0ece0309496/subsys/bluetooth/controller/ll_sw/ull_adv_iso.c#L421

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=330660

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

cvinayak, 2024-02-01:
Earlier analysis of the function use was triaged as false positive, by design the implementation (and by review) should not access out-of-bound memory given the parameter values passed.

Relates to:
https://github.com/zephyrproject-rtos/zephyr/issues/58978
https://github.com/zephyrproject-rtos/zephyr/issues/58970
https://github.com/zephyrproject-rtos/zephyr/issues/67968
https://github.com/zephyrproject-rtos/zephyr/issues/67966
https://github.com/zephyrproject-rtos/zephyr/issues/67954
https://github.com/zephyrproject-rtos/zephyr/issues/67955

cvinayak, 2024-04-15:
2024-04-15 19:02:18 (change by vinayak.kariappa.chettimada@nordicsemi.no)
Comment:

Array not accessed out-of-bound based on the supplied hdr_add_fields/hdr_rem_fields parameter values.
2024-04-15 18:58:03 (change by vinayak.kariappa.chettimada@nordicsemi.no)
Classification:	Unclassified	→	False Positive
Severity:	Unspecified	→	Minor
Action:	Undecided	→	Ignore

cvinayak, 2024-04-15:
<cim-triage-history _ngcontent-pnp-c227=""><div id="source-browser-defect-history" class="collapseable-section"><div class="section-content" style="display: block;"><div class="section-content-inner"><div class="history-item"><h4><span><span>2024-04-15 19:05:19</span><span> (change by vinayak.kariappa.chettimada@nordicsemi.no)</span></span></h4>
Classification: | False Positive | → | Intentional
-- | -- | -- | --

</div></div></div></div></cim-triage-history><cim-defect-occurrences _ngcontent-pnp-c227=""><div id="source-browser-defect-occurrences" class="collapseable-section"><h3 class="section-heading"><span class="ui-icon ui-icon-triangle-1-e"></span></h3></div></cim-defect-occurrences>

closed 2024-04-15 by cvinayak