zephyrproject-rtos/zephyr #67955: [Coverity CID: 330659] Out-of-bounds access in subsys/bluetooth/controller/ll_sw/ull_adv_iso.c

zephyrbot, 2024-01-23:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/69522b8694e11a0597935169c7d5c0ece0309496/subsys/bluetooth/controller/ll_sw/ull_adv_iso.c

Category: Memory - corruptions
Function: `adv_iso_chm_complete_commit`
Component: Bluetooth
CID: [330659](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=330659)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/69522b8694e11a0597935169c7d5c0ece0309496/subsys/bluetooth/controller/ll_sw/ull_adv_iso.c

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=330659

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

cvinayak, 2024-02-01:
Earlier analysis of the function use was triaged as false positive, by design the implementation (and by review) should not access out-of-bound memory given the parameter values passed.

Relates to:
https://github.com/zephyrproject-rtos/zephyr/issues/58978
https://github.com/zephyrproject-rtos/zephyr/issues/58970
https://github.com/zephyrproject-rtos/zephyr/issues/67968
https://github.com/zephyrproject-rtos/zephyr/issues/67966
https://github.com/zephyrproject-rtos/zephyr/issues/67954
https://github.com/zephyrproject-rtos/zephyr/issues/67955

cvinayak, 2024-02-13:
Based on earlier analysis of similar use in implementation, false positive, hence setting priority low.

cvinayak, 2024-04-15:
<div class="history-item"><h4><span><span>2024-04-15 19:04:08</span><span> (change by vinayak.kariappa.chettimada@nordicsemi.no)</span></span></h4>
Classification: | Unclassified | → | Intentional
-- | -- | -- | --
Severity: | Unspecified | → | Minor
Action: | Undecided | → | Ignore

<h5 class="triage-comment">Comment:</h5><p class="triage-comment">Array not accessed out-of-bound based on the supplied hdr_add_fields/hdr_rem_fields parameter values.</p></div>

closed 2024-04-15 by cvinayak