zephyrproject-rtos/zephyr #67957: [Coverity CID: 316393] Missing parentheses in drivers/led_strip/tlc5971.c

zephyrbot, 2024-01-23:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/69522b8694e11a0597935169c7d5c0ece0309496/drivers/led_strip/tlc5971.c

Category: Incorrect expression
Function: `tlc5971_data_byte24`
Component: Drivers
CID: [316393](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=316393)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/69522b8694e11a0597935169c7d5c0ece0309496/drivers/led_strip/tlc5971.c#L140

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=316393

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_CONSTANT_EXPRESSION_RESULT). ([CWE-783](http://cwe.mitre.org/data/definitions/783.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

henrikbrixandersen, 2024-01-23:
> I am unable to check the coverity reference for this bug when logging in with my GitHub account. I get a 401 unauthorized error.

Not sure why, but here is the text from the issue:

[image]

str4t0m, 2024-02-06:
Closing as Duplicate of #58577, which has not been addressed yet and is still open.

henrikbrixandersen, 2024-02-06:
> Closing as Duplicate of #58577, which has not been addressed yet and is still open.

The two issues appear to address two different functions in the same file.

str4t0m, 2024-02-06:
Sorry I have mixed up the issues. There have been 4 open Github issues for the CIDS 316178 and 316393 in Coverity.
I wanted to mark is as duplicate of #58590

henrikbrixandersen, 2024-02-07:
Duplicate of #58590

closed 2024-02-07 by henrikbrixandersen