zephyrproject-rtos/zephyr #67964: [Coverity CID: 316178] Missing parentheses in drivers/led_strip/tlc5971.c

zephyrbot, 2024-01-23:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/69522b8694e11a0597935169c7d5c0ece0309496/drivers/led_strip/tlc5971.c

Category: Incorrect expression
Function: `tlc5971_data_byte25`
Component: Drivers
CID: [316178](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=316178)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/69522b8694e11a0597935169c7d5c0ece0309496/drivers/led_strip/tlc5971.c#L127

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=316178

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_CONSTANT_EXPRESSION_RESULT). ([CWE-783](http://cwe.mitre.org/data/definitions/783.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

simonguinot, 2024-01-23:
I am unable to check the coverity reference for this bug when logging in with my GitHub account. I get a 401 unauthorized error.

str4t0m, 2024-02-06:
Closing as duplicate of #58577, which has not been addressed yet and is still open.

closed 2024-02-06 by str4t0m