zephyrproject-rtos/zephyr #67968: [Coverity CID: 248350] Out-of-bounds access in subsys/bluetooth/controller/ll_sw/ull_adv_sync.c

zephyrbot, 2024-01-23:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/69522b8694e11a0597935169c7d5c0ece0309496/subsys/bluetooth/controller/ll_sw/ull_adv_sync.c

Category: Memory - corruptions
Function: `ull_adv_sync_chm_complete`
Component: Bluetooth
CID: [248350](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=248350)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/69522b8694e11a0597935169c7d5c0ece0309496/subsys/bluetooth/controller/ll_sw/ull_adv_sync.c

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=248350

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

ceolin, 2024-01-31:
Changed priority based on Coverity impact and code location.

cvinayak, 2024-02-01:
Earlier analysis of the function use was triaged as false positive, by design the implementation (and by review) should not access out-of-bound memory given the parameter values passed.

Relates to:
https://github.com/zephyrproject-rtos/zephyr/issues/58978
https://github.com/zephyrproject-rtos/zephyr/issues/58970
https://github.com/zephyrproject-rtos/zephyr/issues/67968
https://github.com/zephyrproject-rtos/zephyr/issues/67966

cvinayak, 2024-04-15:
<div class="history-item"><h4><span><span>2024-04-15 19:09:48</span><span> (change by vinayak.kariappa.chettimada@nordicsemi.no)</span></span></h4>
Classification: | Unclassified | → | Intentional
-- | -- | -- | --
Severity: | Major | → | Minor
Action: | Undecided | → | Ignore

</div>

closed 2024-04-15 by cvinayak