zephyrproject-rtos/zephyr #68000: `Unable to connect to a debugger.` error on `west flash`

IvanVnucec, 2024-01-23:
**Describe the bug**
I've tried to flash software to a custom board over J-Link clone. I do `west flash` and the program exits with the message `ERROR: Unable to connect to a debugger.`.

**Logs and console output**
```console
$ west flash
-- west flash: rebuilding
[1/1] cd /home/ivan/zephyrproject/zephyr/build/zephyr && /home/ivan/zephyrproject/.venv/bin/p...d/check_init_priorities.py --elf-file=/home/ivan/zephyrproject/zephyr/build/zephyr/zephyr.elf
-- west flash: using runner nrfjprog
-- runners.nrfjprog: reset after flashing requested
Using board 4294967295
-- runners.nrfjprog: Flashing file: /home/ivan/zephyrproject/zephyr/build/zephyr/zephyr.hex
[error] [ Client] - Encountered error -102: Command connect_to_emu_with_snr executed for 131 milliseconds with result -102
ERROR: Unable to connect to a debugger.
[error] [ Worker] - An unknown error.
ERROR: JLinkARM DLL reported an error. Try again. If error condition
ERROR: persists, run the same command again with argument --log, contact Nordic
ERROR: Semiconductor and provide the generated log.log file to them.
NOTE: For additional output, try running again with logging enabled (--log).
NOTE: Any generated log error messages will be displayed.
FATAL ERROR: command exited with status 33: nrfjprog --program /home/ivan/zephyrproject/zephyr/build/zephyr/zephyr.hex --sectoranduicrerase --verify -f NRF52 --snr 4294967295
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK

IvanVnucec, 2024-01-23:
I've ran `nrfjprog --program /home/ivan/zephyrproject/zephyr/build/zephyr/zephyr.hex --sectoranduicrerase --verify -f NRF52 --snr 4294967295 --log` directly and still the same error.

`log.log` file is:
```log
[2024-Jan-23 13:48:51] [debug] --------------------------------------------------------------------------------
[2024-Jan-23 13:48:51] [debug] nrfjprog --recover --log 
[2024-Jan-23 13:48:51] [debug] nrfjprog version 10.23.4 external
[2024-Jan-23 13:48:51] [debug] --------------------------------------------------------------------------------
[2024-Jan-23 13:48:51] [ info] Load library at /opt/nrf-command-line-tools/lib/libnrfjprogdll.so.
[2024-Jan-23 13:48:51] [ info] Library loaded, loading member functions.
[2024-Jan-23 13:48:51] [ info] Member functions succesfully loaded.
[2024-Jan-23 13:48:51] [debug] [ Client] - open
[2024-Jan-23 13:48:51] [debug] [ Client] - start
[2024-Jan-23 13:48:51] [ info] [ Client] - stdout: Jlinkarm nRF Worker ready. Handling sequence 73df501a-281e-4e7e-afdf-3874e4525e18.
[2024-Jan-23 13:48:51] [debug] [SeggerBackend] - Logger sink registered in Segger backend logger
[2024-Jan-23 13:48:51] [debug] [  JLink] - Logger sink registered in JLink logger
[2024-Jan-23 13:48:51] [debug] [  nRFXX] - open
[2024-Jan-23 13:48:51] [debug] [  nRFXX] - just_check_family
[2024-Jan-23 13:48:51] [debug] [SeggerBackend] - open_dll
[2024-Jan-23 13:48:51] [debug] [SeggerBackend] - No J-Link DLL path was provided. Attempting to auto detect.
[2024-Jan-23 13:48:51] [trace] [ Client] - Command open executed for 7 milliseconds with result 0
[2024-Jan-23 13:48:51] [debug] [ Client] - config
[2024-Jan-23 13:48:51] [trace] [ Client] - Command config executed for 2 milliseconds with result 0
[2024-Jan-23 13:48:51] [debug] [ Client] - enum_emu_snr
[2024-Jan-23 13:48:51] [ info] [SeggerBackend] - Load library at /opt/SEGGER/JLink/libjlinkarm.so.7.
[2024-Jan-23 13:48:51] [ info] [SeggerBackend] - Library loaded, loading member functions.
[2024-Jan-23 13:48:51] [ info] [SeggerBackend] - Member functions succesfully loaded.
[2024-Jan-23 13:48:51] [debug] [SeggerBackend] - Set batch mode
[2024-Jan-23 13:48:51] [debug] [SeggerBackend] - dll_version
[2024-Jan-23 13:48:51] [ info] [SeggerBackend] - Segger dll version 7.94.f loaded.
[2024-Jan-23 13:48:51] [trace] [ Worker] - Command open executed for 5 milliseconds with result 0
[2024-Jan-23 13:48:51] [debug] [  nRFXX] - config
[2024-Jan-23 13:48:51] [debug] [SeggerBackend] - enum_emu_snr
[2024-Jan-23 13:48:51] [trace] [ Worker] - Command config executed for 2 milliseconds with result 0
[2024-Jan-23 13:48:51] [debug] [  nRFXX] - enum_emu_con_info
[2024-Jan-23 13:48:51] [debug] [SeggerBackend] - is_connected_to_emu
[2024-Jan-23 13:48:51] [debug] [SeggerBackend] - enum_emu_con_info
[2024-Jan-23 13:48:51] [trace] [ Client] - Command enum_emu_con_info executed for 4 milliseconds with result 0
[2024-Jan-23 13:48:51] [debug] [ Client] - connect_to_emu_with_snr
[2024-Jan-23 13:48:51] [debug] [SeggerBackend] - is_connected_to_emu
[2024-Jan-23 13:48:51] [trace] [ Worker] - Command enum_emu_con_info executed for 4 milliseconds with result 0
[2024-Jan-23 13:48:51] [debug] [  nRFXX] - connect_to_emu_with_snr
[2024-Jan-23 13:48:51] [debug] [SeggerBackend] - is_connected_to_emu
[2024-Jan-23 13:48:51] [debug] [SeggerBackend] - connect_to_emu_with_snr
[2024-Jan-23 13:48:51] [debug] [SeggerBackend] - is_connected_to_emu
[2024-Jan-23 13:48:51] [debug] [SeggerBackend] - ---just_enum_emu_snr
[2024-Jan-23 13:48:51] [debug] [SeggerBackend] - ---just_get_num_emus
[2024-Jan-23 13:48:51] [debug] [SeggerBackend] - ---just_connect_to_emu_with_snr
[2024-Jan-23 13:48:51] [debug] [SeggerBackend] - ---just_connect_to_emu_without_snr
[2024-Jan-23 13:48:51] [debug] [SeggerBackend] - Segger logging enabled.
[2024-Jan-23 13:48:51] [trace] [  JLink] - Firmware: J-Link ARM V8 compiled Nov 28 2014 13:44:46
[2024-Jan-23 13:48:51] [trace] [  JLink] - Firmware: J-Link ARM V8 compiled Nov 28 2014 13:44:46
[2024-Jan-23 13:48:51] [trace] [  JLink] - Hardware: V8.00
[2024-Jan-23 13:48:51] [trace] [  JLink] - S/N: 4294967295
[2024-Jan-23 13:48:51] [trace] [  JLink] - OEM: SEGGER
[2024-Jan-23 13:48:51] [trace] [  JLink] - Feature(s): RDI,FlashDL,FlashBP,JFlash,GDB
[2024-Jan-23 13:48:51] [trace] [  JLink] - Bootloader: (Could not read)
[2024-Jan-23 13:48:51] [trace] [  JLink] - TELNET listener socket opened on port 19021
[2024-Jan-23 13:48:51] [trace] [  JLink] - WEBSRV WEBSRV_Init(): Starting webserver thread(s)
[2024-Jan-23 13:48:51] [trace] [  JLink] - WEBSRV Webserver running on local port 19080
[2024-Jan-23 13:48:51] [trace] [  JLink] - Looking for J-Link GUI Server exe at: /opt/SEGGER/JLink/JLinkGUIServerExe 
[2024-Jan-23 13:48:51] [trace] [  JLink] - Forking J-Link GUI Server: /opt/SEGGER/JLink/JLinkGUIServerExe 
[2024-Jan-23 13:48:51] [trace] [  JLink] - J-Link GUI Server info: "J-Link GUI server V7.94f " 
[2024-Jan-23 13:48:51] [trace] [  JLink] - - 55.278ms returns "O.K."  
[2024-Jan-23 13:48:51] [trace] [  JLink] - JLINK_HasError()  
[2024-Jan-23 13:48:51] [trace] [  JLink] - JLINK_Lock()  
[2024-Jan-23 13:48:51] [trace] [  JLink] - JLINK_ExecCommand("SetRestartOnClose = 0", ...).   
[2024-Jan-23 13:48:51] [trace] [  JLink] - - 0.003ms returns 0x01  
[2024-Jan-23 13:48:51] [trace] [  JLink] - JLINK_HasError()  
[2024-Jan-23 13:48:51] [trace] [  JLink] - JLINK_ExecCommand("DisableFlashDL", ...).   
[... 22 lines cut ...]
[2024-Jan-23 13:48:51] [trace] [  JLink] - JLINK_HasError()  
[... cut ...]
[2024-Jan-23 13:48:51] [trace] [  JLink] - JLINK_HasError()  
[2024-Jan-23 13:48:51] [trace] [  JLink] - JLINK_GetSN()  
[2024-Jan-23 13:48:51] [trace] [  JLink] - - 0.004ms returns -1  
[2024-Jan-23 13:48:51] [trace] [  JLink] - JLINK_HasError()  
[2024-Jan-23 13:48:51] [trace] [  JLink] - - 0.015ms  
[2024-Jan-23 13:48:51] [trace] [  JLink] - JLINK_Close()  
[2024-Jan-23 13:48:51] [error] [ Client] - Encountered error -102: Command connect_to_emu_with_snr executed for 110 milliseconds with result -102

```

IvanVnucec, 2024-01-23:
I think the problem might be with the J-Link clone because looking at the logs, `JLINK_GetSN()` command returns `-1` multiple times.

nordicjm, 2024-01-23:
> over J-Link clone

What do you expect?

closed 2024-01-23 by nordicjm