zephyrproject-rtos/zephyr #68003: subsys/net/lib/lwm2m/lwm2m_rw_cbor.c violates coding guidelines rule A.5

aescolar, 2024-01-23:
**Describe the bug**
Zephyr's coding guideline rule A.5 states which C library functions can be used by (embedded) code in the Zephyr tree
https://docs.zephyrproject.org/latest/contribute/coding_guidelines/index.html#rule-a-5-c-standard-library-usage-restrictions-in-zephyr-codebase
subsys/net/lib/lwm2m/lwm2m_rw_cbor.c uses gmtime_r:
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/lwm2m/lwm2m_rw_cbor.c#L46
which is not between these, and therefore is breaking this rule.

**To Reproduce**
N/A, code inspection

**Expected behavior**
No use of C library functions beyond those in that list in embedded Zephyr code (outside of the POSIX compatibility API layer)

**Impact**
Prevents progress in cleaning up the too wide setting of _POSIX_C_SOURCE macros.

**Logs and console output**
N/A

**Environment (please complete the following information):**
N/A

**Additional context**
Issue has been present since the file was added 8fd283a788f0198cc31a9ad5fa7f94c6588f81f2

rlubos, 2024-01-23:
@stephanosio I see that when the A.5 rule was introduced, [the commit message](https://github.com/zephyrproject-rtos/zephyr/commit/400c1fa744b0a273bc42fc5c9785b1c14dadfe14) says that:
> The initial list of allowed non-ISO C libc functions is based on the
non-ISO C functions available in the minimal libc.

Just trying to understand here - any particular reason `gmtime_r()` was not included in the exception list? The function has been part of minimal libc for quite some time (https://github.com/zephyrproject-rtos/zephyr/commit/3e8df8b369999d786a76b565f39b4e69afc8e8e6).

cfriedt, 2024-01-23:
> gmtime_s

`gmtime_s()` is part of C11 and appears to be nearly identical to `gmtime_r()` (which is part of C23).

https://en.cppreference.com/w/c/chrono/gmtime

I think either of the possibilities mentioned by @rlubos would be fine - either adding `gmtime_s()` in the minimal libc and changing affected libraries in Zephyr, or simply adding an exception for `gmtime_r()` (since it's a part of the most recent ISO C standard anyway)

aescolar, 2024-01-26:
At this point, I think it is best to lower this issue priority to "low".
After 4ff79cb74ac7c68957ce9a41847da4955dcb2721 the impact of this issue is limited to the rule violation.
As `gmtime_r()` is provided by all libraries we use in-tree it does not seem to break the spirit of rule A.5.
Moreover this function is used in several other components in tree.
So my suggestion would be to add this function to the exceptions of rule A.5.
CC: @keith-zephyr @nashif @carlescufi @jfischer-no @stephanosio @keith-packard @kartben

stephanosio, 2024-01-29:
> @stephanosio I see that when the A.5 rule was introduced, [the commit message](https://github.com/zephyrproject-rtos/zephyr/commit/400c1fa744b0a273bc42fc5c9785b1c14dadfe14) says that:
> 
> > The initial list of allowed non-ISO C libc functions is based on the
> > non-ISO C functions available in the minimal libc.
> 
> Just trying to understand here - any particular reason `gmtime_r()` was not included in the exception list? The function has been part of minimal libc for quite some time ([3e8df8b](https://github.com/zephyrproject-rtos/zephyr/commit/3e8df8b369999d786a76b565f39b4e69afc8e8e6)).

@rlubos I do not remember the exact details; but, looking through the history, it looks like `gmtime_r` was simply missed due to an oversight -- it should be added to the exception list since the standard `gmtime` is not a viable alternative to `gmtime_r`.

Note that `gmtime_r` is currently only part of the minimal libc and, as per the Rule A.5, it is necessary to move its implementation to the common libc in order for it to be added as an exception:

> All non-standard functions and macros listed above must be implemented by the
:ref:`common libc <c_library_common>` in order to make sure that these
functions can be made available when using a C standard library that does not
implement these functions.

As for adding `gmtime_r` (POSIX and C23) vs. `gmtime_s` (Annex K) as an exception, `gmtime_r` should be preferred since even the ISO themselves no longer believe in the future of Annex K and, moreover, the formerly POSIX-only `gmtime_r` has been made part of C23.

PR #68414 (2024-02-27): doc: Add gmtime_r() to rule A.5 exceptions 
Add `gmtime_r()` to rule's A.5 exception list, and move its implementation to common libc area, as per @stephanosio [comment](https://github.com/zephyrproject-rtos/zephyr/issues/68003#issuecomment-1913872276) in #68003.

Fixes #68003
Fixes #68056
closed 2024-02-27 by stephanosio