zephyrproject-rtos/zephyr #68010: doc: Documentation issue in 'develop/getting_started/installation_linux': Missing packages elftools and serial

justpresident, 2024-01-23:
**Describe the bug**

After following the instructions on Fedora I had following python packages missing: elftools and serial.

I installed them manually with:
```
sudo dnf install python3-pyelftools python3-pyserial
```
After that compilation blinky example succeeded with: ```west build -b esp_wrover_kit samples/basic/blinky```

**Environment**

* Page: `develop/getting_started/installation_linux`
* Version: main
* SHA-1: 24cf2f6548edd480ec10719f5f1fe8d1560c034d

kartben, 2024-01-23:
Hi @justpresident -- it looks you may have missed some steps in the Getting Started guide, w.r.t installing the Python dependencies for Zephyr. Please check https://docs.zephyrproject.org/latest/develop/getting_started/index.html#get-zephyr-and-install-python-dependencies.

I apppreciate that maybe the intro in the Getting Started Guide is confusing, as it "diverts" you to the page you linked above to setup your Linux box, but then (too...) implicitly invites you to come back to it for the next steps "Get the Source code" and "Build, Flash, and run a sample application"

justpresident, 2024-01-23:
@kartben , I did follow these steps. And I see now that pyserial and pyelftools are in requirements-base.txt
But for some reason they were not installed or are not picked up during `west build`

kartben, 2024-01-24:
@justpresident I just tried with a vanilla fedora Docker image and it worked fine for me, are you sure you were running from the same python venv both when installing the requirements and building?

kartben, 2024-01-24:
Note that I am proposing an update to the list of packages that need to be installed as part of #68051, as it looks like the list could use an update.

justpresident, 2024-01-24:
> I just tried with a vanilla fedora Docker image and it worked fine for me, are you sure you were running from the same python venv both when installing the requirements and building?

Well, this means that I indeed did something wrong. But I am not sure how is it possible that only these two packages were not installed.
I suppose this can be closed. Sorry for the trouble

kartben, 2024-01-25:
No reason to be sorry! Please do re-open if you end up narrowing it down to an actual problem in the install instructions.
Thanks!

closed 2024-01-25 by kartben