zephyrproject-rtos/zephyr #68018: doc:  SCD30 CO2 sensor support documentation inconsistency in Zephyr (version >= 3.3)

pasrom, 2024-01-23:
**Describe the bug**
In the release notes for version 3.3, the Zephyr documentation explicitly lists support for the SCD30 CO2 sensor.
https://github.com/zephyrproject-rtos/zephyr/blob/1d23ac55777cf6ad8f7cebebeaf043bc7f9d99ff/doc/releases/release-notes-3.3.rst#L3722
This inconsistency is due to an issue being reported (https://github.com/zephyrproject-rtos/zephyr/issues/39234) and a related pull request being submitted (https://github.com/zephyrproject-rtos/zephyr/pull/39236) for the sensor's integration, but the PR was not merged into the main code base.

**Expected behavior**
The documentation should accurately reflect the hardware support that exists in the code base.

**Environment**
Zephyr Version: >= 3.3

**Additional context**
To resolve this, either the documentation should be updated to reflect the actual state of sensor support, or the sensor support should be integrated into the code. For the latter, an updated driver is available [here](https://github.com/nobodyguy/sensirion_zephyr_drivers).

kartben, 2024-01-23:
I am not sure how easy this would be to address, as there's definitely more than just the linked release note entry that's bogus, see #57966. Not sure we'd want to amend past release notes, as it is an artefact of a previous release that one would expect to not fluctuate over time. Maybe we should add a notice to all the impacted release notes that links to #57966 to indicate they might not reflect the reality of things that were actually fixed?

pasrom, 2024-01-24:
> Maybe we should add a notice to all the impacted release notes that links to https://github.com/zephyrproject-rtos/zephyr/issues/57966 to indicate they might not reflect the reality of things that were actually fixed?

This would be a good approach.

closed 2024-06-14 by github-actions[bot]