zephyrproject-rtos/zephyr #68028: ESP32C3 S3 MCUBoot seg faults/hangs - suspect cache corruption

dean-at-evos, 2024-01-23:
**Describe the bug**
Apologies for cross posting from discord, but I am not getting much traction there, so I am widening the net :)

When enabling the MCUBoot bootloader and using sysbuild for the esp32c3 and esp32s3 SOC the application segfaults/or hangs on entry to main. The same application built to use the espressif IDF 2nd stage bootloader runs fine. I can reproduce the issue minimally by building the z-bus hello_world sample.

My strong suspicion is the the cache is not being left in the correct state at the end of the function `map_rom_segments(void)` in `soc/<arch>/espressif_esp32/<soc>/loader.c`. Both SOC have very similar code, which surprised me due to the fact one is dual core and there are on different arch, but maybe the HAL sorts this...

Also suspicious in the fn is the clearing of bits in `EXTMEM_DCACHE_CTRL1_REG` which for both resolves to `0x600c4004`, in both cases according to the reference manuals this register address either is marked 'reserved' or the bit(s) in question have no documented function.

The other clue that this a cache issue is that in the case of the C3 I can resolve the problem by log printing `autoload` after the 'cache_resume'. I think that somehow this triggers a cache reset/reload and the application will run fine. Also issues #64297 and #62172 describe the same behaviour, with the same flaky sometimes works, mostly doesn't thing.

Also the ROM functions to suspend and resume the cache are sparsely documented so I am not sure what they do, but the `autoload` value returned from suspend is 0, which also seems strange as I suspect that the cache needs to be in autoload, unless that doesn't mean what I think it means... also passing `true` to the resume doesn't fix things.

To get further clues I have tried delays, setting/clearing cache clk and reset bits as per the manuals (I did think this worked for a bit, except it was the log print I added!). None of these made any difference.

I guess that the loader code has been copied between SOCs and works naively, and intermittently, so has never really been examined. Also the IDF 2nd stage bootloader works as expected. I am happy to work on and contribute fixes, but without some clearer doc for the SOC and/or guidance on how cache setup *should* happen I am a bit stuck. Also, it may not be a cache issue at all ...

**To Reproduce**
`west build --sysbuild -p -b esp32c3_devkitm samples/subsys/zbus/hello_world -- -DCONFIG_BOOTLOADER_MCUBOOT=y`

`west flash`

`west espressif monitor`

Substitute `esp32s3_devkitm` to do the same for s3

**Expected behavior**

Sample to run as expected.

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd8000,len:0x1018
load:0x403cad98,len:0x3d8c
load:0x403d0000,len:0x1a24
entry 0x403cc724
[esp32c3] [INF] MCUboot 2nd stage bootloader
[esp32c3] [INF] compiled at Jan 24 2024 09:08:18
[esp32c3] [INF] Chip revision: 3
[esp32c3] [INF] SPI Flash speed: 40MHz, mode: DIO, size: 4MB
[esp32c3] [INF] Image index: 0, Swap type: none
[esp32c3] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32c3] [INF] Application start=40381E66h
[esp32c3] [INF] DRAM segment: paddr=00005A68h, vaddr=3FC87010h, size=003AAh (   938) load
[esp32c3] [INF] IRAM segment: paddr=000016F0h, vaddr=40380000h, size=04378h ( 17272) load
[esp32c3] [INF] DROM segment: paddr=00010040h, vaddr=3C000040h, size=016B0h (  5808) (al=00000000) map
[esp32c3] [INF] IROM segment: paddr=00020000h, vaddr=42020000h, size=039B8h ( 14776) map
I: Sensor sample started raw reading, version 0.1-2!
I: Channel list:
I: 0 - Channel acc_data_chan:
I:       Message size: 12
I:       Observers:
I:       - foo_lis
I:       - bar_sub
I: 1 - Channel simple_chan:
I:       Message size: 4
I:       Observers:
I: 2 - Channel version_chan:
I:       Message size: 4
I:       Observers:
I: Observers list:
I: 0 - Subscriber bar_sub
I: 1 - Listener foo_lis
I: From listener -> Acc x=1, y=1, z=1
I: From subscriber -> Acc x=1, y=1, z=1
I: From listener -> Acc x=2, y=2, z=2
I: From subscriber -> Acc x=2, y=2, z=2
I: Pub a valid value to a channel with validator successfully.
I: Pub an invalid value to a channel with validator successfully.
```

**Impact**
This is a showstopper for our application. We have a product in last stage hardware revision, hoping to get some pilot devices in the field soon. This issue has cropped up as we were merging together some modules for the final app. The z-bus is used in our IPC chain and the MCUBoot for our firmware and deployment, it was putting them both together that surfaced the issue.

Our board is C3 based but I happened to have an S3 and decided to try it to see if that helped find the issue. So my priority is getting the C3 working, but I can help out with the S3 as well if it needs similar repairs.

**Logs and console output**

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd8000,len:0x1018
load:0x403cad98,len:0x3d8c
load:0x403d0000,len:0x1a24
entry 0x403cc724
[esp32c3] [INF] MCUboot 2nd stage bootloader
[esp32c3] [INF] compiled at Jan 24 2024 08:58:42
[esp32c3] [INF] Chip revision: 3
[esp32c3] [INF] SPI Flash speed: 40MHz, mode: DIO, size: 4MB
[esp32c3] [INF] Image index: 0, Swap type: none
[esp32c3] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32c3] [INF] Application start=40381E66h
[esp32c3] [INF] DRAM segment: paddr=00005A5Ch, vaddr=3FC87010h, size=003AAh (   938) load
[esp32c3] [INF] IRAM segment: paddr=000016E4h, vaddr=40380000h, size=04378h ( 17272) load
[esp32c3] [INF] DROM segment: paddr=00010040h, vaddr=3C000040h, size=016A4h (  5796) map
[esp32c3] [INF] IROM segment: paddr=00020000h, vaddr=42020000h, size=039B8h ( 14776) map
E: 
E:  mcause: 7, Store/AMO access fault
E:   mtval: 3c001670
E:      a0: 00000014    t0: bae01309
E:      a1: 3c001670    t1: 0000000f
E:      a2: 3c0016e8    t2: 690ae800
E:      a3: 00000000    t3: 3c394a1e
E:      a4: 3c0016c8    t4: f191ad25
E:      a5: 3c0016d8    t5: 2c6a2555
E:      a6: 005b7e31    t6: f526b1e7
E:      a7: 1ddc51d1
E:      ra: 403808ba
E:    mepc: 42022816
E: mstatus: 00001880
E: 
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x3fc844f8 (unknown)
E: Halting system
```

``` console
(.venv) you@yours:~/zephyrproject/zephyr$ addr2line -e build/hello_world/zephyr/zephyr.elf 3c001670
/home/you/zephyrproject/zephyr/samples/subsys/zbus/hello_world/src/main.c:35
```

main.c:35, statically defined data, accessed by a LOG_ early in main.
``` c
ZBUS_CHAN_DEFINE(acc_data_chan,  /* Name */
		 struct acc_msg, /* Message type */

		 NULL,                                 /* Validator */
		 NULL,                                 /* User data */
		 ZBUS_OBSERVERS(foo_lis, bar_sub),     /* observers */
		 ZBUS_MSG_INIT(.x = 0, .y = 0, .z = 0) /* Initial value */
);
```

**Environment (please complete the following information):**
 - OS: GNU/Linux
 - Toolchain: zephyr 0.16.0
 - Main

sylvioalves, 2024-01-24:
@dean-at-evos, I still have to perform a dig analysis. Can you check this patch?
```c
diff --git a/soc/riscv/espressif_esp32/esp32c3/default.ld b/soc/riscv/espressif_esp32/esp32c3/default.ld
index 65984b4fd7..284a756d2b 100644
--- a/soc/riscv/espressif_esp32/esp32c3/default.ld
+++ b/soc/riscv/espressif_esp32/esp32c3/default.ld
@@ -142,8 +142,8 @@ SECTIONS
     #include <snippets-rodata.ld>
 
     . = ALIGN(4);
-    *(EXCLUDE_FILE (*libarch__riscv__core.a:* *libkernel.a:fatal.* *libkernel.a:init.* *libzephyr.a:cbprintf_complete* *libzephyr.a:log_core.* *libzephyr.a:log_backend_uart.* *libdrivers__flash.a:esp32_mp.* *libzephyr.a:log_output.* *libzephyr.a:loader.* *libdrivers__flash.a:flash_esp32.* *libdrivers__serial.a:uart_esp32.* *libzephyr.a:spi_flash_rom_patch.*) .rodata)
-    *(EXCLUDE_FILE (*libarch__riscv__core.a:* *libkernel.a:fatal.* *libkernel.a:init.* *libzephyr.a:cbprintf_complete* *libzephyr.a:log_core.* *libzephyr.a:log_backend_uart.* *libdrivers__flash.a:esp32_mp.* *libzephyr.a:log_output.* *libzephyr.a:loader.* *libdrivers__flash.a:flash_esp32.* *libdrivers__serial.a:uart_esp32.* *libzephyr.a:spi_flash_rom_patch.*) .rodata.*)
+    *(EXCLUDE_FILE (*libsubsys__zbus.a:* *libarch__riscv__core.a:* *libkernel.a:fatal.* *libkernel.a:init.* *libzephyr.a:cbprintf_complete* *libzephyr.a:log_core.* *libzephyr.a:log_backend_uart.* *libdrivers__flash.a:esp32_mp.* *libzephyr.a:log_output.* *libzephyr.a:loader.* *libdrivers__flash.a:flash_esp32.* *libdrivers__serial.a:uart_esp32.* *libzephyr.a:spi_flash_rom_patch.*) .rodata)
+    *(EXCLUDE_FILE (*libsubsys__zbus.a:* *libarch__riscv__core.a:* *libkernel.a:fatal.* *libkernel.a:init.* *libzephyr.a:cbprintf_complete* *libzephyr.a:log_core.* *libzephyr.a:log_backend_uart.* *libdrivers__flash.a:esp32_mp.* *libzephyr.a:log_output.* *libzephyr.a:loader.* *libdrivers__flash.a:flash_esp32.* *libdrivers__serial.a:uart_esp32.* *libzephyr.a:spi_flash_rom_patch.*) .rodata.*)
 
     *(.irom1.text) /* catch stray ICACHE_RODATA_ATTR */
     *(.gnu.linkonce.r.*)
@@ -339,7 +339,7 @@ SECTIONS
     *libdrivers__flash.a:flash_esp32.*(.rodata  .rodata.*)
     *libdrivers__serial.a:uart_esp32.*(.rodata .rodata.*)
     *libzephyr.a:spi_flash_rom_patch.*(.rodata  .rodata.*)
-
+    *libsubsys__zbus.a:(.rodata  .rodata.*)
 
     KEEP(*(.jcr))
     *(.dram1 .dram1.*)
```

dean-at-evos, 2024-01-25:
Thanks @sylvioalves I got a 'corrupt patch' warning with that ... but, linker! I did think about that but I got stuck when a LOG print seemed to fix it. Anyway I applied the below to the c3 and it seemed to work. At least the sample ran as expected.

```
diff --git a/soc/riscv/espressif_esp32/esp32c3/default.ld b/soc/riscv/espressif_esp32/esp32c3/default.ld
index 65984b4fd7..1ae4161af4 100644
--- a/soc/riscv/espressif_esp32/esp32c3/default.ld
+++ b/soc/riscv/espressif_esp32/esp32c3/default.ld
@@ -142,8 +142,8 @@ SECTIONS
     #include <snippets-rodata.ld>
 
     . = ALIGN(4);
-    *(EXCLUDE_FILE (*libarch__riscv__core.a:* *libkernel.a:fatal.* *libkernel.a:init.* *libzephyr.a:cbprintf_complete* *libzephyr.a:log_core.* *libzephyr.a:log_backend_uart.* *libdrivers__flash.a:esp32_mp.* *libzephyr.a:log_output.* *libzephyr.a:loader.* *libdrivers__flash.a:flash_esp32.* *libdrivers__serial.a:uart_esp32.* *libzephyr.a:spi_flash_rom_patch.*) .rodata)
-    *(EXCLUDE_FILE (*libarch__riscv__core.a:* *libkernel.a:fatal.* *libkernel.a:init.* *libzephyr.a:cbprintf_complete* *libzephyr.a:log_core.* *libzephyr.a:log_backend_uart.* *libdrivers__flash.a:esp32_mp.* *libzephyr.a:log_output.* *libzephyr.a:loader.* *libdrivers__flash.a:flash_esp32.* *libdrivers__serial.a:uart_esp32.* *libzephyr.a:spi_flash_rom_patch.*) .rodata.*)
+    *(EXCLUDE_FILE (*libsubsys__zbus.a:* *libarch__riscv__core.a:* *libkernel.a:fatal.* *libkernel.a:init.* *libzephyr.a:cbprintf_complete* *libzephyr.a:log_core.* *libzephyr.a:log_backend_uart.* *libdrivers__flash.a:esp32_mp.* *libzephyr.a:log_output.* *libzephyr.a:loader.* *libdrivers__flash.a:flash_esp32.* *libdrivers__serial.a:uart_esp32.* *libzephyr.a:spi_flash_rom_patch.*) .rodata)
+    *(EXCLUDE_FILE (*libsubsys__zbus.a:* *libarch__riscv__core.a:* *libkernel.a:fatal.* *libkernel.a:init.* *libzephyr.a:cbprintf_complete* *libzephyr.a:log_core.* *libzephyr.a:log_backend_uart.* *libdrivers__flash.a:esp32_mp.* *libzephyr.a:log_output.* *libzephyr.a:loader.* *libdrivers__flash.a:flash_esp32.* *libdrivers__serial.a:uart_esp32.* *libzephyr.a:spi_flash_rom_patch.*) .rodata.*)
 
     *(.irom1.text) /* catch stray ICACHE_RODATA_ATTR */
     *(.gnu.linkonce.r.*)
@@ -339,6 +339,7 @@ SECTIONS
     *libdrivers__flash.a:flash_esp32.*(.rodata  .rodata.*)
     *libdrivers__serial.a:uart_esp32.*(.rodata .rodata.*)
     *libzephyr.a:spi_flash_rom_patch.*(.rodata  .rodata.*)
+    *libsubsys__zbus.a:*(.rodata .rodata.*)
 
 
     KEEP(*(.jcr))
``` 

When I tried the similar thing with the s3 linker file, it still segfaults on start. I didn't really chase this down though, I suspect there is a difference because of the D and I cache on the s3 but I haven't had a chance to look into that too much.

```
diff --git a/soc/xtensa/espressif_esp32/esp32s3/default.ld b/soc/xtensa/espressif_esp32/esp32s3/default.ld
index 4076ea7c1a..bd0e3816d8 100644
--- a/soc/xtensa/espressif_esp32/esp32s3/default.ld
+++ b/soc/xtensa/espressif_esp32/esp32s3/default.ld
@@ -178,8 +178,8 @@ SECTIONS
     #include <snippets-rodata.ld>
 
     . = ALIGN(4);
-    *(EXCLUDE_FILE (*libarch__xtensa__core.a:* *libkernel.a:fatal.* *libkernel.a:init.* *libzephyr.a:cbprintf_complete* *libzephyr.a:log_core.* *libzephyr.a:log_backend_uart.* *libzephyr.a:log_output.* *libzephyr.a:loader.* *libdrivers__serial.a:uart_esp32.*) .rodata)
-    *(EXCLUDE_FILE (*libarch__xtensa__core.a:* *libkernel.a:fatal.* *libkernel.a:init.* *libzephyr.a:cbprintf_complete* *libzephyr.a:log_core.* *libzephyr.a:log_backend_uart.* *libzephyr.a:log_output.* *libzephyr.a:loader.* *libdrivers__serial.a:uart_esp32.*) .rodata.*)
+    *(EXCLUDE_FILE (*libsubsys__zbus.a:* *libarch__xtensa__core.a:* *libkernel.a:fatal.* *libkernel.a:init.* *libzephyr.a:cbprintf_complete* *libzephyr.a:log_core.* *libzephyr.a:log_backend_uart.* *libzephyr.a:log_output.* *libzephyr.a:loader.* *libdrivers__serial.a:uart_esp32.*) .rodata)
+    *(EXCLUDE_FILE (*libsubsys__zbus.a:* *libarch__xtensa__core.a:* *libkernel.a:fatal.* *libkernel.a:init.* *libzephyr.a:cbprintf_complete* *libzephyr.a:log_core.* *libzephyr.a:log_backend_uart.* *libzephyr.a:log_output.* *libzephyr.a:loader.* *libdrivers__serial.a:uart_esp32.*) .rodata.*)
 
     *(.irom1.text) /* catch stray ICACHE_RODATA_ATTR */
     *(.gnu.linkonce.r.*)
@@ -452,6 +452,7 @@ SECTIONS
     *libzephyr.a:loader.*(.rodata .rodata.*)
     *libdrivers__serial.a:uart_esp32.*(.rodata .rodata.*)
     *libdrivers__flash.a:flash_esp32.*(.rodata  .rodata.*)
+    *libsubsys__zbus.a:*(.rodata .rodata.*)
 
     KEEP(*(.jcr))
     *(.dram1 .dram1.*)
```

sylvioalves, 2024-01-30:
@Harakan Hi, you mean you get exceptions? Can you provide a bit more information and which modules/libs you have been using?

Harakan, 2024-01-30:
Thanks for the fast response @sylvioalves! I'm just pulling main and re-running some test cases.

The stripped down test case I have been working with on uses these modules/libs. 
 - esp32s2_fn4r2 soc
 - cellular_modem (BG95)
 - protobuf
 - ti,tca9538 driver for a i2c-gpio extender chip
 - mcuboot
 - CPP support
 
 Given logging/headers in specific spots the system works with mcuboot, but adding something in like Time or the NVS library seems to reliably break it
  - ie: adding headers and calling `gettimeofday(&tv, NULL);`
  - or initializing the nvs via a global `static struct nvs_fs fs;` then `nvs_mount(&fs);`
 
 Note: Using the example at `samples/subsys/zbus/hello_world` I can see the `ti,tca9538` being initialized in the logs, but the modem does not.

Harakan, 2024-01-30:
I just confirmed the bug is still there with latest commit (41f7294d0691ff754df106507a5794a79dc0454b). Including `sys/time.h`, and `time.h`, then adding a call to `gettimeofday(&tv, NULL);` was what broke it this time.

For posterity sake, here are my build steps. It might be worth noting I end up building the firmware a second time after initially flashing the bootloader for the first time. I did this so west flash grabs the right binary (otherwise west flash only flashes the bootloader):
1. `west build -p --sysbuild -b esp32s2_board .`
2. `west flash --esp-device /dev/ttyACM0`
3. `west build -p -b esp32s2_board`
4. `west flash --esp-device /dev/ttyACM0`

with `CONFIG_BOOTLOADER_MCUBOOT=y` in prj.conf:

```
[esp32s2] [INF] compiled at Jan 30 2024 14:18:47                                                                   
[esp32s2] [INF] Chip revision: 0                                                                                   
[esp32s2] [INF] SPI Flash speed: 40MHz, mode: DIO, size: 4MB
[esp32s2] [INF] Image index: 0, Swap type: none
[esp32s2] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32s2] [INF] Application start=4002DD50h
[esp32s2] [INF] DRAM segment: paddr=00016600h, vaddr=3FFCF8B0h, size=020D4h (  8404) load
[esp32s2] [INF] IRAM segment: paddr=00006DB0h, vaddr=40022000h, size=0F850h ( 63568) load
```

without `CONFIG_BOOTLOADER_MCUBOOT=y`

```
application boots and runs
```

sylvioalves, 2024-01-30:
Can you share a bit more of your `prj.conf`? Have you added posix API, picolibc.. ?
```c
CONFIG_REQUIRES_FULL_LIBC=y
CONFIG_POSIX_API=y
....
```

Harakan, 2024-01-30:
Thank you for explaining a better way to build and flash the bootloader.

A detail I forgot to bring up, is the mcuboot fails to build without adding `depends on MULTITHREADING` to two driver Kconfigs: `gpio/Kconfig.pca953x` and `i2c/Kconfig.esp32 `. 

Here is a test case I have currently running, and crashing with logs very similar to the 3rd case I initially outlined
ie: On boot error log: `[00:00:00.067,000] <err> net_l2_ppp: No LCP found!`

```
CONFIG_LED=y
CONFIG_PRINTK=y
CONFIG_INIT_STACKS=y
CONFIG_MAIN_STACK_SIZE=16384
CONFIG_POSIX_CLOCK=y
CONFIG_GPIO=y
CONFIG_WATCHDOG=y
CONFIG_WDT_DISABLE_AT_BOOT=y
CONFIG_CPP=y
CONFIG_REQUIRES_FULL_LIBCPP=y

# Modules:
CONFIG_NANOPB=y

# Logs:
CONFIG_LOG=y
CONFIG_LOG_DEFAULT_LEVEL=2
CONFIG_LOG_MAX_LEVEL=4
CONFIG_NET_LOG=y

CONFIG_PM_DEVICE_POWER_DOMAIN=n
CONFIG_PM_DEVICE=y
# CONFIG_PM is also broken

# Modem support:
CONFIG_MODEM=y
CONFIG_MODEM_CELLULAR=y
CONFIG_MODEM_CELLULAR_APN="hologram"
CONFIG_MODEM_CMUX=y
CONFIG_MODEM_PPP=y
CONFIG_MODEM_CONTEXT=y
CONFIG_MODEM_RECEIVER=y
CONFIG_MODEM_IFACE_UART=y
CONFIG_MODEM_CMD_HANDLER=y
CONFIG_MODEM_CELL_INFO=y
CONFIG_MODEM_SIM_NUMBERS=n
CONFIG_MODEM_CELLULAR_PERIODIC_SCRIPT_MS=30000

# networking support
CONFIG_NETWORKING=y
CONFIG_NET_NATIVE=y
CONFIG_NET_L2_PPP=y
CONFIG_NET_L2_PPP_TIMEOUT=10000
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=n
CONFIG_NET_UDP=y
CONFIG_NET_TCP=n
CONFIG_NET_SOCKETS=y
CONFIG_NET_CONNECTION_MANAGER=y
CONFIG_NET_CONFIG_AUTO_INIT=n

# DNS
CONFIG_DNS_RESOLVER=y
CONFIG_DNS_RESOLVER_MAX_SERVERS=1
CONFIG_DNS_SERVER_IP_ADDRESSES=y
CONFIG_DNS_SERVER1="8.8.8.8"

#Flash and Permanent Storage:
CONFIG_FLASH=y
CONFIG_FLASH_MAP=y
CONFIG_NVS=y

# MCUBOOT Bootloader:
CONFIG_BOOTLOADER_MCUBOOT=y
```

With the main function looking like this:

```
int main(void)
{
  printk("check happened\n");

  printk("Internal!\n");
  
  // Using the NVS seems to be fine:
  storage::Storage eeprom;
  eeprom.FsInit();
  eeprom.SyncFlash();
  printk("Storage!\n");

  // This time breaks it
  //time_t now = rtc::Now();
  //printk("time: %lu", now);

  // The network connect works if it's the only thing done
  int network_init = setupModemAndSocket();

  while (1) {
    k_sleep(K_MSEC(1000));
    printk("check happened\n");
  }
}
```

and rtc code like this:

```
#include "rtc.h"
#include <time.h>
#include <sys/time.h>
#include <zephyr/sys/printk.h>

time_t rtc::Now() {
  struct timeval tv; 
  int res = gettimeofday(&tv, NULL);
  time_t now = time(NULL);
  struct tm tm; 
  localtime_r(&now, &tm);

  if (res < 0) {
    printf("Error in rtc Now()\n");
    return 1;
  }   

  printk("gettimeofday(): HI(tv_sec)=%d, LO(tv_sec)=%d, "
         "tv_usec=%d\n\t%s\n", (unsigned int)(tv.tv_sec >> 32),
         (unsigned int)tv.tv_sec, (unsigned int)tv.tv_usec,
         asctime(&tm));

  return now;
```

Note: a lot of the code has been ported from a previous project, hense the slight namespace abuse

Harakan, 2024-01-30:
Small update, after removing all references to i2c and the pin extender, the application still crashed on boot:

```
[00:00:00.064,000] <err> net_l2_ppp: No LCP found!
*** Booting Zephyr OS build zephyr-v3.5.0-5033-gd7d254140d02 ***
[00:00:00.065,000] <err> os:  ** FATAL EXCEPTION
[00:00:00.065,000] <err> os:  ** CPU 0 EXCCAUSE 29 (store prohibited)
[00:00:00.065,000] <err> os:  **  PC 0x4001ad62 VADDR 0x400
[00:00:00.065,000] <err> os:  **  PS 0x60620
[00:00:00.065,000] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:6 CALLINC:2)
[00:00:00.065,000] <err> os:  **  A0 0x8002a643  SP 0x3ffc7560  A2 0x400  A3 0xaaaaaaaa
[00:00:00.065,000] <err> os:  **  A4 0x400971f0  A5 0x400  A6 0x400975f0  A7 0x400971f
[00:00:00.065,000] <err> os:  **  A8 0x8002a6ad  A9 0x3ffc7520 A10 0x60b20 A11 0x3ffcee68
[00:00:00.065,000] <err> os:  ** A12 0x3ffc2a10 A13 0x400996b4 A14 0 A15 0
[00:00:00.065,000] <err> os:  ** SAR 0x12
[00:00:00.065,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.065,000] <err> os: Current thread: 0x3ffbffc0 (unknown)
[00:00:00.076,000] <err> os: Halting system
```

oddly enough when looking on which line it failed on, the addr2line tool returns ??
```
$ addr2line -e build/zephyr/zephyr.elf 0x4001ad62
??:0

Checking that the tool still works on a random address:

$ addr2line -e build/zephyr/zephyr.elf 0x4002ad62  
/home/will/git/zephyr_workspace/zephyr/kernel/sched.c:1064
```

Harakan, 2024-01-31:
Update on the above, I'm still doing some testing but upgrading the zephyr-sdk to the latest rc1 seems to have fixed it!

Originally the project was setup with `sdk-0.16.3`, and when I tested with `sdk-0.16.4` I could still trigger the bug.

Now using the `sdk-0.16.5-rc1` release I have yet to make it fail, and all previous test cases now are working
- ref:  https://github.com/zephyrproject-rtos/sdk-ng/releases/tag/v0.16.5-rc1 

Thank you very much for helping, and I hope this helps other projects

dean-at-evos, 2024-02-01:
OK, so I think it is definitely alignment of the `rodata` section at link time. I have made the problem with the z-bus sample go away for both the c3 and s3, patches below. The trick is to use Z_VM_KERNEL to force the common.ld scripts to use the linker attribute ALIGN_WITH_INPUT on it's sections.

I would be interested if this approach works for your cases.

As an aside I wonder if all of the exclusion of files and including stuff into DRAM later is necessary ... see the c3 ** aggressive ** patch at the end. This still works. The notes around this about cache not working might be a furphy, the cache is enabled by the loader very early on. I wondered if this got in there as it ** fixed ** what was an alignment problem all along???

``` console
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd8000,len:0x1018
load:0x403cad98,len:0x3d8c
load:0x403d0000,len:0x1a24
entry 0x403cc724
[esp32c3] [INF] MCUboot 2nd stage bootloader
[esp32c3] [INF] compiled at Feb  1 2024 07:35:55
[esp32c3] [INF] Chip revision: 3
[esp32c3] [INF] SPI Flash speed: 40MHz, mode: DIO, size: 4MB
[esp32c3] [INF] Image index: 0, Swap type: none
[esp32c3] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32c3] [INF] Application start=403824E4h
[esp32c3] [INF] DRAM segment: paddr=00006160h, vaddr=3FC87570h, size=00374h (   884) load
[esp32c3] [INF] IRAM segment: paddr=00001888h, vaddr=40380000h, size=048D8h ( 18648) load
[esp32c3] [INF] DROM segment: paddr=00010040h, vaddr=3C000040h, size=01848h (  6216) map
[esp32c3] [INF] IROM segment: paddr=00020000h, vaddr=42020000h, size=045D0h ( 17872) map
I: Sensor sample started raw reading, version 0.1-2!
I: Channel list:
I: 0 - Channel acc_data_chan:
I:       Message size: 12
I:       Observers:
I:       - foo_lis
I:       - bar_sub
I: 1 - Channel simple_chan:
I:       Message size: 4
I:       Observers:
I: 2 - Channel version_chan:
I:       Message size: 4
I:       Observers:
I: Observers list:
I: 0 - Subscriber bar_sub
I: 1 - Listener foo_lis
I: From listener -> Acc x=1, y=1, z=1
I: From subscriber -> Acc x=1, y=1, z=1
I: From listener -> Acc x=2, y=2, z=2
I: From subscriber -> Acc x=2, y=2, z=2
I: Pub a valid value to a channel with validator successfully.
I: Pub an invalid value to a channel with validator successfully.
```

---

## C3

``` diff
diff --git a/soc/riscv/espressif_esp32/esp32c3/default.ld b/soc/riscv/espressif_esp32/esp32c3/default.ld
index 65984b4fd7..7e96c05b02 100644
--- a/soc/riscv/espressif_esp32/esp32c3/default.ld
+++ b/soc/riscv/espressif_esp32/esp32c3/default.ld
@@ -10,6 +10,8 @@
  * Linker script for the esp32c3 platform.
  */
 
+#define Z_VM_KERNEL 1
+
 #include <zephyr/devicetree.h>
 #include <zephyr/linker/sections.h>
 #include <zephyr/linker/linker-defs.h>
@@ -276,7 +278,7 @@ SECTIONS
[... 290 lines cut ...]
     . = ALIGN(0x10000) + 0x20;
-  } GROUP_LINK_IN(FLASH_CODE_REGION)
+  }  > FLASH_CODE_REGION
 
   .flash.text : ALIGN(IROM_SEG_ALIGN)
   {
@@ -454,7 +427,7 @@ SECTIONS
   .rtc.dummy (NOLOAD):
   {
     . = SIZEOF(.rtc.text);
-  } GROUP_LINK_IN(rtc_iram_seg)
+  } > rtc_iram_seg
 
   .rtc.data :
   {
@@ -471,7 +444,7 @@ SECTIONS
     *rtc_wake_stub*.o(.bss .bss.*)
     *rtc_wake_stub*.o(COMMON)
     _rtc_bss_end = ABSOLUTE(.);
-  } GROUP_LINK_IN(rtc_iram_seg)
+  }  > rtc_iram_seg
 
   /**
    * This section located in RTC SLOW Memory area.
```

dean-at-evos, 2024-02-01:
> Update on the above, I'm still doing some testing but upgrading the zephyr-sdk to the latest rc1 seems to have fixed it!
>
> Originally the project was setup with `sdk-0.16.3`, and when I tested with [...]

Huh ... I'll try this as well here. Thanks.

Harakan, 2024-02-01:
Hey @dean-at-evos I did get a lot more cases working with the latest sdk, but still some errors. I'll try adapting your linker patch to the s2 file. Really appreciate the help

Harakan, 2024-02-01:
No joy. I've never dug into linker scripts, so I might have done something silly here. It looks to me like the esp32s3 has more sections in the linker script than the esp32s2 does. Here is my best effort at adapting the S2's linker script to reflect closer to what the modified S3 is doing.

Note: on the following two sections neither one seems to make a difference if it's changed or not:

```diff
@@ -133,11 +135,15 @@ SECTIONS
    * Adding .rodata as first section helps to reduce size of generated binary by
    * few kBs.
    */
-  SECTION_PROLOGUE(_RODATA_SECTION_NAME,,ALIGN(4))
+  SECTION_PROLOGUE(_RODATA_SECTION_NAME,,ALIGN(0x10))
   {
     _rodata_reserved_start = ABSOLUTE(.);
-    __rodata_region_start = ABSOLUTE(.);

+    _rodata_start = ABSOLUTE(.);
+
+    *(.rodata_desc .rodata_desc.*)               /* Should be the first.  App version info.        DO NOT PUT ANYTHING BEFORE IT! */
+    *(.rodata_custom_desc .rodata_custom_desc.*) /* Should be the second. Custom app version info. DO NOT PUT ANYTHING BEFORE IT! */
```
and

```diff
@@ -194,7 +200,7 @@ SECTIONS
   SECTION_PROLOGUE(_RODATA_SECTION_END,,ALIGN(0x10))
   {
     _rodata_reserved_end = ABSOLUTE(.);
-    . = ALIGN(4);
+    . = ALIGN(16);
```

Finally here is the full diff of what I tried. Does the `mcuboot.ld` also need to be changed?

```diff
diff --git a/soc/xtensa/espressif_esp32/esp32s2/default.ld b/soc/xtensa/espressif_esp32/esp32s2/default.ld
index 8a78bb89ac2..14ab624cc9e 100644
--- a/soc/xtensa/espressif_esp32/esp32s2/default.ld
+++ b/soc/xtensa/espressif_esp32/esp32s2/default.ld
@@ -15,6 +15,8 @@
 #include <zephyr/linker/linker-defs.h>
 #include <zephyr/linker/linker-tool.h>
 
+#define Z_VM_KERNEL 1
+
 #define RAM_IRAM_START    0x40020000
 #define RAM_DRAM_START    0x3ffb0000
 
@@ -133,11 +135,15 @@ SECTIONS
    * Adding .rodata as first section helps to reduce size of generated binary by
    * few kBs.
    */
-  SECTION_PROLOGUE(_RODATA_SECTION_NAME,,ALIGN(4))
+  SECTION_PROLOGUE(_RODATA_SECTION_NAME,,ALIGN(0x10))
   {
     _rodata_reserved_start = ABSOLUTE(.);
-    __rodata_region_start = ABSOLUTE(.);
+    _rodata_start = ABSOLUTE(.);
+
+    *(.rodata_desc .rodata_desc.*)               /* Should be the first.  App version info.        DO NOT PUT ANYTHING BEFORE IT! */
+    *(.rodata_custom_desc .rodata_custom_desc.*) /* Should be the second. Custom app version info. DO NOT PUT ANYTHING BEFORE IT! */
 
+    __rodata_region_start = ABSOLUTE(.);
     . = ALIGN(4);
     #include <snippets-rodata.ld>
 
@@ -194,7 +200,7 @@ SECTIONS
   SECTION_PROLOGUE(_RODATA_SECTION_END,,ALIGN(0x10))
   {
     _rodata_reserved_end = ABSOLUTE(.);
-    . = ALIGN(4);
+    . = ALIGN(16);
     _image_rodata_end = ABSOLUTE(.);
   } GROUP_DATA_LINK_IN(RODATA_REGION, ROMABLE_REGION)
 
@@ -241,6 +247,9 @@ SECTIONS
        and uses it in preference to the first symbol in IRAM */
   } GROUP_DATA_LINK_IN(IRAM_REGION, ROMABLE_REGION)
 
+
+
+
   .iram0.text : ALIGN(4)
   {
     . = ALIGN(8);
@@ -301,7 +310,7 @@ SECTIONS
     */
     . = ALIGN (8);
     . = ORIGIN(dram0_0_seg) + _iram_end - _iram_start;
-  } GROUP_LINK_IN(RAMABLE_REGION)
+  } > RAMABLE_REGION
 
   /* Shared RAM */
   SECTION_DATA_PROLOGUE(_BSS_SECTION_NAME,(NOLOAD),)
@@ -326,7 +335,7 @@ SECTIONS
     *(COMMON)
     . = ALIGN (8);
     __bss_end = ABSOLUTE(.);
-  } GROUP_LINK_IN(RAMABLE_REGION)
+  } > RAMABLE_REGION
 
 #if defined(CONFIG_ESP_SPIRAM)
   .ext_ram.bss (NOLOAD):
@@ -358,8 +367,7 @@ SECTIONS
     *(.noinit)
     *(.noinit.*)
     . = ALIGN(8);
-  } GROUP_LINK_IN(RAMABLE_REGION)
-
+  } > RAMABLE_REGION
   .dram0.data :
   {
     . = ALIGN (8);
@@ -420,7 +428,7 @@ SECTIONS
   {
     . = ALIGN(4);
     _iram_end = ABSOLUTE(.);
-  } GROUP_LINK_IN(IRAM_REGION)
+  } > IRAM_REGION
 
   _image_irom_start = LOADADDR(.flash.text);
   _image_irom_size = LOADADDR(.flash.text) + SIZEOF(.flash.text) - _image_irom_start;
@@ -430,7 +438,7 @@ SECTIONS
   {
     . = ALIGN(4);
     . = SIZEOF(_RODATA_SECTION_NAME);
-  } GROUP_LINK_IN(FLASH_CODE_REGION)
+  } > FLASH_CODE_REGION
 
   .flash.text : ALIGN(IROM_SEG_ALIGN)
   {
@@ -492,7 +500,7 @@ SECTIONS
     *rtc_wake_stub*.o(.bss .bss.*)
     *rtc_wake_stub*.o(COMMON)
     _rtc_bss_end = ABSOLUTE(.);
-  } GROUP_LINK_IN(rtc_slow_seg)
+  } > rtc_slow_seg
 
   /* This section located in RTC SLOW Memory area.
      It holds data marked with RTC_SLOW_ATTR attribute.

```

Harakan, 2024-02-01:
Deleted previous comment, after finding a typo.

I can confirm the following linker script along with the new sdk is booting and running the full blown application code. I can't say I really understand what's going on, but am very happy to see this working. Thank you a tonne @dean-at-evos !

```diff
diff --git a/soc/xtensa/espressif_esp32/esp32s2/default.ld b/soc/xtensa/espressif_esp32/esp32s2/default.ld
index 8a78bb89ac2..25867f5cbab 100644
--- a/soc/xtensa/espressif_esp32/esp32s2/default.ld
+++ b/soc/xtensa/espressif_esp32/esp32s2/default.ld
@@ -10,11 +10,14 @@
  * Linker script for the esp32s2 platform.
  */
 
+#define Z_VM_KERNEL 1
+
 #include <zephyr/devicetree.h>
 #include <zephyr/linker/sections.h>
 #include <zephyr/linker/linker-defs.h>
 #include <zephyr/linker/linker-tool.h>
 
+
 #define RAM_IRAM_START    0x40020000
 #define RAM_DRAM_START    0x3ffb0000
 
@@ -133,7 +136,7 @@ SECTIONS
    * Adding .rodata as first section helps to reduce size of generated binary by
    * few kBs.
    */
-  SECTION_PROLOGUE(_RODATA_SECTION_NAME,,ALIGN(4))
+  SECTION_PROLOGUE(_RODATA_SECTION_NAME,,)
   {
     _rodata_reserved_start = ABSOLUTE(.);
     __rodata_region_start = ABSOLUTE(.);
@@ -191,7 +194,7 @@ SECTIONS
 
   /* Create an explicit section at the end of all the data that shall be mapped into drom.
    * This is used to calculate the size of the _image_drom_size variable */
-  SECTION_PROLOGUE(_RODATA_SECTION_END,,ALIGN(0x10))
+  SECTION_PROLOGUE(_RODATA_SECTION_END,,)
   {
     _rodata_reserved_end = ABSOLUTE(.);
     . = ALIGN(4);
@@ -301,7 +304,7 @@ SECTIONS
     */
     . = ALIGN (8);
     . = ORIGIN(dram0_0_seg) + _iram_end - _iram_start;
-  } GROUP_LINK_IN(RAMABLE_REGION)
+  } > RAMABLE_REGION
 
   /* Shared RAM */
   SECTION_DATA_PROLOGUE(_BSS_SECTION_NAME,(NOLOAD),)
@@ -326,7 +329,7 @@ SECTIONS
     *(COMMON)
     . = ALIGN (8);
     __bss_end = ABSOLUTE(.);
-  } GROUP_LINK_IN(RAMABLE_REGION)
+  } > RAMABLE_REGION
 
 #if defined(CONFIG_ESP_SPIRAM)
   .ext_ram.bss (NOLOAD):
@@ -358,7 +361,7 @@ SECTIONS
     *(.noinit)
     *(.noinit.*)
     . = ALIGN(8);
-  } GROUP_LINK_IN(RAMABLE_REGION)
+  } > RAMABLE_REGION
 
   .dram0.data :
   {
@@ -420,7 +423,7 @@ SECTIONS
   {
     . = ALIGN(4);
     _iram_end = ABSOLUTE(.);
-  } GROUP_LINK_IN(IRAM_REGION)
+  } > IRAM_REGION
 
   _image_irom_start = LOADADDR(.flash.text);
   _image_irom_size = LOADADDR(.flash.text) + SIZEOF(.flash.text) - _image_irom_start;
@@ -430,7 +433,7 @@ SECTIONS
   {
     . = ALIGN(4);
     . = SIZEOF(_RODATA_SECTION_NAME);
-  } GROUP_LINK_IN(FLASH_CODE_REGION)
+  } > FLASH_CODE_REGION
 
   .flash.text : ALIGN(IROM_SEG_ALIGN)
   {
@@ -492,7 +495,7 @@ SECTIONS
     *rtc_wake_stub*.o(.bss .bss.*)
     *rtc_wake_stub*.o(COMMON)
     _rtc_bss_end = ABSOLUTE(.);
-  } GROUP_LINK_IN(rtc_slow_seg)
+  } > rtc_slow_seg
 
   /* This section located in RTC SLOW Memory area.
      It holds data marked with RTC_SLOW_ATTR attribute.
```

dean-at-evos, 2024-02-01:
Hmmmm, yeah I am not convinced that that linker script is a robust as it could be. There are a few things in there that I don't think are necessary and it seems overly complex to me. But not being an ld guru and not being across the details of all the espressif socs I am somewhat flying by the pants seat, in the dark, without a net ....  @sylvioalves might be able to help out with some more insights

Can you post the output of `readelf -l build/zephyr/zephyr.elf` that might offer more clues.

Harakan, 2024-02-01:
Further changing the linker file (somewhat blind) to match the esp32s3 file seems working better but I can still find some counter-cases. Hard to say yet.

```diff
diff --git a/soc/xtensa/espressif_esp32/esp32s2/default.ld b/soc/xtensa/espressif_esp32/esp32s2/default.ld
index 8a78bb89ac2..953fa075f31 100644
--- a/soc/xtensa/espressif_esp32/esp32s2/default.ld
+++ b/soc/xtensa/espressif_esp32/esp32s2/default.ld
@@ -10,11 +10,14 @@
  * Linker script for the esp32s2 platform.
  */
 
+#define Z_VM_KERNEL 1
+
 #include <zephyr/devicetree.h>
 #include <zephyr/linker/sections.h>
 #include <zephyr/linker/linker-defs.h>
 #include <zephyr/linker/linker-tool.h>
 
+
 #define RAM_IRAM_START    0x40020000
 #define RAM_DRAM_START    0x3ffb0000
 
@@ -133,11 +136,15 @@ SECTIONS
    * Adding .rodata as first section helps to reduce size of generated binary by
    * few kBs.
    */
-  SECTION_PROLOGUE(_RODATA_SECTION_NAME,,ALIGN(4))
+  SECTION_PROLOGUE(_RODATA_SECTION_NAME,,)
   {
     _rodata_reserved_start = ABSOLUTE(.);
-    __rodata_region_start = ABSOLUTE(.);
+    _rodata_start = ABSOLUTE(.);
 
+    *(.rodata_desc .rodata_desc.*)               /* Should be the first.  App version info.        DO NOT PUT ANYTHING BEFORE IT! */
+    *(.rodata_custom_desc .rodata_custom_desc.*) /* Should be the second. Custom app version info. DO NOT PUT ANYTHING BEFORE IT! */
+
+    __rodata_region_start = ABSOLUTE(.);
     . = ALIGN(4);
     #include <snippets-rodata.ld>
 
@@ -191,10 +198,10 @@ SECTIONS
 
   /* Create an explicit section at the end of all the data that shall be mapped into drom.
    * This is used to calculate the size of the _image_drom_size variable */
-  SECTION_PROLOGUE(_RODATA_SECTION_END,,ALIGN(0x10))
+  SECTION_PROLOGUE(_RODATA_SECTION_END,,)
   {
     _rodata_reserved_end = ABSOLUTE(.);
-    . = ALIGN(4);
+    . = ALIGN(16);
     _image_rodata_end = ABSOLUTE(.);
   } GROUP_DATA_LINK_IN(RODATA_REGION, ROMABLE_REGION)
 
@@ -301,7 +308,7 @@ SECTIONS
     */
     . = ALIGN (8);
     . = ORIGIN(dram0_0_seg) + _iram_end - _iram_start;
-  } GROUP_LINK_IN(RAMABLE_REGION)
+  } > RAMABLE_REGION
 
   /* Shared RAM */
   SECTION_DATA_PROLOGUE(_BSS_SECTION_NAME,(NOLOAD),)
@@ -326,7 +333,7 @@ SECTIONS
     *(COMMON)
     . = ALIGN (8);
     __bss_end = ABSOLUTE(.);
-  } GROUP_LINK_IN(RAMABLE_REGION)
+  } > RAMABLE_REGION
 
 #if defined(CONFIG_ESP_SPIRAM)
[... 89 lines cut ...]

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  LOAD           0x000154 0x00000000 0x00000000 0x00020 0x00020 R   0x1
  LOAD           0x000174 0x00000020 0x00000020 0x00020 0x00020 RW  0x4
  LOAD           0x000194 0x3f000040 0x00000040 0x09644 0x09644 RW  0x4
  LOAD           0x0097d8 0x3f009688 0x00009684 0x00104 0x00104 RW  0x8
  LOAD           0x0098dc 0x40022000 0x00009788 0x0e4f0 0x0e4f0 R E 0x4
  LOAD           0x017dd0 0x3ffd3cf0 0x00017c78 0x02268 0x02268 RW  0x8
  LOAD           0x020000 0x40090000 0x00020000 0x1b240 0x1b240 R E 0x10000
  LOAD           0x000000 0x3ffb2000 0x3ffb2000 0x00000 0x21cf0 RW  0x10
  LOAD           0x000000 0x40080000 0x40080000 0x00000 0x092ec RW  0x10000

 Section to Segment mapping:
  Segment Sections...
   00     .mcuboot_header 
   01     .metadata 
   02     rodata ctors init_array initlevel device_area 
   03     _static_thread_data_area net_socket_register_area ppp_protocol_handler_area shell_area shell_root_cmds_area shell_subcmds_area 
   04     .iram0.vectors .iram0.text 
   05     .dram0.data sw_isr_table device_states pm_device_slots_area log_mpsc_pbuf_area log_msg_ptr_area log_dynamic_area k_mem_slab_area k_heap_area k_mutex_area k_msgq_area k_sem_area k_fifo_area net_buf_pool_area net_if_area net_if_dev_area net_l2_area log_const_area log_backend_area 
   06     .flash.text 
   07     .dram0.dummy bss noinit 
   08     .flash_text_dummy 
```

dean-at-evos, 2024-02-01:
OK, looks like the rodata is still misaligned. The indication there is the section to segment mapping, notice that segments 02 and 03 both contain stuff that should be in one segment, that is the rodata or the DROM segment. I just did this for the s2, I can't test it as I don't have the hw, but see how you go;

PS: No you don't need to touch the mcuboot.ld. That is only used during the linking of the bootloader build. It is much simpler as the bootloader just runs out of RAM.

``` diff
diff --git a/soc/xtensa/espressif_esp32/esp32s2/default.ld b/soc/xtensa/espressif_esp32/esp32s2/default.ld
index 8a78bb89ac..bbcfd7c889 100644
--- a/soc/xtensa/espressif_esp32/esp32s2/default.ld
+++ b/soc/xtensa/espressif_esp32/esp32s2/default.ld
@@ -10,6 +10,8 @@
  * Linker script for the esp32s2 platform.
  */
 
+#define Z_VM_KERNEL 1
+
 #include <zephyr/devicetree.h>
 #include <zephyr/linker/sections.h>
 #include <zephyr/linker/linker-defs.h>
@@ -133,7 +135,7 @@ SECTIONS
    * Adding .rodata as first section helps to reduce size of generated binary by
    * few kBs.
    */
-  SECTION_PROLOGUE(_RODATA_SECTION_NAME,,ALIGN(4))
+  SECTION_PROLOGUE(_RODATA_SECTION_NAME,,)
   {
     _rodata_reserved_start = ABSOLUTE(.);
     __rodata_region_start = ABSOLUTE(.);
@@ -191,7 +193,7 @@ SECTIONS
 
   /* Create an explicit section at the end of all the data that shall be mapped into drom.
    * This is used to calculate the size of the _image_drom_size variable */
-  SECTION_PROLOGUE(_RODATA_SECTION_END,,ALIGN(0x10))
+  SECTION_PROLOGUE(_RODATA_SECTION_END,,)
   {
     _rodata_reserved_end = ABSOLUTE(.);
     . = ALIGN(4);
@@ -301,7 +303,7 @@ SECTIONS
     */
     . = ALIGN (8);
     . = ORIGIN(dram0_0_seg) + _iram_end - _iram_start;
-  } GROUP_LINK_IN(RAMABLE_REGION)
+  } > RAMABLE_REGION
 
   /* Shared RAM */
   SECTION_DATA_PROLOGUE(_BSS_SECTION_NAME,(NOLOAD),)
@@ -326,7 +328,7 @@ SECTIONS
     *(COMMON)
     . = ALIGN (8);
     __bss_end = ABSOLUTE(.);
-  } GROUP_LINK_IN(RAMABLE_REGION)
+  } > RAMABLE_REGION
 
 #if defined(CONFIG_ESP_SPIRAM)
   .ext_ram.bss (NOLOAD):
@@ -358,7 +360,7 @@ SECTIONS
     *(.noinit)
     *(.noinit.*)
     . = ALIGN(8);
-  } GROUP_LINK_IN(RAMABLE_REGION)
+  } > RAMABLE_REGION
 
   .dram0.data :
   {
@@ -420,7 +422,7 @@ SECTIONS
   {
     . = ALIGN(4);
     _iram_end = ABSOLUTE(.);
-  } GROUP_LINK_IN(IRAM_REGION)
+  } > IRAM_REGION
 
   _image_irom_start = LOADADDR(.flash.text);
   _image_irom_size = LOADADDR(.flash.text) + SIZEOF(.flash.text) - _image_irom_start;
@@ -430,7 +432,7 @@ SECTIONS
   {
     . = ALIGN(4);
     . = SIZEOF(_RODATA_SECTION_NAME);
-  } GROUP_LINK_IN(FLASH_CODE_REGION)
+  } > FLASH_CODE_REGION
 
   .flash.text : ALIGN(IROM_SEG_ALIGN)
   {
@@ -492,7 +494,7 @@ SECTIONS
     *rtc_wake_stub*.o(.bss .bss.*)
     *rtc_wake_stub*.o(COMMON)
     _rtc_bss_end = ABSOLUTE(.);
-  } GROUP_LINK_IN(rtc_slow_seg)
+  } > rtc_slow_seg
 
   /* This section located in RTC SLOW Memory area.
      It holds data marked with RTC_SLOW_ATTR attribute.
```

And the sec/seg mapping from read elf looks like this when I build the z-bus sample I have been using as a test.
`west build -p -b esp32s2_saola samples/subsys/zbus/hello_world/ -- -DCONFIG_BOOTLOADER_MCUBOOT=y -DCONFIG_COMPILER_SAVE_TEMPS=y -DCONFIG_DEBUG_OPTIMIZATIONS=y`

``` console
(.venv) deano@deano:~/zephyrproject/zephyr$ readelf -l build/zephyr/zephyr.elf 

Elf file type is EXEC (Executable file)
Entry point 0x40024adc
There are 8 program headers, starting at offset 52

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  LOAD           0x000134 0x00000000 0x00000000 0x00020 0x00020 R   0x1
  LOAD           0x000154 0x00000020 0x00000020 0x00020 0x00020 RW  0x4
  LOAD           0x000178 0x3f000040 0x00000040 0x019b0 0x019b0 RW  0x8
  LOAD           0x001b28 0x40022000 0x000019f0 0x04688 0x04688 R E 0x4
  LOAD           0x0061b0 0x3ffb9580 0x00006078 0x00988 0x00988 RW  0x4
  LOAD           0x010000 0x40090000 0x00010000 0x05130 0x05130 R E 0x10000
  LOAD           0x000000 0x3ffb2000 0x3ffb2000 0x00000 0x07580 RW  0x10
  LOAD           0x000000 0x40080000 0x40080000 0x00000 0x01854 RW  0x10000

 Section to Segment mapping:
  Segment Sections...
   00     .mcuboot_header 
   01     .metadata 
   02     rodata initlevel device_area _static_thread_data_area zbus_channel_area zbus_observer_area zbus_channel_observation_area 
   03     .iram0.vectors .iram0.text 
   04     .dram0.data sw_isr_table device_states k_heap_area k_msgq_area zbus_channel_observation_mask_area log_const_area 
   05     .flash.text 
   06     .dram0.dummy bss noinit 
   07     .flash_text_dummy
```

Harakan, 2024-02-01:
I'll take a couple days to flush out some solid test cases here, and firm up what works best. I think this gives me enough to work around the problem, so that we can take time to find the 'correct' fix (whatever that ends up being). I'll probably end up building a bunch of sample projects and seeing which ones work or not work then adjusting logging levels, and shell features until it breaks. Watching the readelf and counting segments is a nice shortcut. My understanding here is for a specific soc, the number of segments should always be the same? ie: I should always see 7 segments for zephyr applications compiled for the esp32s2?

Finally thanks for the tool examples as well. Even the simple stuff like: `-DCONFIG_COMPILER_SAVE_TEMPS=y -CONFIG_DEBUG_OPTIMIZATIONS=y` I never thought to look for.

dean-at-evos, 2024-02-02:
@Harakan Sounds good. I am unsure if what I have done will make it upstream, although I am confident it solves the problem that we were seeing with our application. Both the minimal test case sample I was using and our application now run without seg faults. I got more confidence that the same solution seemed to work across multiple chipsets. I will probably submit a PR with those changes to get the conversation with the gurus started so something gets pushed upstream.

To really put yourself in a coma `include/zephyr/linker` contains that linking boiler plate and the output of expanding the linker script ends up in `build/zephyr/linker.cmd` this is the file used by `ld` when the application is linked.

To answer your questions, from my understanding after following the load process from boot.

There are six *segments* that contain data and should end up in the flash binary that goes to the device. The 2nd stage bootloader (mcuboot in this case) uses the first two segments to tell it what to do. The first 32 bytes are the bootloader's header, this contains magic stuff like is the image confirmed, signed, slot index, etc...

The next 32 bytes "metadata" contain the addresses for the next four segments that contain the program instructions and data. The loader needs these to setup the SOC for the application.

The "rodata" or DROM segment is mapped to the MMU starting at it's Vaddr. So when the processor wants the data stored at 0x3f000040 up it will look for it in the flash at offset 0x000178+ (sort of, it caches pages of this data in RAM, but...)
The .flash_text or IROM segment is similarly mapped at Vaddr 0x40090000 up. Again the MMU knows where to look in flash for instructions as needed.

The IRAM and DRAM sections are copied from flash (using memset) into RAM at their Vaddr. This is the processors working memory, initialised from flash. The IRAM provides a place for 'rapid access' of instructions, bypassing the cache. That is why stuff like the kernel code ends up there, it's for efficiency. Another complication of these SOC is that the physical memory pointed at by DRAM and IRAM vaddr is the same. That is, address IRAM+123 is **physically** the same location as DRAM+123 although their addresses suggest they are pointing at different spots. So the linker script marks the end of IRAM and starts the DRAM vaddr section after this, otherwise the DRAM/IRAM would end up corrupted. The sections like 'dummy' put these *marks*/*offsets* in.

The dram .bss, .noinit sections don't contain any data (see the length of this segment is 0) but it is a marker for RAM that is allocated for use, but not initialised (the loader fills it with 0)

SO, the problem we were seeing was that for some reason for the DROM segment was being split which broke it's alignment (or the elf file segment was not ending up in the binary properly ... not sure exactly) then when the data that was expected to be at addr xyz was accessed, it wasn't there. The big clue that it was alignment/segment layout was that moving a library at link time made the problem disappear. Also while trying to debug I wanted to see the value of a variable and just that reference added a string to the DROM, and problem gone. That ends my knowledge of the system really, I am not linker guru .... my fix was to use a define that tells the zephyr provided linker scripts that we were on a vaddr/laddr mapped system and their scripts do the heavy lifting.

closed 2024-02-02 by dean-at-evos

naNEQ, 2024-03-01:
@dean-at-evos Just wanted to say that the changes proposed here have also solved the issue I've been facing with my code, as described in #64297.

dean-at-evos, 2024-03-01:
> Just wanted to say that the changes proposed here have also solved the issue I've been facing with my code, as described in #64297.

I did close this, not because the issue is solved but as I have PR open on it. Just to confirm the PR #68583 solved your issues?

naNEQ, 2024-03-04:
> > Just wanted to say that the changes proposed here have also solved the issue I've been facing with my code, as described in #64297.
> 
> I did close this, not because the issue is solved but as I have PR open on it. Just to confirm the PR #68583 solved your issues?

Yes, I am aware of that PR. The thing is, I was a bit lost in the discussions and was not entirely sure where the #68583 PR is based on. I had the impression it's based on Zephyr v3.6 (perhaps I'm wrong though) and since I am still using v3.5, I decided to try it with the proposed changes in this issue. So with the `#define Z_VM_KERNEL 1`.

dean-at-evos, 2024-03-04:
It is a bit of a mess in a sense. I closed this issue as the 'suspect cache corruption' was nowhere near the cause of the issue.
The issue is that the linking is inconsistent ... so I moved the discussion to that PR.
All versions of Zephyr until this is resolved will show the same thing with mcuboot in tree as a 2nd stage bootloader (I haven't tried the out of tree mcuboot esp32 build for a while). The linker scripts aren't suitable IMHO.