zephyrproject-rtos/zephyr #68039: sensor: as6212: Wrong register values upon entering sleep mode

nguyenmthien-gin, 2024-01-24:
**Describe the bug**
For the code that puts the AS6212 to sleep mode, [the driver appears to write all configuration bits to 0](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/sensor/tmp108/tmp108.c#L267).

This contradicts with the usage stated in section 6.2.4 of [the datasheet](https://ams.com/documents/20143/36005/AS621x_DS000677_4-00.pdf).

**Expected behavior**
One of the three outcomes is expected from the driver:
- The driver returns `-ENOTSUP` when requesting TMP108 attributes 
- The driver sets up the register values correctly with TMP108 attributes 
- The driver supports power management

**Impact**
This sensor will not enter sleep mode upon requested.

henrikbrixandersen, 2024-01-24:
@nguyenmthien-gin Thank you for reporting this. Could you send a PR with a proposed fix for this issue?

nguyenmthien-gin, 2024-01-24:
Hey, I need the sensor maintainer to make a decision on how we want to implement the fix. :smiley: 

> @nguyenmthien-gin Thank you for reporting this. Could you send a PR with a proposed fix for this issue?

closed 2024-04-08 by github-actions[bot]