zephyrproject-rtos/zephyr #68062: intel_adsp_cavs25: tests/subsys/logging/log_output/logging.output.thread_id timeout.

gbarkadiusz, 2024-01-24:
**Describe the bug**
`The tests/subsys/logging/log_output/logging.output.thread_id fails.`
```
INFO    - 2446/2454 intel_adsp_cavs25         tests/subsys/logging/log_output/logging.output.thread_id  FAILED Timeout (device: tgl, 60.600s)
INFO    - <path_to>/twister-out/intel_adsp_cavs25/tests/subsys/logging/log_output/logging.output.thread_id/handler.log
```

**To Reproduce**
Steps to reproduce the behavior:
```
 scripts/twister -W -p intel_adsp_cavs25 --device-testing --hardware-map <path_to>/cavs/cavs.map -T tests/subsys/logging/log_output/logging.output.thread_id -v
```

**Logs and console output**
`<path_to>/twister-out/intel_adsp_cavs25/tests/subsys/logging/log_output/logging.output.thread_id/handler.log`
```
INFO:cavs-client:REMOTE HOST: <host_ip> PORT: <port>
INFO:cavs-client:Monitor process
INFO:cavs-client:Sent:     start_log
INFO:cavs-client:Receive: start_log
INFO:cavs-client:Start to monitor log output...
```

**Environment**
 - OS: Linux
 - Commit SHA or Version used: zephyr-v3.5.0-4625-g114548ef607a

nashif, 2024-01-24:
Cant reproduce, your logs are not showing the test failing or loading, could be a FW loading issue on your device?

```
DEBUG   - DEVICE: INFO:cavs-client:Start to monitor log output...
DEBUG   - -- west flash: using runner intel_adsp
-- runners.intel_adsp: Starting Intel ADSP runner

DEBUG   - DEVICE: --
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v3.5.0-4677-g4ce0555d90a2 ***
DEBUG   - DEVICE: Running TESTSUITE test_log_output
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_colors
DEBUG   - DEVICE: PASS - test_colors in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_format_ts
DEBUG   - DEVICE: [00:00:01.000,000] domain/src: test
DEBUG   - DEVICE: PASS - test_format_ts in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_level_flag
DEBUG   - DEVICE: PASS - test_level_flag in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_levels
DEBUG   - DEVICE: PASS - test_levels in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_no_flags
DEBUG   - DEVICE: PASS - test_no_flags in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_no_flags_dname
DEBUG   - DEVICE: PASS - test_no_flags_dname in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_raw
DEBUG   - DEVICE: PASS - test_raw in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_skip_src
DEBUG   - DEVICE: PASS - test_skip_src in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_thread_id
DEBUG   - DEVICE: <err> [test_thread_id] src: Test
DEBUG   - DEVICE: PASS - test_thread_id in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_ts_flag
DEBUG   - DEVICE: PASS - test_ts_flag in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_ts_to_us
DEBUG   - DEVICE: PASS - test_ts_to_us in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: TESTSUITE test_log_output succeeded
DEBUG   - DEVICE:
DEBUG   - DEVICE: ------ TESTSUITE SUMMARY START ------
DEBUG   - DEVICE:
DEBUG   - DEVICE: SUITE PASS - 100.00% [test_log_output]: pass = 11, fail = 0, skip = 0, total = 11 duration = 0.011 seconds
DEBUG   - DEVICE: - PASS - [test_log_output.test_colors] duration = 0.001 seconds
DEBUG   - DEVICE: - PASS - [test_log_output.test_format_ts] duration = 0.001 seconds
DEBUG   - DEVICE: - PASS - [test_log_output.test_level_flag] duration = 0.001 seconds
DEBUG   - DEVICE: - PASS - [test_log_output.test_levels] duration = 0.001 seconds
DEBUG   - DEVICE: - PASS - [test_log_output.test_no_flags] duration = 0.001 seconds
DEBUG   - DEVICE: - PASS - [test_log_output.test_no_flags_dname] duration = 0.001 seconds
DEBUG   - DEVICE: - PASS - [test_log_output.test_raw] duration = 0.001 seconds
DEBUG   - DEVICE: - PASS - [test_log_output.test_skip_src] duration = 0.001 seconds
DEBUG   - DEVICE: - PASS - [test_log_output.test_thread_id] duration = 0.001 seconds
DEBUG   - DEVICE: - PASS - [test_log_output.test_ts_flag] duration = 0.001 seconds
DEBUG   - DEVICE: - PASS - [test_log_output.test_ts_to_us] duration = 0.001 seconds
DEBUG   - DEVICE:
DEBUG   - DEVICE: ------ TESTSUITE SUMMARY END ------
DEBUG   - DEVICE:
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: RunID: 4bb3099e82758b9d771ddafd69c4fec2
DEBUG   - DEVICE: PROJECT EXECUTION SUCCESSFUL
DEBUG   - Process soc/xtensa/intel_adsp/tools/cavstool_client.py,-s,zephyr@admin-fmos.hf.intel.com:59142,-l terminated outs: None errs None
DEBUG   - Expected suite names:['test_log_output', 'test_log_output']
DEBUG   - Detected suite names:['test_log_output']
DEBUG   - run status: intel_adsp_cavs25/tests/subsys/logging/log_output/logging.output.thread_id passed
INFO    - 1/1 intel_adsp_cavs25         tests/subsys/logging/log_output/logging.output.thread_id PASSED (device: tgl2, 2.996s)

```

closed 2024-01-24 by nashif

gbarkadiusz, 2024-01-25:
HW issue, cannot reproduce. Thank you for verification.