zephyrproject-rtos/zephyr #68065: ambiq timer MPU build error

teburd, 2024-01-24:
CI is failing for me when it attempts to build the ambiq_timer driver with what appears to be an unintentional name conflict

CI results: https://github.com/zephyrproject-rtos/zephyr/actions/runs/7633215966/job/20795102064?pr=67857#step:13:1095

```
FAILED: zephyr/drivers/timer/CMakeFiles/drivers__timer.dir/ambiq_stimer.c.obj 
ccache /opt/toolchains/zephyr-sdk-0.16.4/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -DTC_RUNID=326389d8245780771a6157cdaffaee3f -D_POSIX_C_SOURCE=200809 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I/__w/zephyr/zephyr/include -I/__w/zephyr/zephyr/twister-out/apollo4p_blue_kxr_evb/tests/subsys/llext/llext.simple.arm_mpu/zephyr/include/generated -I/__w/zephyr/zephyr/soc/arm/ambiq/apollo4x -I/__w/zephyr/zephyr/soc/arm/common/cortex_m -I/__w/zephyr/zephyr/soc/arm/ambiq/. -I/__w/zephyr/zephyr/subsys/testsuite/include -I/__w/zephyr/zephyr/subsys/testsuite/ztest/include -I/__w/zephyr/modules/hal/cmsis/CMSIS/Core/Include -I/__w/zephyr/zephyr/modules/cmsis/. -I/__w/zephyr/modules/hal/ambiq/CMSIS/AmbiqMicro/Include -I/__w/zephyr/modules/hal/ambiq/utils -I/__w/zephyr/modules/hal/ambiq/mcu/apollo4p -I/__w/zephyr/modules/hal/ambiq/mcu/apollo4p/hal -I/__w/zephyr/modules/hal/ambiq/components/bluetooth -isystem /__w/zephyr/zephyr/lib/libc/common/include -Wshadow -fno-strict-aliasing -Werror -Os -imacros /__w/zephyr/zephyr/twister-out/apollo4p_blue_kxr_evb/tests/subsys/llext/llext.simple.arm_mpu/zephyr/include/generated/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft --sysroot=/opt/toolchains/zephyr-sdk-0.16.4/arm-zephyr-eabi/arm-zephyr-eabi -imacros /__w/zephyr/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/__w/zephyr/zephyr/tests/subsys/llext=CMAKE_SOURCE_DIR -fmacro-prefix-map=/__w/zephyr/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/__w/zephyr=WEST_TOPDIR -ffunction-sections -fdata-sections --specs=picolibc.specs -std=c99 -MD -MT zephyr/drivers/timer/CMakeFiles/drivers__timer.dir/ambiq_stimer.c.obj -MF zephyr/drivers/timer/CMakeFiles/drivers__timer.dir/ambiq_stimer.c.obj.d -o zephyr/drivers/timer/CMakeFiles/drivers__timer.dir/ambiq_stimer.c.obj -c /__w/zephyr/zephyr/drivers/timer/ambiq_stimer.c
In file included from /__w/zephyr/zephyr/include/zephyr/arch/arm/mpu/arm_mpu.h:14,
                 from /__w/zephyr/zephyr/include/zephyr/arch/arm/arch.h:271,
                 from /__w/zephyr/zephyr/include/zephyr/arch/cpu.h:19,
                 from /__w/zephyr/zephyr/include/zephyr/kernel_includes.h:37,
                 from /__w/zephyr/zephyr/include/zephyr/kernel.h:17,
                 from /__w/zephyr/zephyr/drivers/timer/ambiq_stimer.c:16:
/__w/zephyr/zephyr/include/zephyr/arch/arm/mpu/arm_mpu_v7m.h:18:25: error: expected identifier before numeric constant
   18 | #define NO_ACCESS       0x0
      |                         ^~~
/__w/zephyr/modules/hal/ambiq/mcu/apollo4p/hal/mcu/am_hal_mpu.h:93:5: note: in expansion of macro 'NO_ACCESS'
   93 |     NO_ACCESS       = ARM_MPU_AP_NONE,
```

fkokosinski, 2024-01-25:
Hi @teburd, thanks for reporting this issue.

I wasn't able to reproduce this issue on the current mainline Zephyr. Could you provide me with the minimal steps/instructions to reproduce it? Thanks

teburd, 2024-01-25:
> Hi @teburd, thanks for reporting this issue.
> 
> I wasn't able to reproduce this issue on the current mainline Zephyr. Could you provide me with the minimal steps/instructions to reproduce it? Thanks

It may very well have been a test build configuration issue and no longer seems to be a problem, lets ignore it as you weren't able to repro.

closed 2024-01-25 by teburd