zephyrproject-rtos/zephyr #68070: Bluetooth: Host: missing api: bt_uatt_get_mtu

alwa-nordic, 2024-01-24:
As a Bluetooth LE Audio subsystem developer, I need my API to enforce a requirement from the specification: Before I allow some functions, the UATT MTU must be at least 64. I must return an error otherwise.

Currently, there is no way to get the UATT MTU (in a reliable way) from the Host.

**Describe the solution you'd like**
A new API: `uint16_t bt_uatt_get_mtu(struct bt_conn *conn)`

**Describe alternatives you've considered**
Not enforcing the ATT MTU at service level. This could cause hard-to-debug errors for the user.

**Why the current API does not satisfy my requirements**
 - `bt_gatt_get_mtu`: This affected by EATT MTU. I need the UATT MTU.
 - `bt_gatt_exchange_mtu`: The service layer is not the place to initiate the MTU exchange. That is done by the sample/application.
 - `bt_gatt_cb.att_mtu_updated`: This affected by EATT MTU. I need the UATT MTU.

alwa-nordic, 2024-01-24:
Relates to: https://github.com/zephyrproject-rtos/zephyr/issues/64172

ubieda, 2024-05-08:
@alwa-nordic is this issue up for grabs? I could tackle it as it would enable me coming up with a solution for #72018.

ubieda, 2024-08-06:
Sorry for the delay. Picking up this now.

PR #76808 (2024-08-15): bluetooth: host: Add APIs to track Unenhanced ATT MTU
### Description
This PR features an additional API to get the current UATT MTU and a callback on UATT MTU changes. This in order to provide the ability to keep track of this parameter, regardless of the presence of additional EATT channels.

Fixes #68070.

### Changes
- Add `uatt_mtu_updated` to get notified on UATT MTU changes on a given connection.
- Add `bt_gatt_get_uatt_mtu` to get a connection's current UATT MTU.
- Update MTU Update sample to use the above.
- Refactor Babblesim test: MTU Update to have peripheral and central be two sep [... cut]