zephyrproject-rtos/zephyr #68075: doc: Documentation issue in 'kconfig'

PBA12695, 2024-01-24:
**Describe the bug**
CONFIG_BT_LONG_WQ_STACK_SIZE cannot be found in the documentation

**Environment**

* Page: `kconfig`
* Version: main
* SHA-1: fc942ef7d30dea3a5b4e0d9644cc9f586385ecf0

kartben, 2024-01-24:
It's a hidden Kconfig so AFAICT this is intended behavior https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/host/Kconfig#L15-L16
@jori-nordic @nordicjm

closed 2024-01-24 by kartben

PBA12695, 2024-01-25:
Wouldn't it be good to document invisible symbols as well? In some circumstances, you need to modify them.

kartben, 2024-01-25:
> Wouldn't it be good to document invisible symbols as well? In some circumstances, you need to modify them.

Probably very related to #67582