zephyrproject-rtos/zephyr #68095: pytest/twister: reset instead of reflash

M1cha, 2024-01-25:
**Is your enhancement proposal related to a problem? Please describe.**
I know there's `pytest_dut_scope`, but even if you don't use that, I don't understand why the firmware needs to be reflashed for every single test case.

**Describe the solution you'd like**
Flash the firmware once and simply hardware-reset the device for every test case.

gchwier, 2024-01-26:
The idea was to have the same state of the board before every test, so the easiest method was to flash the board before executing each test. Pytest does not ensure that testcases are run in the same order (unless you use some extra extensions). 

`pytest_dut_scope` was added to flash the DUT only once per scope. As you mentioned, you can use it, if you are certain that the state of the board does not depend on other tests, or if it does not matter for the current scenario.

Currently, there is no method in current DUT abstract to reset the device. You can add a new fixture to your project or just a simple method to reset device.
e.g.
`subprocess.check_output(f'nrfjprog --snr {dut.device_config.id} --reset', shell=True)`

M1cha, 2024-01-26:
@gchwier If `The idea was to have the same state of the board before every test` is the goal, then there is a bug. Currently, the tests only flash the new zephyr binary without erasing the whole flash, so stuff like the storage partition will stay intact.

PerMac, 2024-01-26:
@M1cha I had a chat with @gchwier and we don't see how we could help more with your request. We are also not sure how feasible functionality of "reset" would be, which seems you have already noticed: "with some implementations like binary(native_posix) it's not that easy to do that "from the outside" because they launch a process and process it's stdout."

We don't see a point of "reseting" qemu/native, which wouldn't be just restarting the binary ("reflashing"). Native_posix (native_sim) doesn't even have a functionality of reset (consulted with @aescolar). 

As for HW: Flashing is supported in twister (and pytest) as a method which builds a custom command based on the runner used (mrfjprog, pyocd, etc. ) https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/pylib/twister/twisterlib/handlers.py#L480. it is because there is no common interface for flashing. West does it in similar way (e.g. for nrfjprog https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/west_commands/runners/nrfjprog.py). There is no dedicated method for reset in twister nor in west. 
In our opinion, pytest extension to twister is definitely not a place, where such case-specific (low-level) implementation should take place. We wouldn't even be able to implement it into twister nor west, since we don't have all those types of boards to tests different runners. Maybe you can ask community, if there is an option to implement sth like "west reset" to west_commands.

Hence I back gchwier's suggestion. Try using `pytest_dut_scope` to block reflashing for each test. Then you can add a fixture "reset". It can detect the type of dut by looking at its class type, if it's binary or hardware adapter. If it is binary, call flash_and_run method. If it is hardware, it will call your runner with appropriate reset command (since there are even different types of reset and those are not generic to every runner).

PerMac, 2024-01-26:
>Currently, the tests only flash the new zephyr binary without erasing the whole flash, so stuff like the storage partition will stay intact.

You're right, it was a mental shortcut. By default runners don't erase the full memory before flashing (since in most cases it is not needed) and twister follows this workflow. You can use west-flash="erase" if you need full erase.

M1cha, 2024-01-26:
@PerMac Thanks a lot for taking a look at this. You clarified the goal of the current design and the current implementation reflects those, so there are no bugs.

Since resetting instead of erasing is not what we seem to want to do by default, there's also need for a change in that regard. While a reset function in the Device class would be nice, I have to agree, that it might be the wrong place to put those, and I'm also currently fine with using pytest_dut_scope so I don't plan to implement anything myself.

TL;DR: Assuming that nobody wants to implement any way to reset the device, this can be cloced.

closed 2024-01-26 by M1cha