zephyrproject-rtos/zephyr #68097: intel_adsp_cavs25: tests/drivers/dma/loop_transfer/drivers.dma.loop_transfer FAILED Timeout issue

arikgreen, 2024-01-25:
**Describe the bug**
`The tests/drivers/dma/loop_transfer/drivers.dma.loop_transfer fails.`
```
INFO    - <path_to>/twister-out/intel_adsp_cavs25/tests/drivers/dma/loop_transfer/drivers.dma.loop_transfer/handler.log
INFO    - 2444/2456 intel_adsp_cavs25         tests/kernel/threads/dynamic_thread_stack/kernel.threads.dynamic_thread.stack.pool.alloc.user  FAILED Timeout (device: tgl, 60.502s)
INFO    - <path_to>/twister-out/intel_adsp_cavs25/tests/kernel/threads/dynamic_thread_stack/kernel.threads.dynamic_thread.stack.pool.alloc.user/handler.log
```
 
 
**To Reproduce**
Steps to reproduce the behavior:
```
scripts/twister -W -p intel_adsp_cavs25 --device-testing --hardware-map <path_to>/cavs/cavs.map -T tests/drivers/dma/loop_transfer/drivers.dma.loop_transfer -v
```
 
 
**Logs and console output**
`<path_to>/twister-out/intel_adsp_cavs25/tests/drivers/dma/loop_transfer/drivers.dma.loop_transfer/handler.log`
```
ERROR   - INFO:cavs-client:REMOTE HOST: host_name PORT: host_port
INFO:cavs-client:Monitor process
INFO:cavs-client:Sent:     start_log
INFO:cavs-client:Receive: start_log
INFO:cavs-client:Start to monitor log output...
--
```
 
**Environment:**
- OS: Linux
- Commit SHA or Version used: zephyr-v3.5.0-4691-gacfb87dca3e4

nashif, 2024-01-25:
seems to be a DUT issue rathert than a problem with the test.

closed 2024-01-25 by nashif

arikgreen, 2024-01-26:
yes, confirmed