zephyrproject-rtos/zephyr #68103: Display: STM32: display in not working on stm32f429i_disc1

ajarmouni-st, 2024-01-25:
**Describe the bug**
The drivers/display sample gives random noise on the screen. LVGL samples, modules/lvgl/demos & subsys/display/lvgl, give distorted frames.

 - What target platform are you using? stm32f429i_disc1 board
 - What have you tried to diagnose or workaround this issue? Found a fix to ili9341 driver that must be applied on top of PR #66271.
   PR #66271 alone gives a white screen.
 - Is this a regression? No info

**To Reproduce**
Steps to reproduce the behavior:
1. cd ~/zephyrproject/zephyr
2. west build -p always -b stm32f429i_disc1 samples/drivers/display
3. west flash

**Expected behavior**
Correct display of frames.

**Impact**
showstopper

**Logs and console output**
N/A

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: zephyr-sdk-0.16.3
 - Commit SHA or Version used: main, 571b647825b0402b1b46b4347b1db508b14cdf88

**Additional context**
This issue needs to be solved to be able to test PR #64587 on stm32f429i_disc1.

ajarmouni-st, 2024-01-29:
@fabiobaltieri PR #66271 alone does not fix this issue, PR #68105 is needed as well. Will reopen.

fabiobaltieri, 2024-01-29:
> @fabiobaltieri PR #66271 alone does not fix this issue, PR #68105 is needed as well. Will reopen.

Sure, fyi github closed this automatically when the PR got merged, if you don't want a PR to close an issue make sure remove the reference from under "Development" in the PR page.

closed 2024-02-02 by carlescufi