zephyrproject-rtos/zephyr #68115: up_squared: tests/kernel - many tests fail

KamilxPaszkiet, 2024-01-25:
**Describe the bug**
list of failing tests:
```
tests/kernel/mem_protect/protection/kernel.memory_protection.protection
tests/kernel/fifo/fifo_api/kernel.fifo
tests/kernel/semaphore/sys_sem/kernel.semaphore.usage
tests/kernel/semaphore/semaphore/kernel.semaphore
tests/kernel/spinlock/kernel.multiprocessing.spinlock_fairness
tests/kernel/spinlock/kernel.multiprocessing.spinlock
tests/kernel/workq/work/kernel.workqueue.api
tests/kernel/spinlock/kernel.multiprocessing.spinlock.minimallibc
tests/kernel/profiling/profiling_api/kernel.common.profiling
tests/kernel/threads/thread_stack/kernel.threads.thread_stack
tests/kernel/obj_core/obj_core_stats/kernel.obj_core.stats
tests/kernel/obj_core/obj_core_stats_api/kernel.obj_core.stats.api
tests/kernel/mbox/mbox_usage/kernel.mailbox.usage
tests/kernel/events/sys_event/kernel.events.usage
tests/kernel/events/event_api/kernel.events
tests/kernel/obj_core/obj_core/kernel.obj_core
tests/kernel/lifo/lifo_usage/kernel.lifo.usage
tests/kernel/msgq/msgq_api/kernel.message_queue
tests/kernel/common/kernel.common.nano64
tests/kernel/common/kernel.common.minimallibc
tests/kernel/common/kernel.common.misra
tests/kernel/common/kernel.common
tests/kernel/common/kernel.common.tls
tests/kernel/common/kernel.common.nano32
tests/kernel/common/kernel.common.picolibc
tests/kernel/queue/kernel.queue
tests/kernel/lifo/lifo_api/kernel.lifo
tests/kernel/queue/kernel.queue.minimallibc
```
most common errors:
```
ZEPHYR FATAL ERROR 0: CPU exception on CPU 1
ZEPHYR FATAL ERROR 4: Kernel panic on CPU 1
ZEPHYR FATAL ERROR 2: Stack overflow on CPU 1
```

**To Reproduce**
`twister --device-testing --hardware-map up2.map -T tests/kernel/ -vv`

**Logs and console output**
too many tests fail, here is an example output:
```
2024-01-25 13:19:23,768 - twister - DEBUG - DEVICE: *** Booting Zephyr OS build zephyr-v3.5.0-4714-g3fc5d971fe28 ***
2024-01-25 13:19:23,782 - twister - DEBUG - DEVICE: E: Page fault at address 0x22 (error code 0x0)
2024-01-25 13:19:23,785 - twister - DEBUG - DEVICE: Running TESTSUITE obj_core
2024-01-25 13:19:23,789 - twister - DEBUG - DEVICE: E: Linear address not present in page tables
2024-01-25 13:19:23,796 - twister - DEBUG - DEVICE: ===================================================================
2024-01-25 13:19:23,802 - twister - DEBUG - DEVICE: E: Access violation: supervisor thread not allowed to read
2024-01-25 13:19:23,805 - twister - DEBUG - DEVICE: START - test_obj_core_condvar
2024-01-25 13:19:23,807 - twister - DEBUG - DEVICE: E: PTE: not present
2024-01-25 13:19:23,817 - twister - DEBUG - DEVICE: E: RAX: 0x0000000000000000 RBX: 0x0000000000000046 RCX: 0x0000000000000001 RDX: 0x000000000000003b
2024-01-25 13:19:23,826 - twister - DEBUG - DEVICE: E: RSI: 0x0000000000000000 RDI: 0x0000000000000000 RBP: 0x00000000005b3f90 RSP: 0x00000000005b3f80
2024-01-25 13:19:23,836 - twister - DEBUG - DEVICE: E:  R8: 0x80000000fee00a13  R9: 0x0000000000000000 R10: 0x0000000000000000 R11: 0x0000000000000003
2024-01-25 13:19:23,846 - twister - DEBUG - DEVICE: E: R12: 0x00000000006bf7d0 R13: 0x0000000000000000 R14: 0x0000000000000000 R15: 0x0000000000000000
2024-01-25 13:19:23,854 - twister - DEBUG - DEVICE: E: RSP: 0x00000000005b3f80 RFLAGS: 0x0000000000010046 CS: 0x0018 CR3: 0x00000000008e8000
2024-01-25 13:19:23,857 - twister - DEBUG - DEVICE: E: RIP: 0x00000000001099f7
2024-01-25 13:19:23,862 - twister - DEBUG - DEVICE: E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 1
2024-01-25 13:19:23,865 - twister - DEBUG - DEVICE: E: Current thread: 0 (unknown)
2024-01-25 13:19:23,867 - twister - DEBUG - DEVICE: E: Halting system
2024-01-25 13:20:04,652 - twister - DEBUG - Timed out while monitoring serial output on up_squared
2024-01-25 13:20:04,755 - twister - DEBUG - Expected suite names:['obj_core', 'obj_core']
2024-01-25 13:20:04,755 - twister - DEBUG - Detected suite names:['obj_core']
```

**Environment:**
 - OS: Linux
 - Toolchain: zephyr-sdk-0.16.3
 - zephyr-v3.5.0-4714-g3fc5d971fe28

KamilxPaszkiet, 2024-01-26:
I did a bisection and found a commit that breaks the tests: eefaeee
@nashif, @dcpleung , please take a look at this commit.

KamilxPaszkiet, 2024-01-31:
@dcpleung 

I found another group of failing tests:
`tests/lib/c_lib/thrd`

more in initial comment

cfriedt, 2024-01-31:
I've found recently that x86* platforms require stack sizes >= 4k. Is that possibly related?

I know there was also some mention of a particular jump being too far for the instruction being used somewhere.

KamilxPaszkiet, 2024-02-05:
Now **kernel** tests pass. However, **c_lib** still has problems.

Sample log(tests/lib/c_lib/thrd/libraries.libc.c11_threads.minimal):
```
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v3.5.0-5382-g277d649bdaed ***
DEBUG   - DEVICE: Running TESTSUITE libc_cnd
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_cnd_broadcast_wait
DEBUG   - DEVICE: ASSERTION FAIL [__attr_is_initialized(attr)] @ WEST_TOPDIR/zephyr/lib/posix/options/pthread.c:808
DEBUG   - DEVICE: E: RAX: 0x0000000000000004 RBX: 0x000000000013d590 RCX: 0x0000000000000001 RDX: 0x0000000000000000
DEBUG   - DEVICE: E: RSI: 0x0000000000000328 RDI: 0x0000000000135225 RBP: 0x0000000000ceaec0 RSP: 0x0000000000ceaea8
DEBUG   - DEVICE: E:  R8: 0x0000000000000008  R9: 0x0000000000ceac00 R10: 0x00000000ffffffff R11: 0x0000000000cead10
DEBUG   - DEVICE: E: R12: 0x0000000000000000 R13: 0x0000000000100407 R14: 0x000000000013c010 R15: 0x000000000013d590
DEBUG   - DEVICE: E: RSP: 0x0000000000ceaea8 RFLAGS: 0x0000000000000246 CS: 0x0018 CR3: 0x0000000000f1a000
DEBUG   - DEVICE: E: RIP: 0x0000000000103687
DEBUG   - DEVICE: E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
DEBUG   - DEVICE: E: Current thread: 0x13ea40 (test_cnd_broadcast_wait)
DEBUG   - DEVICE: Caught system error -- reason 4 0
DEBUG   - DEVICE: Fatal error was unexpected, aborting...
```

dcpleung, 2024-02-05:
This is a different type of error than the one reported above, which was NULL pointer reference. This one is asserting on POSIX related stuff. Could you open a new bug about that? Thanks.

KamilxPaszkiet, 2024-02-06:
> This is a different type of error than the one reported above, which was NULL pointer reference. This one is asserting on POSIX related stuff. Could you open a new bug about that? Thanks.

I open new bug: https://github.com/zephyrproject-rtos/zephyr/issues/68616

PR #68388 (2024-02-05): kernel: smp: CPU start may result in null pointer access
It is observed that starting up CPU may result in other CPUs crashing due to de-referencing NULL pointers. Note that this happened on the up_squared board, but there was no way to attach debugger to verify. It started working again after moving z_dummy_thread_init() before smp_timer_init(), which was the old behavior before commit
eefaeee061c869cfa33be9693ff186ca60f76ff9 where the issue first appeared. So mimic the old behavior to workaround the issue.

Fixes #68115
commit caacc27d (2024-02-05): kernel: smp: CPU start may result in null pointer access

It is observed that starting up CPU may result in other CPUs
crashing due to de-referencing NULL pointers. Note that this
happened on the up_squared board, but there was no way to
attach debugger to verify. It started working again after
moving z_dummy_thread_init() before smp_timer_init(), which
was the old behavior before commit
eefaeee061c869cfa33be9693ff186ca60f76ff9 where the issue
first appeared. So mimic the old behavior to workaround
the issue.

Fixes #68115
closed 2024-02-06 by KamilxPaszkiet