zephyrproject-rtos/zephyr #68120: logging: multicore logging with backend IPC service asserts with "Unexpected message" error

bogdanm, 2024-01-25:
This occured on an nRF5340, but it likely affects other targets too.

Multidomain logging configuration on the nRF5340 app core:

```
CONFIG_LOG=y
CONFIG_LOG_MULTIDOMAIN=y
CONFIG_LOG_LINK_IPC_SERVICE=y
CONFIG_LOG_PROCESS_THREAD=y
CONFIG_LOG_OUTPUT=y
```

Multidomain logging configuration on the nRF5340 net core:

```
CONFIG_LOG=y
CONFIG_LOG_MULTIDOMAIN=y
CONFIG_LOG_BACKEND_IPC_SERVICE=y
CONFIG_LOG_PROCESS_THREAD=y
CONFIG_LOG_OUTPUT=y
```

With these options enabled, the app core asserts here:

https://github.com/zephyrproject-rtos/zephyr/blob/73396f9f8644762c4b547679d9b2baabadc80924/subsys/logging/log_multidomain_link.c#L82

This happens because `log_multidomain_link_on_recv_cb` (in the previous link) doesnt handle the value `Z_LOG_MULTIDOMAIN_ID_DROPPED` that might be sent by the backend IPC service running on the net core. A patch that fixes the issue is this:

```diff
diff --git a/subsys/logging/log_multidomain_link.c b/subsys/logging/log_multidomain_link.c
index c730f40545..57a513b230 100644
--- a/subsys/logging/log_multidomain_link.c
+++ b/subsys/logging/log_multidomain_link.c
@@ -77,6 +77,7 @@ void log_multidomain_link_on_recv_cb(struct log_multidomain_link *link_remote,
                link_remote->dst.set_runtime_level.level = msg->data.set_rt_level.runtime_level;
                break;
        case Z_LOG_MULTIDOMAIN_ID_READY:
+       case Z_LOG_MULTIDOMAIN_ID_DROPPED:
                break;
        default:
                __ASSERT(0, "Unexpected message");
```

However, I don't know if this is the best way to handle this issue.

My Zephyr version: https://github.com/nrfconnect/sdk-zephyr/tree/v3.4.99-ncs1-1

henrikbrixandersen, 2024-01-25:
Thank you for reporting this. However - unless you are able to reproduce this issue with upstream Zephyr `main` - please report issues with the nRF Connect SDK (NCS) on the [Nordic Semiconductor DevZone](https://devzone.nordicsemi.com/).

bogdanm, 2024-01-25:
I can't compile my code with mainline Zephyr. I can open an issue on the NCS repository instead, but I doubt that the results will be different. In https://github.com/zephyrproject-rtos/zephyr/blob/73396f9f8644762c4b547679d9b2baabadc80924/subsys/logging/backends/log_multidomain_backend.c#L222-L237 you can see how the IPC log backend sends `Z_LOG_MULTIDOMAIN_ID_DROPPED` which isn't handled by `log_multidomain_link.c` (as described above).

lopsided98, 2024-04-13:
This bug can be reproduced in mainline Zephyr and can cause memory corruption and crashes if assertions are disabled. The fix proposed in this issue fixes the assertion, but not the memory corruption. See #71464 for more explanation.

henrikbrixandersen, 2024-04-13:
@anangl Please take a look.

closed 2024-04-18 by aescolar