zephyrproject-rtos/zephyr #68129: test-ci: drivers: adp_xc7k_ae350 fails to build DMA loop_transfer test

DerekSnell, 2024-01-25:
Twister found the loop_transfer test fails to build on the board adp_xc7k_ae350.  When building with commit 73396f9f8644762c4b547679d9b2baabadc80924

Build command and error:
```
$ west build -b adp_xc7k_ae350 tests/drivers/dma/loop_transfer/
...
/zephyrproject/zephyr/drivers/dma/dma_andes_atcdmac300.c:14:10: fatal error: soc.h: No such file or directory
   14 | #include <soc.h>
      |          ^~~~~~~
compilation terminated.
```

teburd, 2024-01-25:
@kevinwang821020 help needed here

kevinwang821020, 2024-01-30:
> @kevinwang821020 help needed here, please also setup the MAINTAINERS file for the Andes related drivers, noticeably absent

Thank you for your reminder. I have submitted a pull request (PR) addressing this bug. As for the MAINTAINERS file, updates will be made later.

closed 2024-02-01 by dleach02