zephyrproject-rtos/zephyr #68173: Bluetooth: TBS: Value Changed During Long Read error code

piotrnarajowski, 2024-01-26:
**Describe the bug**
These test cases verify the behavior of the server IUT when we update TBS characteristics with value greater then ATT_MTU-3. We change the value of characteristic (e.g. Bearer Provider Name) with value len greater than ATT_MTU-3. PTS then reads the characteristic value by executing GATT Read Long Characteristic Value with offset set to 0. By the time there is second long read request with a Value Offset of (ATT_MTU-3), we change the value one more time. Zephyr server IUT doesn't respond with Value Changed During Read Long (0x80) error code.

**To Reproduce**
Can be reproduced with AutoPTS.

**Expected behavior**
According to [TBS spec](https://www.bluetooth.com/specifications/tbs-1-0/), section 1.4:

> The server shall return an ATT Error Response with the application error code Value Changed During Read Long (0x80) as defined in Table 1.2 to any ATT_READ_BLOB_REQ with a non-zero Value Offset ifthe characteristic value has changed since the last zero offset read response of that characteristic value to that client.

**Impact**
Failing test cases:
TBS/SR/SPN/BV-05-C
TBS/SR/SPN/BV-06-C
TBS/SR/SPN/BV-07-C
TBS/SR/SPN/BV-08-C
TBS/SR/SPN/BV-09-C
TBS/SR/SPN/BV-10-C
TBS/SR/SPN/BV-11-C
Also same test cases with GTBS prefix

**Logs and console output**
[TBS_SR_SPN_BV_05_C_2024_01_26_17_27_12.zip](https://github.com/zephyrproject-rtos/zephyr/files/14067280/TBS_SR_SPN_BV_05_C_2024_01_26_17_27_12.zip)

**Environment (please complete the following information):**
 - OS: Windows 11
 - Toolchain: Zephyr SDK 0.16.4
 - https://github.com/zephyrproject-rtos/zephyr/pull/68116

Thalley, 2024-02-16:
@piotrnarajowski how does your PR (https://github.com/zephyrproject-rtos/zephyr/pull/68367) relate to this? Your PR is required to trigger long reads, right? So this can't really be worked on, before your PR has been done (or at least the work should continue on top of your PR), right?

piotrnarajowski, 2024-02-16:
@Thalley Yes, PR #68367 is necessary to properly trigger long reads and to reproduce this issue. I'll try, but I'm not sure I will be able to finish it as LE Audio project is coming to an end for me.

Thalley, 2024-03-06:
@piotrnarajowski Will your PR ever go out of draft, or will it be abandoned now?

piotrnarajowski, 2024-03-06:
@Thalley for now @sjanc will take a look at this, please assign him

Thalley, 2024-03-12:
Related to https://github.com/zephyrproject-rtos/zephyr/issues/64175

Thalley, 2024-11-13:
@sjanc any plan for this? It's one of the few open bugs we have for LE Audio, so would like to get it closed in the near future

sjanc, 2024-11-14:
I'm not sure when I'll have time to look at this

Thalley, 2024-11-14:
> I'm not sure when I'll have time to look at this

No problem. I'll re-assign to me, and hopefully have a look at in soon