zephyrproject-rtos/zephyr #68248: flash_handler.c - wrong syscall includes

boaks, 2024-01-29:
[flash_handler.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/flash/flash_handlers.c) includes [<syscalls/flash_sfdp_read.c>](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/flash/flash_handlers.c#L97) and [<syscalls/flash_sfdp_jedec_id.c>](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/flash/flash_handlers.c#L106).

On "my machine" (NCS 2.5.1 / v3.4.99-ncs1-1) this files are named `<syscalls/flash_sfdp_read_mrsh.c>` and `<syscalls/flash_read_jedec_id_mrsh.c>`.

Building with "CONFIG_USERSPACE=y" fails therefore with:

> zephyr/drivers/flash/flash_handlers.c:97:10: fatal error: syscalls/flash_sfdp_read.c: No such file or directory
   97 | #include <syscalls/flash_sfdp_read.c>

Building with "CONFIG_USERSPACE=n" works.

Changing the filenames to the available "_mrsh.c" ones makes building with "CONFIG_USERSPACE=y" also working.

I guess this are just typos / left overs.

henrikbrixandersen, 2024-01-29:
Thank you for reporting this. However - unless you are able to reproduce this issue with upstream Zephyr `main` - please report issues with the nRF Connect SDK (NCS) on the [Nordic Semiconductor DevZone](https://devzone.nordicsemi.com/).

kartben, 2024-01-29:
I am able to reproduce... kinda. Looks like this might be a build system or west issue (or even expected behavior, I am really not familiar at all with userspace and syscalls)?

This works:

```
west build -p always -b frdm_k64f samples/userspace/hello_world_user -- -DCONFIG_FLASH=y
```

This does not:

```
west build -p always -b frdm_k64f samples/userspace/hello_world_user
west build -p auto -b frdm_k64f samples/userspace/hello_world_user -- -DCONFIG_FLASH=y

...
-- west build: building application
[58/150] Building C object zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/flash_handlers.c.obj
FAILED: zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/flash_handlers.c.obj 
ccache /Users/kartben/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DCPU_MK64FN1M0VLL12 -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I/Users/kartben/zephyrproject/zephyr/include -I/Users/kartben/zephyrproject/zephyr/build/zephyr/include/generated -I/Users/kartben/zephyrproject/zephyr/soc/arm/nxp_kinetis/k6x -I/Users/kartben/zephyrproject/zephyr/soc/arm/nxp_kinetis/common -I/Users/kartben/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/Users/kartben/zephyrproject/zephyr/modules/cmsis/. -I/Users/kartben/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/devices/MK64F12 -I/Users/kartben/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/devices/MK64F12/drivers -I/Users/kartben/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/common -I/Users/kartben/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/uart -I/Users/kartben/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/flash -I/Users/kartben/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/port -I/Users/kartben/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/sysmpu -I/Users/kartben/zephyrproject/zephyr/modules/hal_nxp/. -isystem /Users/kartben/zephyrproject/zephyr/lib/libc/common/include -Wshadow -fno-strict-aliasing -Os -imacros /Users/kartben/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft --sysroot=/Users/kartben/zephyr-sdk-0.16.1/arm-zephyr-eabi/arm-zephyr-eabi -imacros /Users/kartben/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/Users/kartben/zephyrproject/zephyr/samples/userspace/hello_world_user=CMAKE_SOURCE_DIR -fmacro-prefix-map=/Users/kartben/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/Users/kartben/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections --specs=picolibc.specs -std=c99 -MD -MT zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/flash_handlers.c.obj -MF zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/flash_handlers.c.obj.d -o zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/flash_handlers.c.obj -c /Users/kartben/zephyrproject/zephyr/drivers/flash/flash_handlers.c
/Users/kartben/zephyrproject/zephyr/drivers/flash/flash_handlers.c:19:10: fatal error: syscalls/flash_read_mrsh.c: No such file or directory
   19 | #include <syscalls/flash_read_mrsh.c>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[71/150] Building C object zephyr/drivers/gpio/CMakeFiles/drivers__gpio.dir/gpio_handlers.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /opt/homebrew/bin/cmake --build /Users/kartben/zephyrproject/zephyr/build
...

```

Laczen, 2024-01-30:
I can confirm this is a bug, the includes in `flash_handlers` should have the suffix `_mrsh.c`.

boaks, 2024-01-30:
> However - unless you are able to reproduce this issue with upstream Zephyr main - please report issues with the nRF Connect SDK (NCS) on the [Nordic Semiconductor DevZone](https://devzone.nordicsemi.com/).

I assumed, that the links I provided (pointing to the current sources in main) and the description, how I reproduced it, should be enough information. Sure, I don't know, if the generated file names or the includes are wrong. Hope that the other comments here are already providing the missing information and I guess, reporting it to Nordic's DevZone will not be required.

boaks, 2024-01-30:
> zephyr/drivers/flash/flash_handlers.c:19:10: fatal error: syscalls/flash_read_mrsh.c: No such file or directory
   19 | #include <syscalls/flash_read_mrsh.c>

Hm, that may point more to the general question, if the generated files or the includes are wrong. I consider, that the developers need to sync on that topic.

kartben, 2024-01-30:
> I can confirm this is a bug, the includes in `flash_handlers` should have the suffix `_mrsh.c`.
> 

It's pretty odd that it works with a clean build though?? See my comment above

boaks, 2024-01-30:
> It's pretty odd that it works with a clean build though?? See my comment above

I may be wrong, but in my experience, changing the config as "-DCONFIG_FLASH=y" does, works "best", if you use "--pristine" ;-).
It's not easy to build such a powerful build system with intermediate files in an "incremental way". Therefore I'm in the meantime used to do pretty frequently a "--pristine" build, if something in the config has changed.

Laczen, 2024-01-30:
> > I can confirm this is a bug, the includes in `flash_handlers` should have the suffix `_mrsh.c`.
> 
> It's pretty odd that it works with a clean build though?? See my comment above

The build will fail when building for a board that uses `flash_sfdp_read` and `flash_read_jedec_id`, the clean build that is shown does not provide any information about this.

kartben, 2024-01-30:
> I may be wrong, but in my experience, changing the config as "-DCONFIG_FLASH=y" does, works "best", if you use "--pristine" ;-).
> It's not easy to build such a powerful build system with intermediate [...]

I don't understand your comment, sorry.

My point is that I am not able to reproduce the bug when I perform a clean build of the Hello World userspace sample with CONFIG_FLASH=Y (regardless of whether I pass the option via `-D` or in `prj.conf`)

Laczen, 2024-01-30:
It is easy to reproduce this bug on a nrf52840dk_nrf52840 when compiling with the config options:
```
CONFIG_FLASH=y
CONFIG_FLASH_JESD216_API=y
```
This will even reveal another bug in `nrf_qspi_nor.c`.

kartben, 2024-01-30:
> The build will fail when building for a board that uses `flash_sfdp_read` and `flash_read_jedec_id`, the clean build that is shown does not provide any information about this.

I see. Still weird that this would fail when adding CONFIG_FLASH as par of an incremental build.

FWIW, and to your point, this fails 100%:

```
west build -p -b efr32mg_sltb004a samples/drivers/jesd216 -- -DCONFIG_USERSPACE=y
```

boaks, 2024-01-30:
> I don't understand your comment, sorry.
>
> My point is that I am not able to reproduce the bug when I perform a clean build of the Hello World userspace sample with CONFIG_FLASH=Y (regardless of whether I pass the option via -D or in prj.conf)

> I may be wrong, but in my experience, changing the config as "-DCONFIG_FLASH=y" 

What I want to say is, that "any" change (therefore the "changing the config as") in the config, regardless, if it's passed by -D or by editing a prj.conf, tends to require in my experience a pristine build. If you use "-p always" it works, if you use "-p auto" it seems to not work, maybe, because it doesn't clean the intermediate files.

PR #68277 (2024-01-30): flash: correct userspace flash_handlers
fixes #68248
commit 50597b2e (2024-01-30): flash: correct userspace flash_handlers

fixes #68248
closed 2024-01-30 by nashif