zephyrproject-rtos/zephyr #68272: qemu_x86_tiny@768 is asserting in the philosophers sample

ycsin, 2024-01-30:
**Describe the bug**
`qemu_x86_tiny@768` board is asserting in the philosophers sample:

```
SeaBIOS (version rel-1.16.2-0-gea1b7a073390-prebuilt.qemu.org)
Booting from ROM..*** Booting Zephyr OS build zephyr-v3.5.0-4939-g1e1d2725a712 ***
ASSERTION FAIL [!arch_is_in_isr()] @ WEST_TOPDIR/zephyr/kernel/include/ksched.h:255
```

**To Reproduce**
Steps to reproduce the behavior:
1. Check out main (8873c0784772413ee521abae4b9129dd7cd9e3ee)
2. `west build -b qemu_x86_tiny@768 -t run -p auto -T zephyr/samples/philosophers/sample.kernel.philosopher.fifos`
3. See error

**Expected behavior**
The board should run the philosophers sample properly without assertion.

**Impact**
This is failing in [CI tests](https://github.com/zephyrproject-rtos/zephyr/actions/runs/7702666462/job/20996682868?pr=68188) of #68188

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: 8873c0784772413ee521abae4b9129dd7cd9e3ee

fabiobaltieri, 2024-01-30:
@edersondisouza is this what you bumped into with https://github.com/zephyrproject-rtos/zephyr/pull/67877#issuecomment-1912865782?

edersondisouza, 2024-01-30:
> @edersondisouza is this what you bumped into with [#67877 (comment)](https://github.com/zephyrproject-rtos/zephyr/pull/67877#issuecomment-1912865782)?

Not really - I get an mmu assertion actually:

```
ASSERTION FAIL [page_frames_initialized] @ WEST_TOPDIR/zephyr/kernel/mmu.c:1395
	page fault at 0x401944be happened too early
```
And that's definitely related to my new code there.

cfriedt, 2024-01-31:
@jhedberg, @edersondisouza - recently, on a seemingly unrelated pr, I found that code would only work properly under qemu_x86 and qemu_x86_64 if stack sizes were >= 4096. I wasn't sure why this was necessary all of a sudden. 

Is it possible that this is related to that?

Also, @jhedberg , I would guess that this is related to #68115

cfriedt, 2024-01-31:
> @cfriedt I'm not aware of any recent changes to x86 stack sizes or their requirements. You say they need to be over 4k. Do all stacks need this or just some? What are the sizes now? Did you enable the stack size analysis to see exactly how much is needed?

I needed 4k on qemu_x86, for example, even with `K_THREAD_STACK_ARRAY_DEFINE()`, just to not page fault.

Did not look into it in detail unfortunately.

cfriedt, 2024-01-31:
Could it be alignment or padding requirements for MMU pages?

PR #68379 (2024-02-01): boards: x86: more stack space for qemu_x86_tiny@768
qemu_x86_tiny@768 has coverage enabled by default. Because of this, it requires more stack space for running tests. The increases needed are verified via twister.

Fixes #68272
commit a053c55c (2024-02-01): boards: x86: more stack space for qemu_x86_tiny@768

qemu_x86_tiny@768 has coverage enabled by default. Because of
this, it requires more stack space for running tests.
The increases needed are verified via twister.

Fixes #68272
closed 2024-02-01 by dleach02