zephyrproject-rtos/zephyr #68273: `qemu_xtensa` is failing in `tests/subsys/llext/hello_world/llext.simple.modules_enabled`

ycsin, 2024-01-30:
**Describe the bug**
`qemu_xtensa` is failing in `tests/subsys/llext/hello_world/llext.simple.modules_enabled`:

```
Generating files from /Users/ycsin/zephyrproject/build/zephyr/zephyr.elf for board: qemu_xtensa
[134/135] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: dc233c
*** Booting Zephyr OS build zephyr-v3.5.0-4944-g8873c0784772 ***
Running TESTSUITE llext
===================================================================
START - test_llext_simple
[00:00:00.000,000] <dbg> llext: llext_load: Loading relocatable or shared elf
[00:00:00.000,000] <dbg> llext: do_llext_load: Finding ELF tables...
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 0 at 3d0: name 0, type 0, flags 0, addr 0, size 0
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 1 at 3f8: name 11, type 5, flags 2, addr 94, size 48
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 2 at 420: name 17, type 11, flags 2, addr c4, size 112
[00:00:00.000,000] <dbg> llext: llext_find_tables: symtab at 2
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 3 at 448: name 25, type 3, flags 2, addr 134, size 51
[00:00:00.000,000] <dbg> llext: llext_find_tables: strtab at 3
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 4 at 470: name 33, type 4, flags 2, addr 168, size 72
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 5 at 498: name 43, type 4, flags 42, addr 1b0, size 12
[00:00:00.010,000] <dbg> llext: llext_find_tables: section 6 at 4c0: name 53, type 1, flags 6, addr 1bc, size 76
[00:00:00.010,000] <dbg> llext: llext_find_tables: section 7 at 4e8: name 59, type 1, flags 2, addr 208, size 52
[00:00:00.010,000] <dbg> llext: llext_find_tables: section 8 at 510: name 67, type 1, flags 2, addr 23c, size 16
[00:00:00.010,000] <dbg> llext: llext_find_tables: section 9 at 538: name 76, type 6, flags 3, addr 124c, size 160
[00:00:00.010,000] <dbg> llext: llext_find_tables: section 10 at 560: name 85, type 1, flags 3, addr 12ec, size 4
[00:00:00.010,000] <dbg> llext: llext_find_tables: section 11 at 588: name 90, type 1, flags 3, addr 12f0, size 8
[00:00:00.010,000] <dbg> llext: llext_find_tables: section 12 at 5b0: name 104, type 1, flags 30, addr 0, size 32
[00:00:00.010,000] <dbg> llext: llext_find_tables: section 13 at 5d8: name 113, type 7, flags 0, addr 0, size 56
[00:00:00.010,000] <dbg> llext: llext_find_tables: section 14 at 600: name 1, type 3, flags 0, addr 0, size 126
[00:00:00.010,000] <dbg> llext: llext_find_tables: shstrtab at 14
[00:00:00.010,000] <dbg> llext: do_llext_load: Allocate and copy strings...
[00:00:00.020,000] <dbg> llext: do_llext_load: Mapping ELF sections...
[00:00:00.020,000] <dbg> llext: llext_map_sections: section 0 name 
[00:00:00.020,000] <dbg> llext: llext_map_sections: Not copied section 
[00:00:00.020,000] <dbg> llext: llext_map_sections: section 1 name .hash
[00:00:00.020,000] <dbg> llext: llext_map_sections: Not copied section .hash
[00:00:00.020,000] <dbg> llext: llext_map_sections: section 2 name .dynsym
[00:00:00.020,000] <dbg> llext: llext_map_sections: Not copied section .dynsym
[00:00:00.020,000] <dbg> llext: llext_map_sections: section 3 name .dynstr
[00:00:00.020,000] <dbg> llext: llext_map_sections: Not copied section .dynstr
[00:00:00.020,000] <dbg> llext: llext_map_sections: section 4 name .rela.dyn
[00:00:00.020,000] <dbg> llext: llext_map_sections: Not copied section .rela.dyn
[00:00:00.020,000] <dbg> llext: llext_map_sections: section 5 name .rela.plt
[00:00:00.020,000] <dbg> llext: llext_map_sections: Not copied section .rela.plt
[00:00:00.020,000] <dbg> llext: llext_map_sections: section 6 name .text
[00:00:00.020,000] <dbg> llext: llext_map_sections: section 7 name .rodata
[00:00:00.030,000] <dbg> llext: llext_map_sections: section 8 name .got.loc
[00:00:00.030,000] <dbg> llext: llext_map_sections: Not copied section .got.loc
[00:00:00.030,000] <dbg> llext: llext_map_sections: section 9 name .dynamic
[00:00:00.030,000] <dbg> llext: llext_map_sections: Not copied section .dynamic
[00:00:00.030,000] <dbg> llext: llext_map_sections: section 10 name .got
[00:00:00.030,000] <dbg> llext: llext_map_sections: Not copied section .got
[00:00:00.030,000] <dbg> llext: llext_map_sections: section 11 name .exported_sym
[00:00:00.030,000] <dbg> llext: llext_map_sections: section 12 name .comment
[00:00:00.030,000] <dbg> llext: llext_map_sections: Not copied section .comment
[00:00:00.030,000] <dbg> llext: llext_map_sections: section 13 name .xtensa.info
[00:00:00.030,000] <dbg> llext: llext_map_sections: Not copied section .xtensa.info
[00:00:00.030,000] <dbg> llext: llext_map_sections: section 14 name .shstrtab
[00:00:00.030,000] <dbg> llext: llext_map_sections: Not copied section .shstrtab
[00:00:00.030,000] <dbg> llext: do_llext_load: Allocate and copy sections...
[00:00:00.040,000] <dbg> llext: do_llext_load: Counting exported symbols...
[00:00:00.040,000] <dbg> llext: llext_count_export_syms: symbol count 7
[00:00:00.040,000] <dbg> llext: llext_count_export_syms: unhandled symbol 1, name symbol_hello_world, type tag 1, bind 1, sect 11
[00:00:00.040,000] <dbg> llext: llext_count_export_syms: unhandled symbol 2, name printk, type tag 0, bind 1, sect 0
[00:00:00.040,000] <dbg> llext: llext_count_export_syms: unhandled symbol 3, name __bss_start, type tag 0, bind 1, sect 11
[00:00:00.040,000] <dbg> llext: llext_count_export_syms: function symbol 4, name hello_world, type tag 2, bind 1, sect 6
[00:00:00.040,000] <dbg> llext: llext_count_export_syms: unhandled symbol 5, name _edata, type tag 0, bind 1, sect 11
[00:00:00.040,000] <dbg> llext: llext_count_export_syms: unhandled symbol 6, name _end, type tag 0, bind 1, sect 11
[00:00:00.040,000] <dbg> llext: do_llext_load: Allocating memory for symbol table...
[00:00:00.040,000] <dbg> llext: do_llext_load: Copying symbols...
[00:00:00.040,000] <dbg> llext: llext_copy_symbols: function symbol 0 name hello_world addr 0x12474
[00:00:00.040,000] <dbg> llext: do_llext_load: Linking ELF...
[00:00:00.050,000] <dbg> llext: llext_link_plt: Found 0x123ad in PLT 4 size 12 cnt 6 text 0x1244c
[00:00:00.050,000] <dbg> llext: llext_link_plt: symbol  offset 0xc r-offset 0x1c8 .text offset 0x1bc stb 0
[00:00:00.050,000] <dbg> llext: llext_link_plt: symbol  offset 0x14 r-offset 0x1d0 .text offset 0x1bc stb 0
[00:00:00.050,000] <dbg> llext: llext_link_plt: symbol  offset 0x134 r-offset 0x12f0 .text offset 0x1bc stb 0
[00:00:00.050,000] <dbg> llext: llext_link_plt: symbol  offset 0 r-offset 0x1bc .text offset 0x1bc stb 0
[00:00:00.050,000] <dbg> llext: llext_link_plt: symbol  offset 0x4 r-offset 0x1c0 .text offset 0x1bc stb 0
[00:00:00.050,000] <dbg> llext: llext_link_plt: symbol hello_world offset 0x138 r-offset 0x12f4 .text offset 0x1bc stb 1
[00:00:00.050,000] <dbg> llext: llext_link_plt: Found 0x123b7 in PLT 4 size 12 cnt 1 text 0x1244c
[00:00:00.050,000] <dbg> llext: llext_link_plt: symbol printk offset 0x10 r-offset 0x1cc .text offset 0x1bc stb 1
[00:00:00.050,000] <dbg> llext: llext_export_symbols: sym 0 name  in 0x1256c
[00:00:00.050,000] <dbg> llext: do_llext_load: loaded module, .text at 0x1244c, .rodata at 0x1249c
[00:00:00.060,000] <inf> llext: Loaded extension hello

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/llext/hello_world/src/test/test_llext_simple.c:46: llext_test_llext_simple: (hello_world_fn is NULL)
hello_world should be an exported symbol
 FAIL - test_llext_simple in 0.061 seconds
===================================================================
TESTSUITE llext failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -   0.00% [llext]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.061 seconds
 - FAIL - [llext.test_llext_simple] duration = 0.061 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION FAILED
```

**To Reproduce**
Steps to reproduce the behavior:
1. Check out main (8873c0784772413ee521abae4b9129dd7cd9e3ee)
2. `west build -b qemu_xtensa -t run -p auto -T zephyr/tests/subsys/llext/hello_world/llext.simple.modules_enabled`
3. See error

**Expected behavior**
The board should finishes the test properly without assertion.

**Impact**
This is failing in [CI tests](https://github.com/zephyrproject-rtos/zephyr/actions/runs/7707647237/job/21005329683?pr=68268) of #68268

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: 8873c0784772413ee521abae4b9129dd7cd9e3ee

teburd, 2024-01-30:
The issue seems to be two things..

* llext on xtensa has only been tested to work with the WRITABLE_STORAGE Kconfig
* when WRITABLE_STORAGE is not enabled the exported symbol name pointer is not valid and subsequently the symbol lookup fails (and asserts in the test).

This changed recently as the testcase.yaml was updated in 92a5e3ace755f33006b0579e874fc926041977f5 though not really the direct cause I believe ungated some testcase builds/runs that maybe weren't quite ready to be ungated.

ycsin, 2024-01-31:
fixed by #68305

closed 2024-01-31 by ycsin