zephyrproject-rtos/zephyr #68283: LE Audio: Broadcast Assistant should discover PACS

Thalley, 2024-01-30:
**Is your enhancement proposal related to a problem? Please describe.**
A BAP broadcast assistant should discover PACS on the Broadcast Sink/Scan Delegator device to determine if it supports a broadcast source. Currently this isn't possible with the API. 

**Describe the solution you'd like**
Similar to how the `bt_bap_unicast_client_discover` discover PACS, the `bt_bap_broadcast_assistant` should also discover PACS on the remote device. 

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

jthm-ot, 2024-06-28:
I can take it

Chris-StJohn-Bose, 2024-11-07:
@jthm-ot @Thalley please be aware that there are proposals to extend PACS in the near future. I'm running IOP with a modified version of `pacs.c` and `bap_unicast_client.c`. It would be nice if your refactor could make extending PACS a little easier.

Thalley, 2024-11-13:
@jthm-ot Any plans for this in the near future? 
Since this will likely conflict with https://github.com/zephyrproject-rtos/zephyr/issues/70826 I'm like to have at least 1 of the 2 implemented in the near future

Thalley, 2024-11-14:
> It uses a single discovery callback method rather than a chain of them. Just an idea...

We actually used to have that, but we changed it to a chain. 

There are a couple of reasons for this:
1) If you only discover a single characteristic at a time using the UUID (compared to a general discovery for `BT_GATT_DISCOVER_CHARACTERISTIC`, then you don't get much value out of having a combine callback fucntion, as nearly nothing will be shared. 
2) If you do a more general discover, then this function may be called multiple time from the same read request response, as it it called _per_ handle. This is great for optimization, but one needs to be aware of that, especially if you want to do calls to `bt_gatt_subscribe`, as `bt_gatt_subscribe` is a blocking operation, and you e.g. get 5 handles in a response, then you may end up calling `bt_gatt_subscribe` 5 times in the same response, and if you run out of ATT buffers, then your call to `bt_gatt_subscribe` will fail, and you won't subscribe to all characteristics, which is mandatory in some cases (like BAP). The only way to circumvent this, is to have enough buffers, but since you may find up to 255 sink ASEs and 255 source ASEs, that's too many buffers to handle worst case. Discovering them one by one limits it to only require a single buffer. 
3) Finally to subscribe you need to know the CCC handle. The CONFIG_BT_GATT_AUTO_DISCOVER_CCC feature is pretty neat for this, but it requires a discovery parameter. Similar to the issue in 2), doing parallel discoveries will require additional discover parameters. Discovering it in a chain will only require 1 such parameter, and you can even re-use the discover parameter that was used to find the characteristic you want to subscribe to in the first place. This will however only work if you always return `BT_GATT_ITER_STOP` if you want to reused the discover paramters. 

So in short, we did use that approach earlier and it could speed up discovery, but at the cost of (significant) memory. It is how e.g. the TBS client works today (https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/audio/tbs_client.c#L1554-L1719), but where TBS needs up to 11 ATT buffers for discovery, and that's only if no other ATT operation is done at the same time.

Thalley, 2026-03-20:
I think perhaps a good solution is to move PACS discovery out from the BAP unicast client and into it's own implementation/function, i.e. create a `bt_pacs_client` API for this (and other PACS related stuff, since it's not specific to one BAP role). Since BAP unicast client discovery works on a single direction at a time (for now), and and since BAP Broadcast Assistant is always looking for BASS (sink endpoint), it may make sense to not do PACS discovery as part of either the unicast client's or broadcast assistant's discovery functions, but only as a standalone discovery function. That way the application can discover PACS and the read the records first, and then decide if it wants to continue with any broadcast or unicast audio discovery.