zephyrproject-rtos/zephyr #68306: Add driver that support cf1133 touchscreen controller from Sitronix

clemdy, 2024-01-30:
Currently Zephyr proposes no driver to support Sitronix I2C touchscreen controller.

Adding an out of tree driver that support Sitronix I2C touchscreen controller.

kartben, 2024-01-30:
Just to make sure I understand correctly, it looks likes you are intersted in contributing your driver (https://github.com/zephyrproject-rtos/zephyr/commit/14197b3075c237e1ff649989b4430a3d0805d45f), right?
Thanks in advance :)

kartben, 2024-01-31:
> Hey yes exactly ! But i'm wondering if a pull request would be accepted right now or if i need to do more things

Of course it would be accepted! You could even have directly submitted the PR and not bothered with opening an issue, even if having the issue is nice for tracking that there's a need and WIP for this.

PR #68321 (2024-03-26): Driver sitronix CF1133 controller touchscreen
Add a new out of tree driver that support sitronix CF1133 controller touchscreen (I2C). Can be multitouch if needed.

Clément Dysli / clementdy@kickmaker.net

Fixes #68306
closed 2024-03-26 by fabiobaltieri