zephyrproject-rtos/zephyr #68313: async stm32 uart driver: uart_rx_buf_rsp function does not return errors

mkaranki, 2024-01-30:
**Describe the bug**

`uart_rx_buf_rsp` is [documented](https://docs.zephyrproject.org/latest/hardware/peripherals/uart.html#c.uart_rx_buf_rsp
) to return error `-EACCES` in case receiver is already disabled when the function is called.

stm32 uart driver is never returning any errors even if the rx has been disabled due to running out of buffers. See https://github.com/zephyrproject-rtos/zephyr/blob/b1a15718aa9990c8820c2f426d7a559b88f74e29/drivers/serial/uart_stm32.c#L1683-L1693

The problem in that is that at least Zephyr's async serial shell backend relies on getting `-EACCES` error if the rx has been disabled. See https://github.com/zephyrproject-rtos/zephyr/blob/b1a15718aa9990c8820c2f426d7a559b88f74e29/subsys/shell/backends/shell_uart.c#L468-L477

As a consequence, if no buffers available situation happens, then the async uart shell is dead and cannot receive anymore any characters.

**To Reproduce**
Steps to reproduce the behavior:
1. Have an stm32 board and async uart enabled
2. Enable shell with async serial backend
3. Have some high prio task, and spam the shell with some shell command so much that the uart runs out of buffers.
4. Notice the shell input is stuck. You could put some indicator to https://github.com/zephyrproject-rtos/zephyr/blob/d7d254140d02ef30821262853e46c794f7b515e5/drivers/serial/uart_stm32.c#L1489-L1497 to see that the cause of the jam is running out of buffers in the driver (yes, that happens).

**Expected behavior**
- `uart_rx_buf_rsp` returns the documented error code leading to the situation that async serial shell rx does not stop completely working if the call of `uart_rx_buf_rsp`  is done too late, or alternatively
- async serial shell does not rely on error return code of `uart_rx_buf_rsp` in this case but would handle `UART_RX_DISABLED` event.

Dropping some received characters on shell is more okay than stopping the whole shell rx functioning.

**Impact**
The shell is unusable and cannot be recovered easily without a reboot. The shell is used on the test automation, and in my application, TA can occasionally cause this to happen.

The issue can be set to be less likely to happen by tuning `CONFIG_SHELL_THREAD_PRIORITY_OVERRIDE` + `CONFIG_SHELL_THREAD_PRIORITY` values. By default shell has the lowest application priority so with it does not require very much else happening on the CPU to see the driver running out of buffers.

ajarmouni-st, 2024-02-14:
Hi @mkaranki, thank you for reporting this & for the detailed report. Indeed I was able to replicate the issue on an nucleo_l4r5zi.

mkaranki, 2024-02-14:
I am not sure is it worth to share this, but anyways, here I go... I have applied to following patch to make the issue less likely to happen:

```diff
diff --git a/drivers/serial/uart_stm32.c b/drivers/serial/uart_stm32.c
index 7ce07b930c9..b12889c380c 100644
--- a/drivers/serial/uart_stm32.c
+++ b/drivers/serial/uart_stm32.c
@@ -1475,7 +1475,7 @@ void uart_stm32_dma_rx_cb(const struct device *dma_dev, void *user_data,
                 * called in ISR context. So force the RX timeout
                 * to minimum value and let the RX timeout to do the job.
                 */
-               k_work_reschedule(&data->dma_rx.timeout_work, K_TICKS(1));
+               k_work_reschedule(&data->dma_rx.timeout_work, K_NO_WAIT);
        }
 }
 
@@ -1668,9 +1668,22 @@ static int uart_stm32_async_rx_buf_rsp(const struct device *dev, uint8_t *buf,
        struct uart_stm32_data *data = dev->data;
 
        LOG_DBG("replace buffer (%d)", len);
+
+       const unsigned int key = irq_lock();
+
+       if (data->rx_next_buffer != NULL) {
+               irq_unlock(key);
+               return -EBUSY;
+       } else if (!data->dma_rx.enabled) {
+               irq_unlock(key);
+               return -EACCES;
+       }
+
        data->rx_next_buffer = buf;
        data->rx_next_buffer_len = len;
 
+       irq_unlock(key);
+
        return 0;
 }
```

Although I can't get the shell anymore to jam, I think this might not be a complete fix since there's still a time window from the scheduling the timeout work to getting the flag data->dma_rx.enabled to false. During that time a call of `uart_stm32_async_rx_buf_rsp` would look like a successful one although it is not. Maybe intoducing a new flags is needed, idk.

ajarmouni-st, 2024-02-14:
> I am not sure is it worth to share this, but anyways, here I go... I have applied to following patch to make the issue less likely to happen

All contributions are welcome & appreciated :) will look into it & see if a permanent fix can be pushed this week.

PR #69038 (2024-02-26): drivers: serial: uart_stm32: fix for async_rx_buf_rsp
`uart_stm32_async_rx_buf_rsp()` does not return the necessary errors when `rx_next_buffer` is already set & when `async uart rx` is disabled.

One of the consequences of this is that the `async_read()` in `shell_uart` driver fails to call `uart_rx_enable()` when it's needed.

This patch was submitted by @mkaranki & fixes #68313 .