zephyrproject-rtos/zephyr #68329: doc: Documentation issue in 'samples/bluetooth/mtu_update/README'

CodingGhost, 2024-01-31:
on that Doc page the following is written:
"According to the Bluetooth specification, MTU is the maximum size of SDUs. However, in Zephyr, we can assume that it also represents the maximum size of the PDUs. Because, in Bluetooth LE, 3 unless we are using L2CAP dynamic channels, SDUs are not segmented."

but the MTU size does not have to be the same as the PDU size?
I can set a MTU of 1024 bytes with a PDU size of 251 bytes and then call bt_gatt_write with 1024 bytes of data. this will lead to the 1024 byte packet to get segmented.
so SDUs can be segmented, or not?
**Environment**

* Page: `samples/bluetooth/mtu_update/README`
* Version: main
* SHA-1: 43144808263ed65aa0ba45781aa88f6fcf4c49c5

jori-nordic, 2024-01-31:
SDUs can be segmented, just like you describe.
We are just saying that:
An SDU is always < MTU
a PDU is also always < MTU

Not that SDU == PDU == MTU

CodingGhost, 2024-01-31:
But it also says that SDUs are not Segmented. and the PDU cannot reach a value of more than 251 I think? the text suggests that a PDUs maximum size is equal to the maximum size of the MTU and SDU

CodingGhost, 2024-01-31:
could it be that it already changed? my code does not use dynamic channels but the SDUs do get segmented. I can do a gatt_write with 512 bytes of data at once, but it will be segmented into 2x251 bytes.

jori-nordic, 2024-01-31:
Interesting, I'll check then. Could you post the config you're using with the sample?

CodingGhost, 2024-01-31:
Im not using this sample but my own code. the sample should work in the same way though.
this is the relevant config options im using:
CONFIG_BT_USER_DATA_LEN_UPDATE=y
CONFIG_BT_CTLR_DATA_LENGTH_MAX=251
CONFIG_BT_L2CAP_TX_MTU=1024
CONFIG_BT_BUF_ACL_TX_COUNT=7 
CONFIG_BT_BUF_ACL_RX_SIZE=1028
CONFIG_BT_BUF_ACL_TX_SIZE=1028

alwa-nordic, 2024-02-01:
> could it be that it already changed? my code does not use dynamic channels but the SDUs do get segmented. I can do a gatt_write with 512 bytes of data at once, but it will be segmented into 2x251 bytes.

You are seeing a GATT long write (Core 5.4 3 G 4.9.4). This is fragmentation at ATT level. It has nothing to do with SDU segmentation. `bt_gatt_write` does this automagically for you. This btw surprises everyone using our API, and possibly broken or under-specified in combination with EATT. It's not a good feature.

alwa-nordic, 2024-02-01:
I just want to make sure I don't kill this discussion with my previous answer. Whatever the conclusion is here, I think we want to make the README as good as possible. Any confusion experienced by our users is a valid concern and likely experienced by many for each reported issue. It means the text is less-than-perfect.

CodingGhost, 2024-02-03:
thanks, this clears things up for me! if this information would be present in the readme somehow I think I would not have opened this issue, so maybe put it there (and in the gatt_write doc)?

jori-nordic, 2024-07-04:
Hi @CodingGhost . @theob-pro recently updated the documentation for L2CAP.
Could you check it out and see if it clears up the confusion? If you have any questions still unanswered after reading, please comment again, and we'll try to fix the docs once more :+1: 

https://docs.zephyrproject.org/latest/connectivity/bluetooth/bluetooth-le-host.html#l2cap

kartben, 2024-12-18:
Going to assume that the latest comment from @jori-nordic addresses the reported issue as the original author hasn't commented further.

closed 2024-12-18 by kartben