zephyrproject-rtos/zephyr #68336: Bluetooth: Controller: Connection Interval seems wrong

CodingGhost, 2024-01-31:
using the Zephyr Controller Implementation, the connection interval seems to be double of what is negotioated. This can be seen with wireshark, I attached the logs. when using the Softdevice controller it is behaving like expected.
Im not sure if my Wireshark maybe skews the results but as it measures correctly with the softdevice it seems to be working. It would be Interesting if someone could verify this behaviour.
I am currently using NCS 2.5, but as I tried with both Controllers, this shouldnt be a NCS only Issue.
at this stage I would first like to know if anyone can reproduce this or if its specific to my setup.
@cvinayak 
[wireshark.zip](https://github.com/zephyrproject-rtos/zephyr/files/14110188/wireshark.zip)

kartben, 2024-01-31:
+ @theob-pro who authored this code sample.

CodingGhost, 2024-01-31:
> + @theob-pro who authored this code sample.

what code sample do you mean?

kartben, 2024-01-31:
sorry, posted on the wrong issue!

cvinayak, 2024-01-31:
tested using shell application, and i do not see any issue using 7.5 ms conn interval either with coded phy supported (but the slot reservation turned off):
[image]
Timing seems ok too:
[image]

CodingGhost, 2024-01-31:
I will try to create a minimal sample to demonstrate it and also try to verify that it works correct with the console application. Maybe this is harder to replicate than I thought. What I know for sure right now is that with LL_SW_SPLIT it behaves "wrong" and then I only set LL_SW_SPLIT=n and everything works as it should. I will update this issue with my findings once I have some more time todo specific testing.

henrikbrixandersen, 2024-02-06:
Thank you for reporting this. However - unless you are able to reproduce this issue with upstream Zephyr `main` - please report issues with the nRF Connect SDK (NCS) on the [Nordic Semiconductor DevZone](https://devzone.nordicsemi.com/).

CC: @carlescufi

CodingGhost, 2024-02-07:
I thought that it would fit better here, because its about the Zephyr BLE Controller implementation and not about the softdevice, but If you think Its better to put it in the devzone I will do that. 
I will get around to more testing in about two weeks, but Im pretty sure it has something todo with some kind of time reservation for the scanning. If I start a scan with coded phy window and interval set in the bt_le_scan_param and then stop it and connect, the connection interval is wrong. It doesnt even matter if coded phy scanning was enabled, its just about if the values for window and interval have been set. and the scan was stopped before connecting and never restarted, so something is going on there. But as I said, i will put together a sample to demonstrate it once I have some spare time.

CodingGhost, 2024-02-21:
as Vidar Berg pointed out in his analysis on the Devzone post, the Controller is skipping every other event. The cause is unknown though. Is there any log that I can enable where I would see this skipping happen? Also, could this all be because of a misalignment of event scheduling? like somehow the events are scheduled for the same time? as far as I know, the LL connection update has a offset field for exactly this purpose, but neither the Softdevice nor the Opensource Controller are using this feature. Maybe this would solve the problem..?

cvinayak, 2024-02-26:
> he LL connection update has a offset field for exactly this purpose, but neither the Softdevice nor the Opensource Controller are using this feature. Maybe this would solve the problem..?

You could try the Zephyr Controller in Zephyr v3.3 release with `CONFIG_BT_LL_SW_LLCP_LEGACY=y`, i believe the older implementation had code to find offset and use it in connection update. If you do not see offset used in sniffer log with v3.3 release, do revert back and i can check if I have any local implementation that does it... do post sniffer logs here anyway.

CodingGhost, 2024-02-26:
[wireshark_wrongintervaldemo.pcapng.txt](https://github.com/zephyrproject-rtos/zephyr/files/14404583/wireshark_wrongintervaldemo.pcapng.txt)
I tested it as you said, and it really does use an offset! But unfortunately it did not change anything, it was still reproducable within 2 tries. What I think is a little bit odd, because I would have assumed that with the offset there should be no collisions directly after the negotiation, but eventually a long time later when the system clocks have drifted enough. But what you can see in the sniffer log is, that the problem is there from the first packet( packet no. 11564) with the updated interval onwards.

I cant upload .pcapng files, so i just added .txt at the end, so just rename it.

cvinayak, 2024-02-26:
> eventually a long time later when the system clocks have drifted enough

If I remember correct there was some automatic detection of drift and the connection parameter request was auto-initiated, but I do not think it is there anymore in any release of Zephyr.

Alternatively, applications can periodically initiate connection parameter request to mitigate the eventual overlap... may be you can try that, say a connection parameter update every 30 seconds or so.

CodingGhost, 2024-02-26:
yeah, that could work, BUT in order for that to work, it first has to work at all after the update. Currently as you see in the log, there is a offset of 5msec applied, but they seem to still overlapp / Controller is still skipping events right after the update.

cvinayak, 2024-02-27:
> it first has to work at all after the update

In your sniffer trace your Peripheral (I think a non Zephyr Controller) is not using the LL Connection Parameter Request (but L2CAP Connection Parameter Update), hence the Central device (I think a Zephyr Controller) is not provided with the scheduling offset hints.

Anyway, we are diverging away from the fact that Central clock would drift and that any measure taken will eventually lead to the Central connection events drifting into and overlapping with other roles in the Peripheral. LL Connection Parameter Request Control Procedure would to some extent help as long as all devices in the topology co-operate in the scheduling.

CodingGhost, 2024-02-27:
Sorry, I forgot to also apply this config to the Peripheral. Now it looks different, but somehow the connection param request now gets rejected. This is how the request looks:
[image]

By the way, If I get it to work like that, and just do a update every 30sec or so, how will that work in a current version of Zephyr? Why was this feature even removed?