zephyrproject-rtos/zephyr #68339: up_squared: tests/lib/c_lib/thrd/ fail

KamilxPaszkiet, 2024-01-31:
**Describe the bug**
These 6 tests fail:

```
tests/lib/c_lib/thrd/libraries.libc.c11_threads.minimal on up_squared
tests/lib/c_lib/thrd/libraries.libc.c11_threads.picolibc on up_squared
tests/lib/c_lib/thrd/libraries.libc.c11_threads.picolibc.module on up_squared
tests/lib/c_lib/thrd/libraries.libc.c11_threads.picolibc.notls on up_squared
tests/lib/c_lib/thrd/libraries.libc.c11_threads.newlib on up_squared
tests/lib/c_lib/thrd/libraries.libc.c11_threads.newlib_nano on up_squared
```

**To Reproduce**
`twister --device-testing --hardware-map up2.map -T tests/lib/c_lib/thrd/ -vv`

**Logs and console output**
The log is too long to be readable in github. I will paste part of the log from one of the tests.
```
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v3.5.0-5059-gb3e67cf2a62b ***
DEBUG   - DEVICE: E: Page fault at address 0x150 (error code 0x2)
DEBUG   - DEVICE: Running TESTSUITE libc_cnd
DEBUG   - DEVICE: E: Linear address not present in page tables
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: E: Access violation: supervisor thread not allowed to write
DEBUG   - DEVICE: START - test_cnd_broadcast_wait
DEBUG   - DEVICE: E: PTE: not present
DEBUG   - DEVICE: ASSERTION FAIL [__attr_is_initialized(attr)] @ WEST_TOPDIR/zephyr/lib/posix/pthread.c:779
DEBUG   - DEVICE: E: RAX: 0x0000000000000000 RBX: 0x0000000000006d50 RCX: 0x0000000000000000 RDX: 0x0000000000000022
DEBUG   - DEVICE: E: RAX: 0x0000000000000004 RBX: 0x0000000000127590 RCX: 0x0000000000000001 RDX: 0x0000000000000000
DEBUG   - DEVICE: E: RSI: 0x0000000000000000 RDI: 0x0000000000006d50 RBP: 0x0000000000006d40 RSP: 0x0000000000bc8fc8
DEBUG   - DEVICE: E: RSI: 0x000000000000030b RDI: 0x000000000011f425 RBP: 0x0000000000cd4ec0 RSP: 0x0000000000cd4ea8
DEBUG   - DEVICE: E:  R8: 0x0000000000000000  R9: 0x0000000000000000 R10: 0x0000000000d04d40 R11: 0x0000000000008248
DEBUG   - DEVICE: E:  R8: 0x0000000000000007  R9: 0x0000000000cd4c00 R10: 0x00000000ffffffff R11: 0x0000000000cd4d10
DEBUG   - DEVICE: E: R12: 0x0000000000000000 R13: 0x0000000000000022 R14: 0x0000000000000000 R15: 0x0000000000000000
DEBUG   - DEVICE: E: R12: 0x0000000000000000 R13: 0x0000000000100407 R14: 0x0000000000126010 R15: 0x0000000000127590
DEBUG   - DEVICE: E: RSP: 0x0000000000bc8fc8 RFLAGS: 0x0000000000010246 CS: 0x0018 CR3: 0x0000000000f07000
DEBUG   - DEVICE: E: RSP: 0x0000000000cd4ea8 RFLAGS: 0x0000000000000246 CS: 0x0018 CR3: 0x0000000000f07000
DEBUG   - DEVICE: E: RIP: 0x000000000000163e
DEBUG   - DEVICE: E: RIP: 0x0000000000103687
DEBUG   - DEVICE: E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 1
DEBUG   - DEVICE: E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
DEBUG   - DEVICE: E: Current thread: (nil) (unknown)
DEBUG   - DEVICE: E: Current thread: 0x128a40 (test_cnd_broadcast_wait)
DEBUG   - DEVICE: Caught system error -- reason 0 1
DEBUG   - DEVICE: Caught system error -- reason 4 0
DEBUG   - DEVICE: Fatal error expected as part of test case.
DEBUG   - DEVICE: Fatal error was unexpected, aborting...
DEBUG   - DEVICE: E: Page fault at address 0x18 (error code 0x0)
DEBUG   - DEVICE: E: Linear address not present in page tables
DEBUG   - DEVICE: E: Access violation: supervisor thread not allowed to read
DEBUG   - DEVICE: E: PTE: not present
DEBUG   - DEVICE: E: RAX: 0x0000000000000001 RBX: 0x0000000000000000 RCX: 0x0000000000000001 RDX: 0x0000000000000001
DEBUG   - DEVICE: E: RSI: 0x0000000000006d50 RDI: 0x00000000005b2500 RBP: 0x0000000000006cf0 RSP: 0x0000000000006cd0
DEBUG   - DEVICE: E:  R8: 0x0000000000000000  R9: 0x0000000000006a00 R10: 0x00000000ffffffff R11: 0x0000000000006b10
DEBUG   - DEVICE: E: R12: 0x0000000000000247 R13: 0x0000000000000247 R14: 0x0000000000000000 R15: 0x0000000000000000
DEBUG   - DEVICE: E: RSP: 0x0000000000006cd0 RFLAGS: 0x0000000000010002 CS: 0x0018 CR3: 0x0000000000f07000
DEBUG   - DEVICE: E: RIP: 0x0000000000110e8d
DEBUG   - DEVICE: E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 1
```

**Environment:**
 - OS: Linux
 - Toolchain: zephyr-sdk-0.16.3
 - zephyr-v3.5.0-5059-gb3e67cf2a62b

aescolar, 2024-01-31:
Removing myself as assignee as this is not my area.
CC test author @cfriedt  and Intel x86 guys @jhedberg @tbursztyka @laurenmurphyx64

aescolar, 2024-01-31:
Is this related to the failure in #68115? CC @dcpleung

KamilxPaszkiet, 2024-01-31:
> Is this related to the failure in #68115? CC @dcpleung

Yep, looks like the same problem. I just checked it a moment ago.
I will add this information to that thread.

closed 2024-01-31 by KamilxPaszkiet

cfriedt, 2024-01-31:
Assigned Daniel, since iirc there was a similar bug covered yesterday during triage.