zephyrproject-rtos/zephyr #68341: ARC SMP mode doesn't initialize MPU to all cores

avivbarkan, 2024-01-31:
**Describe the bug**
Running ARC HS48x, dual core with SMP and multi-threading enabled.
It seems that the function _arc_mpu_init_ being called for core-0 only.
That means that for core-1 the MPU remains disabled the default access attributes are being used.

This caused a problem not only when accessing a forbidden area but also when the attributes of ranges being changed. 
For example if the NV bit is set.

In my case we change a memory are to be un-cached area. In this the range was set as un-cached for core-0 and cached for core-1

ruuddw, 2024-02-05:
Acknowledged, so far we didn't support configurations with MPU.
What MPU features are most important to you? The cache attributes?
I'll see what we can do to add MPU support for SMP, would you be able/willing to prepare a PR for this?

avivbarkan, 2024-02-05:
Hi,
What we are trying to get is the same MPU configuration for both cores in SMP mode.
I'm referring to the configuration that being done as part of the setting defined in the _mpu_regions_  (_arc_mpu_regions.c_).

I don't really have a PR for it. We just worked around it by adding an additional call to _arc_mpu_init_ in _z_arc_slave_start_ function (_arc_smp.c_)