zephyrproject-rtos/zephyr #68343: Function sys_clock_tick_get return wrong value in ARC SMP mode

avivbarkan, 2024-01-31:
**Describe the bug**
When calling the _sys_clock_tick_get_ function on ARC SMP mode, the return value can be wrong.
Configuration:
ARC HS48x, dual core 
SMP and multi thread configuration enabled.

In this configuration the _sys_clock_tick_get_ function will eventually call the _elapsed_ function.
Since this function using ARC timer0 which exist on each cores, it should be used on a **single** core system only. 

On multi-core the "gfrc" timer should be used (_z_arc_connect_gfrc_read_).

henrikbrixandersen, 2024-02-05:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

avivbarkan, 2024-02-06:
> Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

Done, see above

avivbarkan, 2024-02-06:
> Can you check if SMP_TIMER_DRIVER is defined? elapsed shouldn't be called in the SMP case: `#if SMP_TIMER_DRIVER cyc = (z_arc_connect_gfrc_read() - last_time); #else cyc = elapsed() + cycle_count - announced_cycles; #endif`

Hi,
Yes, it is enable since both CONFIG_SMP and  CONFIG_MP_MAX_NUM_CPUS is set to 2.

The function _elapsed_ are called from the  _sys_clock_tick_get_ (_timeout.c_)

ruuddw, 2024-02-06:
ok, we'll need to look deeper into this., I still don't understand the problem. The ARC timer driver does use gfrc, but it seems the issue you report is in the generic kernel timer code and the curr_tick use?
Sharing more details to reproduce the issue e.g. on nSIM would help.

avivbarkan, 2024-02-06:
I will try to get more info, but as I understood the root-cause, is a race on the _curr_tick_ updating.
Our workaround was removing the use of this parameter. We can do that since the gfrc counter is a 64b counter. So there is no chance of wraparound of the counter

ruuddw, 2024-02-06:
I don't see anything obviously suspicious, including the sys_clock_announce() that should be called with 0 ticks on SMP CPUs that get a timer interrupt that was already announced by another CPU.
To further check, some log files and/or steps to reproduce are needed.

avivbarkan, 2024-02-06:
OK, I will try to reproduce and add some results.
2 point's that I didn't mentioned: a) I'm using the 3.5 version b) The _CONFIG_SYS_CLOCK_TICKS_PER_SEC_ is set to 100

avivbarkan, 2024-02-07:
Please see our workaround code. Our suspicion was that it's related to the global parameter _curr_tick_ .
You can see that the code avoid using it. However, it is possible that we are looking at the wrong place.
Please advise, 
[SMP_timer_fix.patch](https://github.com/zephyrproject-rtos/zephyr/files/14194751/SMP_timer_fix.patch)

ruuddw, 2024-02-07:
Thanks, the patch clarifies your workaround - it looks like I expected from the earlier description.
However, I still don't understand the problem this workaround solves. The patch may change semantics to something you expect but wasn't originally intended.
A small example and logs that can be used for reproducing the issue and further analysis are really needed.

closed 2024-07-25 by github-actions[bot]