zephyrproject-rtos/zephyr #68345: `FLASH_AREA_1_SIZE` is omited when `MCUBOOT_IMAGE_NUMBER=1`

ithinuel, 2024-01-31:
**Describe the bug**
`west build -p -b lpcxpresso55s69_ns samples/hello_world/ -D CONFIG_TFM_MCUBOOT_IMAGE_NUMBER=1` fails to build with

```
Exception: The script was probably invoked manually with having certain macros nested in flash_layouts.h.
Please revisit the flash_layout.h file and hardcode values for the (NON-)SECURE_IMAGE_OFFSET and (NON-)SECURE_IMAGE_MAX_SIZE macros
ninja: build stopped: subcommand failed.
```

This is caused by `FLASH_AREA_1_SIZE` being left undefined after macro expansion expansion.
I tracked it down to this being omited when `MCUBOOT_IMAGE_NUMBER` is not defined or set to 1 [here](https://github.com/zephyrproject-rtos/trusted-firmware-m/blob/main/platform/ext/target/nxp/lpcxpresso55s69/partition/flash_layout.h#L99-L120).

**To Reproduce**
Steps to reproduce the behavior:
1. run `west build -p -b lpcxpresso55s69_ns samples/hello_world/ -D CONFIG_TFM_MCUBOOT_IMAGE_NUMBER=1` from your zephyr work directory.
4. See error

**Expected behavior**
The image to build with secure and non-secure being signed as a single image.

**Impact**
It prevents the usecase where tfm & ns application are atomically updated.

**Environment (please complete the following information):**
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: 60c58fe91801e69aa1d443b5196b934596f6f3fd

**Question**
Is this an issue in tf-m's port to lpcxpresso55s69 or should this be overridden by a zephyr config/dts?

butok, 2024-01-31:
So I have tried to compile it for other platforms with CONFIG_TFM_MCUBOOT_IMAGE_NUMBER=1. And it is failed with the same error. 
Also the older TFM build is failed too.

So this is the common TFM issue.

ithinuel, 2024-01-31:
I checked with our TF-M team and AFAIU, platforms are not tested with both.
They would typically only support one of them and most recent platform only support 2.

So the issue should probably be that it fails after trying to compile while the board/port should be what sets this configuration and it should fail if a user tries to set it manually instead.

ithinuel, 2024-02-06:
Note: this is not specific to this target. 
`west build -p -b mps2_an521_ns samples/hello_world/ -D CONFIG_TFM_MCUBOOT_IMAGE_NUMBER=1` also triggers the same failure.

PR #74363 (2024-07-13): modules: tfm: fix `TFM_MCUBOOT_IMAGE_NUMBER == 1`
Fix builds with `CONFIG_TFM_MCUBOOT_IMAGE_NUMBER=1`. The merged binary should be signed with the same argumements as a secure build, not a non-secure build, except with a different layout file generated by the TF-M build system.

Fixes #68345.

Note that the test case provided works correctly for 3 boards, but results in `imgtool` overflows (which are marked as "skipped" tests) for the following platforms:
 * bl5340_dvk/nrf5340/cpuapp/ns
 * stm32l562e_dk/stm32l562xx/ns
 * mps2/an521/cpu0/ns

Resolving these will require delving into the TF-M re [... cut]
commit 1bbfbef8 (2024-07-13): modules: tfm: fix `TFM_MCUBOOT_IMAGE_NUMBER == 1`

Fix builds with `CONFIG_TFM_MCUBOOT_IMAGE_NUMBER=1`. The merged binary
should be signed with the same argumements as a secure build, not a
non-secure build, except with a different layout file generated by the
TF-M build system.

Fixes #68345.
closed 2024-07-13 by nashif