zephyrproject-rtos/zephyr #68352: net: socket: recvmsg() doesn't update msg_controllen accordingly

axelnxp, 2024-01-31:
**Describe the bug**

When returning from `recvmsg`, even when there is not control data, `msg_controllen` field is still set to its initial value.
According to manpage of `recvmsg`:

`
The field msg_control, which has length msg_controllen, points to a buffer for other protocol  control-related
messages  or miscellaneous ancillary data.  When recvmsg() is called, msg_controllen should contain the length
of the available buffer in msg_control; upon return from a successful call it will contain the length  of  the
control message sequence.
`

Currently, the second part of this statement is not fulfilled.
I've been facing an issue with our Matter application, a function tries to parse control data after returning from `recvmsg`, using `CMSG_FIRSTHDR` and `CMSG_NXTHDR` macros, but as the `msg_controllen` field is incorrect, it is stuck in a for loop.
`csmg_len` is set to 0 (used by `CMSG_NXTHDR` to determine next control message).

**To Reproduce**
One can reuse the matter function where the issue is observed: 
https://github.com/project-chip/connectedhomeip/blob/b1bf11eae359e6744ba571676f210617802dab9c/src/inet/UDPEndPointImplSockets.cpp#L571
And make sure no control data is received.

**Expected behavior**
`msg_controllen` field shall be updated by `recvmsg` according to the control data received.

**Impact**
Impact is high, in the current state, any Matter application over Zephyr would be impacted, likely unable to even perform a commissioning.

**Logs and console output**
[image]

axelnxp, 2024-01-31:
cc @jukkar

jukkar, 2024-01-31:
@axelnxp please take a look for the proposed fix at #68376 if it works for you

axelnxp, 2024-01-31:
Thanks @jukkar for the quick fix proposal.
It works on my side, I added a comment in the PR.

commit 692ebf40 (2024-02-02): net: sockets: Update msg_controllen in recvmsg properly

If recvmsg() does not update control data, then it must
set msg_controllen length to 0 so that the caller can
understand this.

Fixes #68352
closed 2024-02-02 by dleach02