zephyrproject-rtos/zephyr #68360: Enhancing Application Versioning Feature

hieuv, 2024-01-31:
**Is your enhancement proposal related to a problem? Please describe.**
The Application Version Management feature generate app_version.h from the input VERSION file. However, not all fields of the VERSION file are easily accessible from app_version.h.

**Describe the solution you'd like**
app_version.h should also include a VERSION_TWEAK and an EXTRA_VERSION macro to access the value of those fields.

**Describe alternatives you've considered**
Deriving VERSION_TWEAK and EXTRA_VERSION from the existing APPVERSION and APP_VERSION_STRING macros are possible, but that is just a workaround and isn't very clean.
An application developer can also use CMake to generate new preprocessors to access VERSION_TWEAK and EXTRA_VERSION, but still, direct support from the build system is ideal.

**Additional context**
N/A

kartben, 2024-01-31:
cc @nordicjm

nordicjm, 2024-01-31:
There is this https://github.com/zephyrproject-rtos/zephyr/pull/68160 please check it matches the format you want

commit a3063978 (2024-02-01): version: cmake: kconfig: introduce extra <type>_VERSION_<x>_STRING

Fixes: #68360

This commit introduces <type>_VERSION_TWEAK_STRING which includes
the tweak field in the string, but without the extra version.
This format is used by MCUboot / imgtool, and thus makes it easier to
align code to the format used by MCUboot.

This commit also introduces <type>_VERSION_EXTENDED_STRING which
includes the tweak field in the string in addition to the extra version
field.

The new defines / variables is available in code, CMake, and Kconfig,
and it defined for KERNEL, APP, and custom types, such as MCU [... cut]