zephyrproject-rtos/zephyr #68365: ESP32C3: `assert sch_arb.c 890, param 00000000 00000000` during BLE connection set-up or data transfer over BLE [zephyr-v3.5.0]

abdulhaseebayub, 2024-01-31:
**Describe the bug**
I have a BLE Peripheral based on Espressif `esp32c3` running Zephyr `v3.5.0`. Sometimes, either when a Central is being connected or data is being transferred from the Central to the Peripheral, the module gets stuck. The log message shows the following error, followed by a reset triggered by the watchdog timer:
`assert sch_arb.c 890, param 00000000 00000000`

There are no other debug log messages indicating an issue either before or after this message. Since I don't have access to the Espressif library where this message comes from, I'm not sure what it means. I'm unable to find a case where I could reproduce this issue every time.

Previously this issue came up when I was testing how many different centrals can be bonded with my peripheral (with `CONFIG_BT_MAX_PAIRED=40`). At that time it was recommended to apply [this fix](https://github.com/espressif/esp-idf/issues/9238), which resolved the issue for that use case.

**Expected behavior**
The module should not get stuck.

**Impact**
Showstopper, since my application gets interrupted during critical operations.

**Environment (please complete the following information):**
- Devcontainer based on image: zephyrprojectrtos/zephyr-build:v0.26.6
- Toolchain: Zephyr SDK 0.16.4
- Zephyr branch: zephyr-v3.5.0
- Target: esp32c3 (DevkitM-1)

nordicjm, 2024-02-01:
@nashif since this file and error message seems to have nothing to do with zephyr (i.e. binary blob?) should this issue be closed?

abdulhaseebayub, 2024-03-01:
> @nashif since this file and error message seems to have nothing to do with zephyr (i.e. binary blob?) should this issue be closed?

Correct, testing with the new Espressif HAL v5.1 seems to have resolved this issue (PR still open: https://github.com/zephyrproject-rtos/zephyr/pull/68760).

closed 2024-03-01 by abdulhaseebayub