zephyrproject-rtos/zephyr #68369: samples: build error

ottojo, 2024-01-31:
**Describe the bug**
Starting with 9dcbfa8bf23fb73a0d4ab14f93e1f14fb07022c8, the following build failure occurs
```
/home/jonas/Workspace/zephyr_sandbox_2/zephyr/subsys/llext/shell.c: In function 'cmd_llext_load_hex':
/home/jonas/Workspace/zephyr_sandbox_2/zephyr/subsys/llext/shell.c:124:26: error: implicit declaration of function 'strnlen'; did you mean 'strlen'? [-Werror=implicit-function-declaration]
  124 |         size_t hex_len = strnlen(argv[2], CONFIG_LLEXT_SHELL_MAX_SIZE*2+1);
      |                          ^~~~~~~
      |                          strlen
```

**To Reproduce**
Steps to reproduce the behavior:
```
west twister --build-only --inline-logs  --platform teensy41 -s samples/subsys/llext/shell_loader/sample.llext.shell -v
```

**Expected behavior**
No build error

**Impact**
Caused CI failure for OOT board repo, which builds samples against custom board nightly: https://github.com/ottojo/otterpill_zephyr/actions/runs/7674769452/job/20919951824

**Environment (please complete the following information):**
 - OS: Linux 6.6.12-1-MANJARO
 - Toolchain: zephyr sdk 0.16.1
 - Bisected first bad commit: 9dcbfa8bf23fb73a0d4ab14f93e1f14fb07022c8

I see there is discussion around some posix stuff already, but i'm not sure if this is the same problem, as this seems to occur with default configuration?

teburd, 2024-01-31:
Thank you for the report, will take a look

ottojo, 2024-01-31:
It seems the outdated SDK seems to be the culprit here... Locally i had SDK 16.1 installed, updating to the latest 16.4 solved the issue locally, so it seems like the problem was indeed fixed between SDK releases 16.1 and 16.2.
Thanks for the suggestion, and sorry for the false alarm :smile:

closed 2024-01-31 by ottojo