zephyrproject-rtos/zephyr #68398: Sample:LVGL widgets emulation crashes with segfault

everedero, 2024-02-01:
**Describe the bug**
LVGL demo app "Widgets" crashes with a segmentation fault when using native_sim_64 emulation.
It runs without crashing on my hardware platform (https://github.com/everedero/board_st_mb1283).
The other samples are working correctly, both on target and in native_sim_64 emulation.

**To Reproduce**
Steps to reproduce the behavior:
1. cd ~/zephyrproject/zephyr/samples/modules/lvgl
2. west build -b native_sim_64 -p always ./demos -DCONFIG_LV_Z_DEMO_WIDGETS=y
3. ./build/zephyr/zephyr.exe

```
uart connected to pseudotty: /dev/pts/4
[00:00:00.000,000] <inf> sdl_input: Init 'input-sdl-touch' device
*** Booting Zephyr OS build zephyr-v3.5.0-5156-g417c18c974da ***
Erreur de segmentation (core dumped)
```

**Impact**
Not a major issue since the other demos are working.

**Environment**
 - OS: GNU/Linux (LMDE 5)
 - Toolchain: Zephyr SDK
 - LVGL module #2b76c641, Zephyr main #417c18c974

kartben, 2024-02-01:
I can confirm this is happening for me as well on main / 5957da58301bb05d58e5bbaee98b22be55e937d2

aescolar, 2024-02-01:
The segfault is caused here:
``` 
==871774==    at 0x416D36: lv_obj_mark_layout_as_dirty (lv_obj_pos.c:290)
==871774==    by 0x415E47: lv_obj_class_init_obj (lv_obj_class.c:106)
==871774==    by 0x43F901: lv_checkbox_create (lv_checkbox.c:60)
==871774==    by 0x407EBF: shop_create (lv_demo_widgets.c:971)
==871774==    by 0x407EBF: lv_demo_widgets (lv_demo_widgets.c:196)
``` 
Where in the 4th call to lv_checkbox_create in https://github.com/zephyrproject-rtos/lvgl/blob/zephyr/src/widgets/lv_checkbox.c#L59
```
sys_heap_alloc() at zephyr/lib/heap/heap.c:274 0x408662	
lvgl_malloc() at zephyr/modules/lvgl/lvgl_mem.c:23 0x447ded	
lv_mem_alloc() at modules/lib/gui/lvgl/src/misc/lv_mem.c:136 0x439eee	
lv_obj_class_create_obj() at modules/lib/gui/lvgl/src/core/lv_obj_class.c:47 0x415ca7	
lv_checkbox_create() at modules/lib/gui/lvgl/src/widgets/lv_checkbox.c:59 0x43f8f5	
``` 
The `sys_heap_alloc()` returns null as it seems to run out of heap, but the failure is not properly handled by `lv_checkbox_create()` which just calls into `lv_obj_class_init_obj()` with that null pointer causing the segfault.

As an immediate quick fix you can set `-DCONFIG_LV_Z_MEM_POOL_SIZE=128000` 

I think it would be good if:
* somebody increased the heap size in this sample config (at least for this board) so this does not break by default
* the lvgl handled more gracefully running out of heap

CC @faxe1008 , @brgl , @pdgendt  , @danieldegrasse 

As this is an issue in lvgl and its sample I remove myself as asignee.

kartben, 2024-02-01:
Thanks for the detailed investigation @aescolar!

faxe1008, 2024-02-05:
@aescolar @kartben @everedero Thanks for the investigation. I'll raise a smol PR upstream tomorrow, but I suspect that is not the only case where LVGL does not OOM gracefully.

PR #69097 (2024-02-16): samples/modules/lvgl/demos: Increase heap for native 64 targets
For these targets, the widgets demo run out of heap and crashes. Let's just increase the heap size.

Fixes: #68398
closed 2024-02-16 by henrikbrixandersen