zephyrproject-rtos/zephyr #68401: DT_N_INST_0_arm_v7m_nvic_P_arm_num_irq_priority_bits' undeclared

akhil-bhat, 2024-02-01:
i am getting error 'DT_N_INST_0_arm_v7m_nvic_P_arm_num_irq_priority_bits' undeclared (first use in this function) for the board s32g274aabvuc . I am currently trying to compile and build and got this build error.
what should i add into devicetree and how to add?
this is my board devicetree

```
/dts-v1/;
#include <arm/nxp/nxp_s32g274aabvuc_a53.dtsi>
#include "s32g274aabvuc_a53.dtsi"

/ {
model = "NXP S32G374AABVUC on Cortex-A53 cores"; //changed
compatible = "nxp,s32g274"; //changed

chosen {
	zephyr,sram = &sram0;
	zephyr,canbus = &can0;
	zephyr,console = &uart0;
	zephyr,shell-uart = &uart0;
};

aliases {
	watchdog0 = &swt0;
};

};

&mru0 {
rx-channels = <1>;
status = "okay";
};

&enetc_psi0 {
mboxes = <&mru0 0>;
mbox-names = "rx";
};
```

erwango, 2024-02-01:
@henrikbrixandersen Looks like a NXP target.

henrikbrixandersen, 2024-02-01:
> @henrikbrixandersen Looks like a NXP target.

My bad, sorry.

slim-shaddy-shades, 2024-02-02:
https://github.com/zephyrproject-rtos/zephyr/issues/38398

isn't this similar issue??

slim-shaddy-shades, 2024-02-02:
okkkk