zephyrproject-rtos/zephyr #68408: drivers: ili9341 fails to build

everedero, 2024-02-01:
**Describe the bug**
Board with ili9341 driver fails to build since recent changes.

**To Reproduce**
Steps to reproduce the behavior:
1. Find a board with an ili9341 driver
2. west build -b $BOARD -p always ./lvgl

Result: lots of devicetree complaints like this:
```
zephyr_st25/build/zephyr/include/generated/devicetree_generated.h:18262:53: error: 'DT_N_S_soc_S_spi_40003800_S_ili9341_0_P_width' undeclared here (not in a function); did you mean 'DT_N_S_soc_S_spi_40003800_S_ili9341_0_P_reg'?
18262 | #define DT_CHOSEN_zephyr_display                    DT_N_S_soc_S_spi_40003800_S_ili9341_0
      |                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/eve/zephyrproject/zephyr/include/zephyr/devicetree.h:4452:29: note: in definition of macro 'DT_CAT3'
 4452 | #define DT_CAT3(a1, a2, a3) a1 ## a2 ## a3
```
Git bisect dates this back to:
[571de47e1645622d16b20584d6068980e8622463] drivers: mipi_dbi: add SPI based MIPI DBI mode C driver

**Environment:**
- OS: GNU/Linux (LMDE 5)
- Toolchain: Zephyr SDK
- Zephyr main #417c18c974

kartben, 2024-02-01:
I can't reproduce? Did you run `west update`? Or maybe share which board specifically is causing issues

```
 west build -b stm32f429i_disc1 -p always samples/subsys/display/lvgl
```

everedero, 2024-02-01:
It’s an out-of-tree board : https://github.com/everedero/board_st_mb1283
west update did not do much.
There may something wrong with my config, it is just weird that it breaks now

kartben, 2024-02-01:
> It’s an out-of-tree board : https://github.com/everedero/board_st_mb1283 west update did not do much. There may something wrong with my config, it is just weird that it breaks now

Ah! Based on the changes in https://github.com/zephyrproject-rtos/zephyr/pull/64587/commits/1cfed06929087701fec8edcabbc78f92ddcad926 I guess it needs updating?

@danieldegrasse It looks like a note in the release notes and migration guides is in order?

everedero, 2024-02-01:
OK, that is what I was missing! Thanks

kartben, 2024-02-01:
I am tempted to leave this open as a placeholder for the work that needs to happen for 3.6 re: updating release notes / migration guide. @danieldegrasse, agreed?

henrikbrixandersen, 2024-02-01:
> I am tempted to leave this open as a placeholder for the work that needs to happen for 3.6 re: updating release notes / migration guide. @danieldegrasse, agreed?

It's already there? https://docs.zephyrproject.org/latest/releases/migration-guide-3.6.html#device-drivers-and-device-tree

kartben, 2024-02-01:
> It's already there? https://docs.zephyrproject.org/latest/releases/migration-guide-3.6.html#device-drivers-and-device-tree

Duh. Sorry! Completely missed it when looking at the original PR. Sincere apologies for the noise.

closed 2024-02-01 by kartben

erwango, 2024-02-01:
@everedero Congrats for your blog post https://redero.fr/?p=257 A nice Zephyr introduction!