zephyrproject-rtos/zephyr #68449: STM32: QSPI: Dual Flash mode not always supported

erwango, 2024-02-02:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/68423

<div type='discussions-op-text'>

<sup>Originally posted by **tt0776** February  1, 2024</sup>
Maybe a Q & A for @erwango ?

In advance a apology on the way doing this, I am new in reporting this kind of things.
A the issue text directed me to make a discussion first before continue, this is it :-)

----

qspi is failing in compile stm32l476xx + stm32l486xx 

Reported error is 
`warning: implicit declaration of function 'HAL_QSPI_SetFlashID'`

Is missing define of bits QUADSPI_CR_DFM (also missing QUADSPI_CR_FSEL ..)

- [stm32l476xx.h](https://github.com/zephyrproject-rtos/hal_stm32/blob/main/stm32cube/stm32l4xx/soc/stm32l476xx.h)
- [stm32l486xx.h](https://github.com/zephyrproject-rtos/hal_stm32/blob/main/stm32cube/stm32l4xx/soc/stm32l486xx.h)

Fix is needed to be added after 
`#define QUADSPI_CR_SSHIFT              QUADSPI_CR_SSHIFT_Msk                   /*!< Sample Shift */`

Fix is to place bits missing (Copied from stm32l496xx.h)
```
#define QUADSPI_CR_DFM_Pos             (6U)
#define QUADSPI_CR_DFM_Msk             (0x1UL << QUADSPI_CR_DFM_Pos)           /*!< 0x00000040 */
#define QUADSPI_CR_DFM                 QUADSPI_CR_DFM_Msk                      /*!< Dual-flash mode */
#define QUADSPI_CR_FSEL_Pos            (7U)
#define QUADSPI_CR_FSEL_Msk            (0x1UL << QUADSPI_CR_FSEL_Pos)          /*!< 0x00000080 */
#define QUADSPI_CR_FSEL                QUADSPI_CR_FSEL_Msk                     /*!< Flash memory selection */
```

That result is tested to work on board [stm32l476g_disco](https://docs.zephyrproject.org/latest/boards/arm/stm32l476g_disco/doc/index.html)

Still having what looks to be a known warning and I see it also when compiling on other ST boards. 
Looks to works anyway on my stm32l476g_disco
```
Warning (spi_bus_bridge): /soc/quadspi@a0001000: node name for SPI buses should be 'spi'
<stdout>: Warning (spi_bus_reg): Failed prerequisite 'spi_bus_bridge'
```

There is [2.6 QUADSPI errata](https://www.st.com/resource/en/errata_sheet/dm00111498-stm32l476xxstm32l486xx-device-errata-stmicroelectronics.pdf).
Not sure it is a valide reason for not including bits and the option for getting samples working :-)
 

Using sample [samples/drivers/spi_flash](https://docs.zephyrproject.org/latest/samples/drivers/spi_flash/README.html)

```
*** Booting Zephyr OS build zephyr-v3.5.0 ***

qspi-nor-flash@0 SPI flash testing
==========================

Perform test on single sector
Test 1: Flash erase
Flash erase succeeded!

Test 2: Flash write
Attempting to write 4 bytes
Data read matches data written. Good!!

Perform test on multiple consequtive sectors
Test 1: Flash erase
Flash erase succeeded!

Test 2: Flash write
Attempting to write 4 bytes at offset 0xff000
Data read matches data written. Good!!
Attempting to write 4 bytes at offset 0x100000
Data read matches data written. Good!!
```

"stm32l476g_disco.conf" added to "boards" in sample
`CONFIG_SPI=n`

"stm32l476g_disco.overlay" added to board in sample
```
/ {

    chosen {
		zephyr,flash-controller = &n25q128a1;
	};

    aliases {
		spi-flash0 = &n25q128a1; 
	};
};

&dma2 {
	status = "okay";
};

&quadspi {
	pinctrl-0 = <&quadspi_bk1_io0_pe12 &quadspi_bk1_io1_pe13
				&quadspi_bk1_io2_pe14 &quadspi_bk1_io3_pe15
				&quadspi_ncs_pe11 &quadspi_clk_pe10>;
	pinctrl-names = "default";
	
	// Copied from stm32l496 that had the STM32_DMA_.. values as 0x480>; 
	dmas = <&dma2 7 3 (STM32_DMA_PERIPH_TX | STM32_DMA_PERIPH_RX | STM32_DMA_PRIORITY_HIGH)>; /* channel 7 request 3 on DMA2 */
	dma-names = "tx_rx";

	flash-id = <1>;
	status = "okay";
	
    n25q128a1: qspi-nor-flash@0 {
		compatible = "st,stm32-qspi-nor";
		reg = <0>;
		qspi-max-frequency = <DT_FREQ_M(108)>;
		size = <DT_SIZE_M(128)>; /* 16 MBytes */
		status = "okay";
		spi-bus-width = <4>;
		
		// Was original writeoc = "PP_1_4_4", the stm32l476g_disco only worke with PP_1_1_4 ;
		writeoc = "PP_1_1_4"; 
	};

    partitions {
         compatible = "fixed-partitions";
         #address-cells = <1>;
         #size-cells = <1>;

         partition@0 {
             label = "nor";
             reg = <0x00000000 DT_SIZE_M(4)>;
         };
 	};
};
```
</div>

FRASTM, 2024-02-06:
I put a commit 9c29505cfef600226730dffea830eb9d59599427 in my PR #68607 to accept the  DualFlash Mode of dual qspi configuration (when the DTS has no <flash-id> )

erwango, 2024-02-06:
Converting to enhancement.

FRASTM, 2024-05-06:
The stm32L47x  and stm32L48x  does NOT support the Dual-flash Mode
Refer to the [RM0351](https://www.st.com/resource/en/reference_manual/rm0351-stm32l47xxx-stm32l48xxx-stm32l49xxx-and-stm32l4axxx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf)

FRASTM, 2024-05-06:
However we could still add a check   on  bitfield  **QUADSPI_CR_FSEL**   

```patch
diff --git a/drivers/flash/flash_stm32_qspi.c b/drivers/flash/flash_stm32_qspi.c
index 326c1129a36..6c6ab7fd6cf 100644
--- a/drivers/flash/flash_stm32_qspi.c
+++ b/drivers/flash/flash_stm32_qspi.c
@@ -1283,7 +1283,8 @@ static int flash_stm32_qspi_init(const struct device *dev)
 
 	HAL_QSPI_Init(&dev_data->hqspi);
 
-#if DT_NODE_HAS_PROP(DT_NODELABEL(quadspi), flash_id)
+#if DT_NODE_HAS_PROP(DT_NODELABEL(quadspi), flash_id) && \
+	defined(QUADSPI_CR_FSEL)
 	uint8_t qspi_flash_id = DT_PROP(DT_NODELABEL(quadspi), flash_id);
 
 	HAL_QSPI_SetFlashID(&dev_data->hqspi,
```

closed 2024-05-16 by carlescufi