zephyrproject-rtos/zephyr #68497: doc: Documentation issue in 'boards/arm/nrf52840dongle_nrf52840/doc/index': pkg command does not exist

masisley, 2024-02-03:
**Describe the bug**

nrfutil does not seem to contain a pkg command in the most recently available version: https://www.nordicsemi.com/Products/Development-tools/nrf-util

nrfutil search returns 7 installable commands, none of which are pkg.

**Environment**

* Page: `boards/arm/nrf52840dongle_nrf52840/doc/index`
* Version: main
* SHA-1: 4f1cd0e42858b8b7f02512bfca603332bc90edee

masisley, 2024-02-03:
Perhaps i need to use the  version here, but it is marked as deprecated: https://github.com/NordicSemiconductor/pc-nrfutil

I also read that python >=3.10 has problems with nrfutil: https://github.com/NordicSemiconductor/pc-nrfutil/issues/366 - I am currently using 3.11 which i believe was installed by the Nordic installer (?), maybe i will have better luck if i also install 3.9…

masisley, 2024-02-03:
I had to run:
nrfutil install nrf5sdk-tools
After that, pkg was available. We should update the documentation to point to the non- deprecated version and include install instructions for the new nrf5sdk-tools

henrikbrixandersen, 2024-02-06:
CC: @carlescufi

closed 2024-06-23 by github-actions[bot]