zephyrproject-rtos/zephyr #68508: Building with CMake no longer works

talih0, 2024-02-04:
Building with CMake directly gives errors:

$ cmake -Bbuild -DBOARD=xmc47_relax_kit samples/hello_world/
$ make -Cbuild

100%] Linking C executable zephyr.elf
/home/andriy/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: cannot open linker script file zephyr/isr_tables_swi.ld: No such file or directory
collect2: error: ld returned 1 exit status

While building with west works fine:
$ west build -p -b xmc47_relax_kit samples/hello_world/

This issue is most likely caused by PR: 
https://github.com/zephyrproject-rtos/zephyr/pull/66392

talih0, 2024-02-04:
@rakons please have a look

henrikbrixandersen, 2024-02-04:
For me, building with `west` doesn't work either after https://github.com/zephyrproject-rtos/zephyr/commit/1ff24b34ebe568a5d606928af69914de946b9762:

```
$ west build -b frdm_k64f samples/hello_world/ 2>&1 | tail -n 7
[ 97%] Linking C executable zephyr.elf
/opt/zephyr-sdk/0.16.5/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: cannot open linker script file zephyr/isr_tables_swi.ld: No such file or directory
collect2: error: ld returned 1 exit status
gmake[2]: *** [zephyr/CMakeFiles/zephyr_final.dir/build.make:143: zephyr/zephyr.elf] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:3216: zephyr/CMakeFiles/zephyr_final.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2
FATAL ERROR: command exited with status 2: /home/brix/Projects/zephyrproject/venv/bin/cmake --build /home/brix/Projects/zephyrproject/zephyr/build

```

rakons, 2024-02-05:
Sure. It obviously is. I am surprised it was not detected by CI when the PR was processed. Well - sometimes green CI gives us too much of confidence. Seems like easy fix - give me like an hour.

rakons, 2024-02-05:
I am a little bit confused. As it builds on my machine. I had tested it for https://github.com/zephyrproject-rtos/zephyr/commit/1ff24b34ebe568a5d606928af69914de946b9762, current main, and my merge branch. Could you possibly provide the full build log? @henrikbrixandersen  @talih0

57300, 2024-02-05:
I can reproduce this with Makefile generator, but not Ninja.

henrikbrixandersen, 2024-02-05:
@rakons Full build log:

```$ west build -p always -b frdm_k64f samples/hello_world/
-- west build: making build dir /home/hebad/Projects/zephyrproject/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/hebad/Projects/zephyrproject/zephyr/samples/hello_world
-- CMake version: 3.27.7
-- Found Python3: /home/hebad/Projects/zephyrproject/venv/bin/python3 (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/hebad/.cache/zephyr
-- Zephyr version: 3.5.99 (/home/hebad/Projects/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: frdm_k64f
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.5 (/opt/zephyr-sdk/0.16.5)
-- Found toolchain: zephyr 0.16.5 (/opt/zephyr-sdk/0.16.5)
-- Found Dtc: /opt/zephyr-sdk/0.16.5/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/hebad/Projects/zephyrproject/zephyr/boards/arm/frdm_k64f/frdm_k64f.dts
-- Generated zephyr.dts: /home/hebad/Projects/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/hebad/Projects/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/hebad/Projects/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing /home/hebad/Projects/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/hebad/Projects/zephyrproject/zephyr/boards/arm/frdm_k64f/frdm_k64f_defconfig'
Merged configuration '/home/hebad/Projects/zephyrproject/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/hebad/Projects/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/hebad/Projects/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /opt/zephyr-sdk/0.16.5/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk/0.16.5/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
Load components for MK64F12:
driver_common component is included.
driver_reset component is included.
device_CMSIS component is included.
CMSIS_Include_core_cm component is included.
device_system component is included.
driver_uart component is included.
driver_port component is included.
driver_sysmpu component is included.
-- Using ccache: /usr/bin/ccache
CMake Warning at /home/hebad/Projects/zephyrproject/zephyr/CMakeLists.txt:864 (message):
  No SOURCES given to Zephyr library: lib__posix__options

  Excluding target from build.

-- Configuring done (20.2s)
-- Generating done (0.1s)
-- Build files have been written to: /home/hebad/Projects/zephyrproject/zephyr/build
-- west build: building application
[  0%] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
[  2%] Built target parse_syscalls_target
[  3%] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h, include/generated/otype-to-size.h
[  3%] Built target kobj_types_h_target
[  4%] Generating include/generated/syscall_dispatch.c, include/generated/syscall_list.h
[  4%] Built target syscall_list_h_target
[  5%] Generating include/generated/driver-validation.h
[  5%] Built target driver_validation_h_target
[  6%] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
[  6%] Built target offsets
[  7%] Generating include/generated/offsets.h
[  7%] Built target offsets_h
[  7%] Generating include/generated/version.h
-- Zephyr version: 3.5.99 (/home/hebad/Projects/zephyrproject/zephyr), build: zephyr-v3.5.0-5374-g826d67af9169
[  7%] Built target version_h
[  7%] Built target zephyr_generated_headers
[  8%] Building C object CMakeFiles/app.dir/src/main.c.obj
[  8%] Linking C static library app/libapp.a
[  8%] Built target app
[  9%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/heap/heap.c.obj
[... 112 lines cut ...]
[ 89%] Built target drivers__timer
[ 89%] Building C object modules/hal_nxp/hal_nxp/CMakeFiles/..__modules__hal__nxp.dir/mcux/mcux-sdk/devices/MK64F12/drivers/fsl_clock.c.obj
[ 90%] Building C object modules/hal_nxp/hal_nxp/CMakeFiles/..__modules__hal__nxp.dir/mcux/mcux-sdk/drivers/common/fsl_common.c.obj
[ 91%] Building C object modules/hal_nxp/hal_nxp/CMakeFiles/..__modules__hal__nxp.dir/mcux/mcux-sdk/drivers/common/fsl_common_arm.c.obj
[ 92%] Building C object modules/hal_nxp/hal_nxp/CMakeFiles/..__modules__hal__nxp.dir/mcux/mcux-sdk/devices/MK64F12/system_MK64F12.c.obj
[ 92%] Building C object modules/hal_nxp/hal_nxp/CMakeFiles/..__modules__hal__nxp.dir/mcux/mcux-sdk/drivers/uart/fsl_uart.c.obj
[ 93%] Building C object modules/hal_nxp/hal_nxp/CMakeFiles/..__modules__hal__nxp.dir/mcux/mcux-sdk/drivers/sysmpu/fsl_sysmpu.c.obj
[ 94%] Linking C static library lib..__modules__hal__nxp.a
[ 94%] Built target ..__modules__hal__nxp
[ 95%] Building C object zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj
[ 96%] Linking C executable zephyr_pre0.elf
Logical command for additional byproducts on target: zephyr_pre0
[ 96%] Built target zephyr_pre0
[ 96%] Generating linker.cmd
[ 96%] Built target linker_zephyr_final_script_target
[... cut ...]
/opt/zephyr-sdk/0.16.5/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: cannot open linker script file zephyr/isr_tables_swi.ld: No such file or directory
collect2: error: ld returned 1 exit status
gmake[2]: *** [zephyr/CMakeFiles/zephyr_final.dir/build.make:143: zephyr/zephyr.elf] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:3215: zephyr/CMakeFiles/zephyr_final.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2
FATAL ERROR: command exited with status 2: /home/hebad/Projects/zephyrproject/venv/bin/cmake --build /home/hebad/Projects/zephyrproject/zephyr/build```

henrikbrixandersen, 2024-02-05:
> I can reproduce this with Makefile generator, but not Ninja.

I can confirm that changing the CMake generator from "Unix Makefiles" to "Ninja" allows the linking phase to complete.

carlescufi, 2024-02-05:
> > I can reproduce this with Makefile generator, but not Ninja.
> 
> I can confirm that changing the CMake generator from "Unix Makefiles" to "Ninja" allows the linking phase to complete.

@rakons is working on a hotfix right now

rakons, 2024-02-05:
Quick fix: https://github.com/zephyrproject-rtos/zephyr/pull/68530

I am working on the fix for all generators - but this one is valid anyway.

PR #68530 (2024-02-05): arch: arm, arm64: Disable swi_tables.ld file when not required
This commit removes the need of swi_tables.ld file if the ISR table generator is not configured to use it.

Fixes #68508
closed 2024-02-05 by henrikbrixandersen