zephyrproject-rtos/zephyr #68512: Add Support for Adafruit ESP32-S3 and ESP32-S2 Feather Boards

zpm1066, 2024-02-04:
**Is your feature request related to a problem? Please describe.**
This is a request to add the very popular family of `Adafruit ESP32-S3 Feather` and `ESP32-S2 Feather` boards to Zephyr.

**Describe the solution you'd like**
`Xtensa ESP32-S3 and ESP32-S2`development boards from `Espressif` and a few others are already supported.

I'd like to see a common board definition to support these two `Adafruit ESP32-S3 & ESP32-S2 Feather` boards. Folks can then add their own overlays to support the board variants with on-board TFT, fuel gauge, sensors, or Flash/PSRAM.

**Describe alternatives you've considered**
Currently there isn't any support for the `Adafruit ESP32-S3 & ESP32-S2 Feather` boards.

**Additional context**
Here is a list of the current `Adafruit ESP3-S3 & ESP32-S2 Feather` boards that this request will be able to support.

- **Adafruit ESP32-S3 Feather**
    - 8MB Flash / No PSRAM, Fuel Gauge MAX17048,  https://www.adafruit.com/product/5323 
    - 4MB Flash / 2MB PSRAM, MAX17048 https://www.adafruit.com/product/5477 
    - 4MB Flash / 2MB PSRAM, TFT ST7789 240x135, MAX17048 https://www.adafruit.com/product/5483 
    - 4MB Flash / 2MB PSRAM, MAX17048, Reverse TFT ST7789 240x135 + buttons https://www.adafruit.com/product/5691 
    
- **Adafruit ESP32-S2 Feather**
    - 8MB Flash / 2MB PSRAM , https://www.adafruit.com/product/5300 
    - 4MB Flash / 2MB PSRAM, MAX17048, https://www.adafruit.com/product/5000 
    - 4MB Flash / 2MM PSRAM, MAX17048, TFT ST7789 240x135, https://www.adafruit.com/product/5300 
    - 4MB Flash / 2MB PSRAM, MAX17048, Reverse TFT ST7789 240x135 + buttons https://www.adafruit.com/product/5345 
    - 4MB Flash / 2MB PSRAM, MAX17048, BME280 sensor, https://www.adafruit.com/product/5303
    - 16MB Flash / 8MB PSRAM , https://www.adafruit.com/product/4769

zpm1066, 2024-02-09:
Hello,

I’ve been looking to see if I can leverage board xiao_esp32s3 or esp32s3_devkitm, as the Adafruit esp32s3 Feather also uses ESP32-S3-Mini-1 SoC. 

If so, I expect the changes to add support for the Adafruit esp32s3 boards may be relatively small.

I’ll be available for any testing if needed.

zpm1066, 2024-02-09:
Just ran a few quick tests. It looks like I’ll be able use board xiao_esp32s3 for some initial tests with the Adafruit ESP32S3 Feather boards. 

I expected esp32s3_devkitm boards to work with the Adafruit esp32s3 boards as both are ESP32-S3-Mini-1 but no. Perhaps some mismatch flash partitions. 

Anyway, for now I’ll override GPIOs and I2C in an overlay to conduct some tests using xiao_esp32s3 and provide some updates.

sylvioalves, 2024-02-09:
You should be able to get that working adding overlays and necessary configs. You are welcome to submit a PR to bring support to this boards.

zpm1066, 2024-02-09:
Yes - xiao_esp32s3 board works!
Build using  esp32s3_ devkitm did’t work though it is ESP32-S3-Mini-1.

@sylvioalves - Any reason why esp32s3_ devkitm didn’t work? The esp32s3 TFT Feather is 4MB Flash and 2MB PSRAM.

zpm1066, 2024-02-14:
I've setup a new board `Adafruit_esp32s3_tft_feather` locally.

Added  and tested: 
- onboard GPIOs - builtin led, user button
- I2C with SSD1306 and a sensor
- SPI with ST7789 135x240 (same as the onboard display)
- feather_header connector
- on-board `MAX17048`

Items remaining to configure & test
- `regulator  control via GPIO` to control power to the `I2C STEMMA` connector and onboard ST7789 display via GPIO 21
- I2C device connected to STEMMA connector
- On-board ST7789 135x240 display
- `Neopixel` (Zephyr GPIO "bit-banging" is only supported for couple of boards, not sure what options exist)

When using the regulator devicetree binding control via GPIO 21, I've verified that the regulator comes on at power-on I2C STEMMA connector, seeing 3.26V. I2C is in a READY state but the SSD1306 is DISABLED.

Below, you can see that I2C is  READY and has sent commands to SSD1306 but its initialization failed.

Under Arduino, init_variant() call sets GPIO 21 high to enable power to I2C STEMMA and onboard TFT. The regulator devicetree binding does the same in Zephyr.

I expected SSD1306 to come up. 
Does anyone have any suggestion what may be going on?

A few other things to try
-  use `gpio-hog` but I haven't found any sample code and also don't know if it'll work with ESP32-S3 if there isn't a Espressif compatible that supports gpio-hog.

-  use a `pre-kernel SYS_INIT() `call.

Does anyone have any suggestions as how best to initialize the GPIO for the TFT/I2C STEMMA connector power?

Thanks.

```
00:00:00.228,000] <err> i2c_esp32: I2C transfer error: -14
[00:00:00.228,000] <err> ssd1306: Failed to initialize device!
*** Booting Zephyr OS build zephyr-v3.5.0-4872-gcf14d4f1fd6d ***
[00:00:00.228,000] <inf> main: main_thread
Display device not ready
[00:00:00.228,000] <dbg> cfb: cfb_framebuffer_init: number of fonts 2
[00:00:00.317,000] <inf> display: Index[0] font dimensions  5x8
[00:00:00.317,000] <inf> display: Index[1] font dimensions  8x8
[00:00:00.317,000] <inf> display: Selected font: index[3]
[00:00:00.317,000] <inf> display: x_res 128, y_res 64, ppt 8, rows 8, cols 128
[00:00:00.317,000] <inf> display: Zephyr v3.5.99

[00:00:00.317,000] <inf> display: xiao_esp32s3
uart:~$ device list
devices:
- rtc@60021000 (READY)
- gpio@60004800 (READY)
- gpio@60004000 (READY)
- trng@6003507c (READY)
- uart@60000000 (READY)
- uart@60038000 (READY)
- i2c@60013000 (READY)
- reg-tft-power (READY)
- ssd1306@3c (DISABLED)
uart:~$

```

DBS06, 2025-02-03:
Thanks to the awesome work done by @leonrinkel I was able to get the display working.

List of working and tested features/peripherals for [Adafruit ESP32-S2 TFT Feather](https://learn.adafruit.com/adafruit-esp32-s2-tft-feather/overview):

- [x] On-Board_LED
- [x] On-Board-Button 
- [x] NeoPixel
- [x] TFT-Display
- [ ] Add STEMMA QT Connector -> shares the same power pin as the display.
- [ ] UART over USB connector -> still a pain in the a**
- [ ] USB -> seems not to be supported until now
- [ ] Battery Monitor (I have a board with the Adafruit LC709203F, but there is also a version out there MAX17048)

> No matter what I do or try, I am not getting it to work, or to show up as a Serial on my System (WSL2 Ubuntu)
I am not 100% sure, but I think the USB will work only directly and not over USB-to-UART and it seems USB-OTG is not yet supported, see here: https://github.com/zephyrproject-rtos/zephyr/issues/29394

If I am able to get the UART and/or USB working over USB I will create a PR. Most likely and from what I already read, I would need to wait if it will be added to the SoC, if thats the case, it is what it is.
Would be nice if somebody could verify that.

If I will find my FTDI USB to UART adapter, I could test if the UART works at all.

Regarding the Battery Monitor I actually don't know how I should proceed, because there are two versions out there, a driver for the  MAX17048 already exists in zephyr, but not for the Adafruit LC709203F. Because of this, I think I will add this feature after writing the driver.

rftafas, 2025-02-03:
> I am not 100% sure, but I think the USB will work only directly and not over USB-to-UART and it seems USB-OTG is not yet supported, see here: https://github.com/zephyrproject-rtos/zephyr/issues/29394
> Does anyone can verify it?

The usage of USB on ESP32-S2 for flashing and console depends on some running code. It is NOT a hardware thing, like ESP32-C3. While implemented on ESP-IDF, it is missing in Zephyr. Yes, USB-OTG would be needed and yes, it is missing.

DBS06, 2025-02-03:
> The usage of USB on ESP32-S2 for flashing and console depends on some running code. It is NOT a hardware thing, like ESP32-C3. While implemented on ESP-IDF, it is missing in Zephyr. Yes, USB-OTG would be needed and yes, it is missing.

Thanks a lot! So it seems at least I am not stupid or overlooked something. So therefore as long as USB-OTG is not supported we are not able to use it. At least it seems that somebody will add it, or is willing to -> https://github.com/zephyrproject-rtos/zephyr/issues/29394#issuecomment-2613914376

kartben, 2025-05-22:
Hey, unless I'm missing something, we can call victory on this one (thanks @DBS06!)

closed 2025-05-22 by kartben

DBS06, 2025-05-22:
@kartben you and the community are welcome!
And yes we can call this a victory 😉

DBS06, 2025-05-22:
@kartben @zpm1066 also @leonrinkel should be mentioned too, he implemented the Feather ESP32-S3 boards and did a lot of pre-work which was very useful for me to implement the feather ESP32-S2 boards.