zephyrproject-rtos/zephyr #68537: kernel: sleep: kernel.common.timing.minimallibc fails

katgiadla, 2024-02-05:
**Describe the bug**
The test `kernel.common.timing.minimallibc` fails.

Observed for `nrf5340dk_nrf5340_cpunet`.

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf5340dk_nrf5340 connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/kernel/sleep -p nrf5340dk_nrf5340_cpunet --device-testing --device-serial /dev/ttyACM1 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.5.0-5140-g055ac615427f ***
Running TESTSUITE sleep
===================================================================
START - test_sleep
PASS - test_sleep in 2.005 seconds
===================================================================
START - test_sleep_forever
PASS - test_sleep_forever in 0.001 seconds
===================================================================
START - test_usleep
Assertion failed at ../../../../../../zephyr/tests/kernel/sleep/src/usleep.c:93: sleep_test_usleep: (elapsed_ms <= UPPER_BOUND_MS is false)
overslept
FAIL - test_usleep in 31.642 seconds
===================================================================
TESTSUITE sleep failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -  66.67% [sleep]: pass = 2, fail = 1, skip = 0, total = 3 duration = 33.648 seconds
- PASS - [sleep.test_sleep] duration = 2.005 seconds
- PASS - [sleep.test_sleep_forever] duration = 0.001 seconds
- FAIL - [sleep.test_usleep] duration = 31.642 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: 26e78a0f9a32cacbd08c0ab625110e38
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.16.5
- Commit SHA or Version used: zephyr-v3.5.0-5140-g055ac615427f

katgiadla, 2024-03-12:
The same issue has occurred on `nrf9160dk/nrf9160` on version `v3.6.0-530-g7df7e834d910`.

katgiadla, 2024-04-24:
The same issue has occurred on `nrf5340dk/nrf5340/cpuapp/ns` on version `v3.6.0-2817-gdbf62a5a2a84`.

bjarki-andreasen, 2024-07-10:
Built test suite for `nrf5340dk/nrf5340/cpuapp/ns`, `nrf5340dk/nrf5340/cpunet` and `nrf9160dk/nrf9160`   with `v3.7.0-rc2` using west with no errors.
`west build -p -b nrf5340dk/nrf5340/cpuapp/ns zephyr/tests/kernel/sleep/ -T kernel.common.timing.minimallibc`
```
*** Booting Zephyr OS build v3.7.0-rc2-326-g6fc525cc20d4 ***
Running TESTSUITE sleep
===================================================================
START - test_sleep
 PASS - test_sleep in 2.005 seconds
===================================================================
START - test_sleep_forever
 PASS - test_sleep_forever in 0.001 seconds
===================================================================
START - test_usleep
 PASS - test_usleep in 3.001 seconds
===================================================================
TESTSUITE sleep succeeded

------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [sleep]: pass = 3, fail = 0, skip = 0, total = 3 duration = 5.007 seconds
 - PASS - [sleep.test_sleep] duration = 2.005 seconds
 - PASS - [sleep.test_sleep_forever] duration = 0.001 seconds
 - PASS - [sleep.test_usleep] duration = 3.001 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION SUCCESSFUL
```
Closing as issue is no longer reproducible.

closed 2024-07-10 by bjarki-andreasen